duhow/xiaoai-patch #1: Does it add snapcast functionality for Xioai device

hillbicks, 2021-03-15:
Hey David,

hope you don't mind that I open an issue to ask you question. I saw your [comment ](https://github.com/badaix/snapcast/issues/750#issuecomment-752793836)on one of the snapcast issues. 

I wasn't sure whether I understood your comment correctly. Is this patch really adding snapcast functionality (amongst other things) to the xioaAI device? Meaning, a standalone speaker that can act as a snapcast client? I would appreciate it, if you can elaborate a bit. 

If that is really the case, that would be huge imho. Looking forward to your answer!

Thanks and BR

duhow, 2021-03-15:
Hi @hillbicks , 

This set of scripts and patches removes all XiaoAI propietary software and allows to install other open-source programs, build during the process.
Some of them include MPD music server, Upmpdcli, and Snapcast, both Snapserver and Snapclient, allowing to send music to speaker and act as server, with other speakers with same image being as client, thus allowing multi-room audio open source :) 

This repo is for personal usage, but the end goal is to have a "local alexa-like smart speaker" powered by open source only (or mostly).

hillbicks, 2021-03-15:
Dude, that is *huge* news. There are *so* many people actually looking for a device like this. Decent speaker with voice assistant capabilities without cloud services attached to it. It's basically the missing hardware for a lot of software projects.

I was not aware there is any hardware available which would be able to run snapcast, let alone one that is a smart speaker. I hope it's ok if I ask a couple more questions? If you're not comfortable, no worries, I understand, since it is just for your personal usage. 

- Which versions of the xioai devices have you tested this with? 
- What voice assistant system do you want to implement? I recently looked at rhasspy, but currently hardware, especially mics wihtout beamforming seem to be the problem. I would expect the hardware from xiaomi to be better in this regard. And if your process allows other opensource software to be run on this device, getting rhasspy as a client should at least be possible
- I would love to tinker around with this as well if that's not an issue for you. 

Let me know if that would be of interest to you, would love to help out here.

duhow, 2021-03-15:
I'm testing with LX06 originally. LX01 also works, but need some tweaks on the image/process.
The newer speaker L09G would be also interesting, as it improves in hardware resources, but i'm having issues with serial being blocked by the system, so didn't made any progress in there.
Also i'm blocked with LX05 as I cannot flash properly the image to the system, it may be signed or something weird.

duhow, 2021-03-15:
The process for installing this program, is to extract the original image - from same Flash Memory `dd` or some update, then use it as "base", and perform the modifications on a new squashfs image. Finally just flash the new "update" to the other partition - speaker contains 2 partitions with bootloader and system in order to rollback to previous.

As I've told earlier, I'm removing all the propietary programs, so miio is also unavailable. But you can comment part of the process to keep that programs and run them.

hillbicks, 2021-03-15:
partition size makes sense, hadn't thought about that. Even having a wifi speaker with snapcast is still a huge improvement in terms of hardware and a good starting point I think. 

I thought you were already at the stage of development where you're flashing the updated image via miio, that's at least how I've done it for the xiaomi vacuums that I own. But I guess you're using the serial ports to access it, or in case of the L09G, trying to access it. 

Since I couldn't find the LX06, I ordered the L09G and see what I can figure out, although I don't have high hopes for that, since I have zero experience with this kind of hardware reverse engineering. I also took the liberty to ask in the dustcloud telegram channel if anybody is already working on a L09G. If we're lucky, one of the dudes responsible for originally rooting the xiaomi vacuum is interested. I already offered donating a device to him, since I'm profiting from his work for some time now (3 vacuums with rooted firmware)

duhow, 2021-03-15:
Yep, in this case the flashing is done by having access to serial port, to the Linux console.
I was inspired by the CVE-2020-10263 and started testing from there - seriously, why is that a CVE? 😂

hillbicks, 2021-03-15:
Oh wow, that is a good find! Really strange that this is listed as CVE. I just joined the telegram channel for dustcloud and linking the CVE resulted in the exact same reactions. WHY THE FUCK is that a CVE?

I also wrote the authors of the CVE, asking whether they looked at the L09G. But judging from the responses I got in the telegram channel for dustcloud, I don't have high hopes for the L09G. It seems that xiaomi is cracking down on the new vacuums they're releasing, so we might have no luck there, but let's see. 

Have you enabled ssh on the fallback partition or how are you reflashing new builds onto the device? While I wait for the L09G to arrive, I'll read up on the dustcloud docs, they should give good insight into the xiaomi architecture. Any other resources you can recommend in the meantime?

duhow, 2021-03-16:
Beware this is not related to dustcloud, as that is meant only for vacuums. It would be possible to add some offline agent to have miio control anyway, but i'm not interested on it - yet. Maybe i'll need to rely on it for the Infrared control but still need to research more on that.

At the moment I'm just focusing on what programs or services can I use to make the voice assistant itself. So far the combination would be something like:

[Porcupine](https://github.com/Picovoice/porcupine) > ALSA record (Opus?) > [websocat](https://github.com/vi/websocat) > External [Vosk API server](https://github.com/alphacep/vosk-api) > `jq` > [Home Assistant](https://www.home-assistant.io/) or [NanoTTS](https://github.com/gmn/nanotts)

Probably need to deploy some small CLI/tools for Home Assistant, and I'm still having problems with `websocat` connecting but not sending the data to server.

If you check the scripts, you'll see the patching enables SSH on the image, so if you just flash the patched image on both partitions, you'll have SSH enabled permanently.

hillbicks, 2021-03-16:
Oh yeah, sure, I'm aware of that. I just mentioned it, because there are likely some similarities between the device groups that we can look at. And the miio tools should probably come into play if everything else is running and you want to go "public" with it. I guess flashing via UART is not a viable option for most people :p

One step at a time I guess, I'm just excited about the possibilities this hardware can provide and most importantly for me personally, whether the L09G can even be accessed at all. That is going to be the first challenge and probably the biggest one.

hillbicks, 2021-03-18:
Alright, just got the L09G and took it apart, fortunately that's very easy. It seems the UART port on the front are not actually connected to the board, so I removed the board from the speaker and connected the actual pins on the back of the board. When you tried to connect via serial, did you use the holes on the front or the pins on the back?

For my first test, I just used tape to attach TX, RX and GND and lo and behold, I got a serial connection. Well, sort of, it seems I can't get the serial speed right. Tried the usual ones: 1200, 2400, 4800, 9600, 19200, 38400, 57600, and 115200. What are you using for the LX06?

But I can see some boot messages, just not something that is readable and it stops after the boot is finished. 

I'll solder the pins for UART now, just taping it doesn't really cut it.

UART Front
https://i.imgur.com/uKn6U1a.png

UART Back
https://i.imgur.com/0fwQnEq.png

hillbicks, 2021-03-18:
Alright, this is all I got now. Will have a more detailed look later. 

```
AXG:BL1:d1dbf2:a4926f;FEAT:F0DC31BC:2000;POC:F;EMMC:800;NAND:0;READ:0;0.0;0.0;CHK:0;
sdio debug board detected 
TE: 140564

BL2 Built : 11:48:35, Mar 10 2020. axg gf91bf0a - jenkins@walle02-sh

set vcck to 1050 mv
set vddee to 950 mv
Board ID = 3
CPU clk: 1200MHz
DDR low power enabled
DDR3 chl: Rank0 16bit @ 912MHz
bist_test rank: 0 1d 00 3a 26 0b 41 1a 00 34 28 0f 42 00 00 00 00 00 00 00 00 00 00 00 00 693   - PASS
Rank0: 512MB(auto)-2T-13
AddrBus test pass!
NAND init
Load FIP HDR from NAND, src: 0x0000c000, des: 0x01700000, size: 0x00004000, part: 0
Load BL3x from NAND, src: 0x00010000, des: 0x01704000, size: 0x000afc00, part: 0
NOTICE:  BL31: v1.3(release):d5a9e97
NOTICE:  BL31: Built : 17:38:06, Mar 12 2020
NOTICE:  BL31: AXG secure boot!
NOTICE:  BL31: BL33 decompress pass
OPS=0x43
[Image: axg_v1.1.3489-8f09446 2020-03-12 13:58:51 jenkins@walle02-sh]
25 0c 43 00 52 d3 09 cf 88 9e 44 51 47 2e ee 46 
bl30:axg ver: 9 mode: 0
bl30:axg thermal0
[0.015886 Inits done]
secure task start!
high task start!
low task start!
ERROR:   Error initializing runtime service opteed_fast

U-Boot 2015.01-gfe79c6daed-dirty (Aug 27 2020 - 14:13:41)

DRAM:  512 MiB
Relocation Offset is: 1eec3000
mmu cfg end: 0x20000000
mmu cfg end: 0x20000000
register usb cfg[0][1] = 000000001ff74ad0
aml_i2c_init_port init regs for 0
NAND:  get_sys_clk_rate_mtd() 292, clock setting 200!
bus cycle0: 6,timing: 7
NAND device id: 98 da 90 15 76 16 
NAND device: Manufacturer ID: 0x98, Chip ID: 0x98 (Toshiba A revision NAND 2Gib TC58NVG1S3HBAI4 )
get_sys_clk_rate_mtd() 292, clock setting 200!
m3_nand_adjust_timing() sys_clk_rate 200, bus_c 6, bus_t 7
oob_fill_cnt =32 oob_size =64, bch_bytes =14
ecc mode:6 ecc_page_num=2 eep_need_oobsize=16
plane_num=1 writesize=0x800 ecc.size=0x200 bch_mode=1
oob avail size 6
Creating 1 MTD partitions on "A revision NAND 2Gib TC58NVG1S3HBAI4 ":
0x000000000000-0x000000200000 : "bootloader"
A revision NAND 2Gib TC58NVG1S3HBAI4  initialized ok
get_sys_clk_rate_mtd() 292, clock setting 200!
bus cycle0: 6,timing: 7
NAND device id: 98 da 90 15 76 16 
NAND device: Manufacturer ID: 0x98, Chip ID: 0x98 (Toshiba A revision NAND 2Gib TC58NVG1S3HBAI4 )
get_sys_clk_rate_mtd() 292, clock setting 200!
m3_nand_adjust_timing() sys_clk_rate 200, bus_c 6, bus_t 7
oob_fill_cnt =32 oob_size =64, bch_bytes =14
ecc mode:6 ecc_page_num=2 eep_need_oobsize=16
PLANE change!
plane_num=1 writesize=0x800 ecc.size=0x200 bch_mode=1
aml_nand_init :oobmul=1,oobfree.length=8,oob_size=64
oob avail size 8
nbbt=20
[... 237 lines cut ...]
[    8.450434@1] init: '/bin/logwrapper' exited with status 0
[    8.450567@1] init: command 'exec' r=0
[    8.454704@1] init: executing '/system/bin/ifconfig'
[    9.299965@3] start_addr=(0x8000), end_addr=(0x10000), buffer_size=(0x8000), smp_number_max=(4096)
[    9.333961@0] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[    9.344392@2] init: '/system/bin/ifconfig' exited with status 0
[    9.344777@2] init: command 'exec' r=0
[    9.348506@2] init: executing '/system/bin/iw'
command failed: No such device (-19)
[    9.404366@3] init: '/system/bin/iw' exited with status 237
[    9.404501@3] init: command 'exec' r=237
[    9.408991@3] init: starting 'wpa_supplicant'
[    9.412999@3] init: command 'start' r=0
Serial Number: A0SF66176
Initializing random number generator...
[   10.195598@3] unifykey:  key_unify_init() already inited!
[   10.195719@3] unifykey: name_store() 1302, name deviceid, 8
[   10.208107@3] unifykey: name_store() 1311
[   10.221560@0] unifykey: name: deviceid, size 15
[   10.271223@2] capability: warning: `dhcpcd' uses 32-bit capabilities (legacy support in use)
[   11.252859@2] configfs-gadget ff400000.dwc2_a: failed to start amlogic: -19
[   11.273853@0] BT_RADIO going: off
[   11.273889@0] AML_BT: going OFF

```

hillbicks, 2021-03-18:
Alright, last post for today. either minicom didn't work or I missed a parameter. Using putty/screen I was able to interrupt autoboot and get into the u-boot command line interface

```
axg_s420_v1_gva#printenv
EnableSelinux=enforcing
active_slot=normal
adb_setting=if itest ${factory_mode} == 1; then run clr_adb_debuggable;else run set_adb_debuggable;fi;
baudrate=115200
bcb_cmd=get_valid_slot;
boot_part=boot
bootargs=rootfstype=ramfs init=/init ramoops.pstore_en=1 ramoops.record_size=0x8000 ramoops.console_size=0x4000 mtdblock.ro_fspart=system mtdblock.ro_fspart_2nd=chrome loglevel=0 quiet logo=,loaded,androidboot.selinux=enforcing androidboot.firstboot=1 jtag=disable androidboot.hardware=amlogic androidboot.bootloader=U-Boot 2015.01-gfe79c6daed-dirty androidboot.build.expect.baseband=N/A androidboot.slot_suffix=normal defendkey=0x08300000,0x100000 androidboot.serialno=29432/A0SF66176 androidboot.deviceid=29432/A0SF66176 androidboot.country=WW android.debuggable=1 android.secure=0 console=ttyS0,115200 no_console_suspend earlycon=aml_uart,0xff803000 androidboot.rpmb_state=0
bootcmd=run storeboot
bootdelay=1
bootloader_version=U-Boot 2015.01-gfe79c6daed-dirty
clr_adb_debuggable=echo disable adb debug prop;setenv bootargs ${bootargs} android.debuggable=0 android.secure=1;setenv bootargs ${bootargs} console=ttyS9,115200 no_console_suspend;
cmdline_keys=if keyman init 0x1234; then if keyman read deviceid ${loadaddr} str; then setenv bootargs ${bootargs} androidboot.serialno=${deviceid};setenv serial ${deviceid};else setenv bootargs ${bootargs} androidboot.serialno=1234567890;setenv serial 1234567890;fi;if keyman read mac ${loadaddr} str; then setenv bootargs ${bootargs} mac=${mac} androidboot.mac=${mac};fi;if keyman read deviceid ${loadaddr} str; then setenv bootargs ${bootargs} androidboot.deviceid=${deviceid};fi;if keyman read lang ${loadaddr} str; then setenv bootargs ${bootargs} androidboot.lang=${lang};fi;if keyman read country ${loadaddr} str; then setenv bootargs ${bootargs} androidboot.country=${country};fi;if keyman read locale_lang ${loadaddr} str; then setenv bootargs ${bootargs} androidboot.locale.lang=${locale_lang};fi;if keyman read locale_region ${loadaddr} str; then setenv bootargs ${bootargs} androidboot.locale.region=${locale_region};fi;fi;
country=WW
deviceid=29432/A0SF66176
dtb_mem_addr=0x1000000
factory_mode=0
factory_reset_poweroff_protect=echo wipe_data=${wipe_data}; echo wipe_cache=${wipe_cache};if test ${wipe_data} = failed; then run storeargs;if mmcinfo; then run recovery_from_sdcard;fi;if usb start 0; then run recovery_from_udisk;fi;run recovery_from_flash;fi; if test ${wipe_cache} = failed; then run storeargs;if mmcinfo; then run recovery_from_sdcard;fi;if usb start 0; then run recovery_from_udisk;fi;run recovery_from_flash;fi; 
fdt_high=0x20000000
firstboot=1
identifyWaitTime=750
initargs=rootfstype=ramfs init=/init ramoops.pstore_en=1 ramoops.record_size=0x8000 ramoops.console_size=0x4000 mtdblock.ro_fspart="system" mtdblock.ro_fspart_2nd="chrome" loglevel=0 quiet
irremote_update=if irkey 2500000 0xe31cfb04 0xb748fb04; then echo read irkey ok!; if itest ${irkey_value} == 0xe31cfb04; then run update;else if itest ${irkey_value} == 0xb748fb04; then run update;\
fi;fi;fi;
jtag=disable
loadaddr=1080000
preboot=run bcb_cmd; run factory_reset_poweroff_protect;run upgrade_check;run storeargs;run adb_setting;run switch_bootmode;
reboot_mode=cold_boot
recovery_from_backup=setenv bootargs ${bootargs} console=ttyS0,115200 no_console_suspend earlycon=aml_uart,0xff803000;setenv bootargs ${bootargs} aml_dt=${aml_dt} recovery_part={recovery_part} recovery_offset={recovery_offset}; ubi part data; ubi getVolName data 0; ubifsmount ubi0:${volName}; ubifsload ${dtb_mem_addr} .data/dtb.img; ubifsload ${loadaddr} .data/recovery.img; wipeisb; bootm ${loadaddr}; 
recovery_from_flash=setenv bootargs ${bootargs} console=ttyS0,115200 no_console_suspend earlycon=aml_uart,0xff803000;setenv bootargs ${bootargs} aml_dt=${aml_dt} recovery_part={recovery_part} recovery_offset={recovery_offset};if imgread kernel ${recovery_part} ${loadaddr} ${recovery_offset}; then wipeisb; bootm ${loadaddr}; fi
recovery_from_sdcard=if fatload mmc 0 ${loadaddr} aml_autoscript; then autoscr ${loadaddr}; fi;if fatload mmc 0 ${loadaddr} recovery.img; then if fatload mmc 0 ${dtb_mem_addr} dtb.img; then echo sd dtb.img loaded; fi;wipeisb; bootm ${loadaddr};fi;
recovery_from_udisk=if fatload usb 0 ${loadaddr} aml_autoscript; then autoscr ${loadaddr}; fi;if fatload usb 0 ${loadaddr} recovery.img; then if fatload usb 0 ${dtb_mem_addr} dtb.img; then echo udisk dtb.img loaded; fi;wipeisb; bootm ${loadaddr};fi;
recovery_offset=0
recovery_part=recovery
rpmb_state=0
sdc_burning=sdc_burn ${sdcburncfg}
sdcburncfg=aml_sdc_burn.ini
serial=29432/A0SF66176
set_adb_debuggable=echo enable adb debug prop;setenv bootargs ${bootargs} android.debuggable=1 android.secure=0;setenv bootargs ${bootargs} console=ttyS0,115200 no_console_suspend earlycon=aml_uart,0xff803000;
stderr=serial
stdin=serial
stdout=serial
storeargs=get_bootloaderversion;setenv bootargs ${initargs} logo=${display_layer},loaded,androidboot.selinux=${EnableSelinux} androidboot.firstboot=${firstboot} jtag=${jtag}; setenv bootargs ${bootargs} androidboot.hardware=amlogic androidboot.bootloader=${bootloader_version} androidboot.build.expect.baseband=N/A;setenv bootargs ${bootargs} androidboot.slot_suffix=${active_slot};setenv bootargs ${bootargs} defendkey=0x08300000,0x100000;run cm [... cut ...]
upgrade_check=echo recovery_status=${recovery_status};if itest.s "${recovery_status}" == "in_progress"; then run storeargs; run recovery_from_flash;else fi;echo upgrade_step=${upgrade_step}; if itest ${upgrade_step} == 3; then run storeargs; run update;else if itest ${upgrade_step} == 4; then run storeargs; run update;fi;fi;
upgrade_key=if gpio input GPIOAO_3; then echo detect upgrade key; run update;fi;
upgrade_step=2
usb_burning=update 1000 3500
wipe_cache=successful
wipe_data=successful

Environment size: 6574/65532 bytes

```

I'm too tired to to look further today, but I think we're in business :rocket:

hillbicks, 2021-03-19:
I just relized the L09G runs android., so that's going to be interesting. 

I guess the LX06 runs GNU linux, right?

duhow, 2021-03-19:
Everything is GNU/Linux in the end 😂 
The other speakers also use an Android boot image / based in OpenWRT.

duhow, 2021-06-14:
Heya @hillbicks ! Do you have any news with this speaker?
Just got the new L15A and I'm struggling to find the TTL connectors in here.. 😂

hillbicks, 2021-06-16:
Hey,

nope, unfortunately not. After several days of trial and error I gave up. I couldn't figure out a way to get past the autoboot prompt and start a shell with the device, but I'm pretty sure that's just due to my lack of knowledge and fear of bricking the damn thing. 

So, if you have any suggestions or tips to share, I'm all ears. 

As for the TTL connectors, maybe they learned and removed them?

duhow, 2021-06-16:
Not at all. They remain the 3 connectors, but somehow I burned my CP2102 by testing multiple pins.
Fortunately I have a backup. :)
The next day I brick L15A into a bootloop. 😬

I'll try do another check to L09G but I expect the same certificate / signature protection set.

If you want to implement this I suggest you find a LX06 and we can check if this is still working (no new firmware version installed with this signature protection)

hillbicks, 2021-06-17:
Well, if you're also testing the L09G as well, I guess I'll wait a bit and see if you're able to get further than me before I buy the LX06. I'm having a hard time finding the LX06 online, it only seems to be available in some shops in India. 

Do you already have the L09G?

duhow, 2021-06-17:
Yep, I have 5 different models :)
https://m.es.aliexpress.com/item/4001310651216.html

mah8050, 2021-06-19:
Good work guys.
Im really interested in your progress.
I have a L09A which is chinese version of L09G
she talks chinese and I have no clue how to talk to her :)
so its really good to try to create something that anyone can speak to
and I like to help
for start I have checked amlogic documents and I want to see if its possible to use factory tools to read partitions or not
links are here:
http://openlinux.amlogic.com:8000/download/A113/
http://openlinux2.amlogic.com/download/doc/
ps: I have access to tools like nand flash reader and smd work station

duhow, 2021-06-20:
@mah8050 Interesting, so the same model L09A (Art) is Chinese while the G is Global and the public name is different...
So far we didn't manage to access the shell, looks like the input is somehow blocked.
Regarding these amlogic tools, I'm not familiar, but we can easily dump and write the MTD partitions with `dd`, no other programs are required. Despite the LX06 for example includes some write_mtd program.

duhow, 2021-06-20:
@hillbicks  from your post I'm checking that you have Uboot bootdelay enabled, so you can interrupt the boot process and get to the Uboot menu, you could do dump operations in there.
```
Hit Enter or space or Ctrl+C key to stop autoboot -- :  0 
```

I'm not seeing this option at the moment, I didn't pair the speaker to Wifi, not sure about your setup.
I found some random post that mentioned this envs include `jtag=disable` which are sent to the Linux Kernel run, and disable the user input. You may check in there.

duhow, 2021-06-20:
WOOOOT - looks like by switching off and on quick again I got into this `reboot_mode=cold_boot` and allowed me to interrupt and get the Uboot menu!

duhow, 2021-06-20:
Still testing. Here's some output. Don't run them at the moment, looks like if missing the "android" envs, device is kind of factory restarted?

```
axg_s420_v1_gva#imgread kernel boot 1080000
[imgread]szTimeStamp[2020090314331755]
[imgread]secureKernelImgSz=0x9a3000
axg_s420_v1_gva#env set bootargs 'rootfstype=ramfs init=/bin/ash ramoops.pstore_en=1 ramoops.record_size=0x8000 ramoops.console_size=0x4000 mtdblock.ro_fspart=system mtdblock.ro_fspart_2nd=chrome loglevel=7 console=ttyS0,115200'
axg_s420_v1_gva#bootm 1080000

aml log : R-2048 check pass!
aml log : R2048 check pass!
aml log : R2048 check pass!
aml log : R2048 check pass!
avb2: 0
save_power_post ...
avb2: 0
## Booting Android Image at 0x01080000 ...
reloc_addr =1dedc780
copy done
[store]Is good fdt check header, no need decrypt!
load dtb from 0x1000000 ......
      Amlogic Multi-DTB tool
      Single DTB detected
   Uncompressing Kernel Image ... OK
   kernel loaded at 0x01080000, end = 0x02121a00
   Loading Ramdisk to 1db42000, end 1deb1800 ... OK
   Loading Device Tree to 000000001db33000, end 000000001db411af ... OK
fdt_fixup_memory_banks, reg:0000000000000000

Starting kernel ...

uboot time: 63101625 us
domain-0 init dvfs: 1
[    0.000000@0] Booting Linux on physical CPU 0x0
[    0.000000@0] Linux version 4.9.113-04598-gde6945f522b7-dirty (ming.liu@droid16-sz) (gcc version 6.3.1 20170109 (Linaro GCC 6.3-2017.02) ) #81 SMP PREEMPT Thu Sep 3 14:18:06 CST 2020
[    0.000000@0] Boot CPU: AArch64 Processor [410fd034]
[    0.000000@0] efi: Getting EFI parameters from FDT:
[    0.000000@0] efi: UEFI not found.
[    0.000000@0]        07400000 - 07500000,     1024 KB, ramoops@0x07400000
[    0.000000@0] __reserved_mem_alloc_size, start:0x0000000005000000, end:0x0000000005400000, len:4 MiB
[    0.000000@0]        05000000 - 05400000,     4096 KB, linux,secmon
[    0.000000@0] cma: Reserved 8 MiB at 0x000000001f800000
[    0.000000@0] psci: probing for conduit method from DT.
[    0.000000@0] psci: PSCIv1.0 detected in firmware.
[    0.000000@0] psci: Using standard PSCI v0.2 function IDs
[    0.000000@0] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000@0] psci: SMC Calling Convention v1.1
[    0.000000@0] percpu: Embedded 26 pages/cpu @ffffffc01f750000 s68632 r8192 d29672 u106496
[    0.000000@0] Detected VIPT I-cache on CPU0
[    0.000000@0] CPU features: enabling workaround for ARM erratum 845719
[    0.000000@0] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 129024
[    0.000000@0] Kernel command line: rootfstype=ramfs init=/bin/ash ramoops.pstore_en=1 ramoops.record_size=0x8000 ramoops.console_size=0x4000 mtdblock.ro_fspart=system mtdblock.ro_fspart_2nd=chrome loglevel=7 console=ttyS0,115200 reboot_mode=cold_boot
[    0.000000@0]
[    0.000000@0] PID hash table entries: 2048 (order: 2, 16384 bytes)
[    0.000000@0] Dentry cache hash table entries: 65536 (order: 7, 524288 bytes)
[    0.000000@0] Inode-cache hash table entries: 32768 (order: 6, 262144 bytes)
[    0.000000@0] Memory: 480040K/524288K available (9468K kernel code, 1222K rwdata, 2508K rodata, 3776K init, 635K bss, 31960K reserved, 12288K cma-reserved)
[    0.000000@0] Virtual kernel memory layout:
[    0.000000@0]     modules : 0xffffff8000000000 - 0xffffff8008000000   (   128 MB)
[    0.000000@0]     vmalloc : 0xffffff8008000000 - 0xffffffbebfff0000   (   250 GB)
[    0.000000@0]       .text : 0xffffff8009080000 - 0xffffff80099c0000   (  9472 KB)
[    0.000000@0]     .rodata : 0xffffff80099c0000 - 0xffffff8009c40000   (  2560 KB)
[    0.000000@0]       .init : 0xffffff8009c40000 - 0xffffff8009ff0000   (  3776 KB)
[    0.000000@0]       .data : 0xffffff8009ff0000 - 0xffffff800a121a00   (  1223 KB)
[    0.000000@0]        .bss : 0xffffff800a121a00 - 0xffffff800a1c06dc   (   636 KB)
[    0.000000@0]     fixed   : 0xffffffbefe7fb000 - 0xffffffbefec00000   (  4116 KB)
[    0.000000@0]     PCI I/O : 0xffffffbefee00000 - 0xffffffbeffe00000   (    16 MB)
[    0.000000@0]     vmemmap : 0xffffffbf00000000 - 0xffffffc000000000   (     4 GB maximum)
[    0.000000@0]               0xffffffbf00000000 - 0xffffffbf00800000   (     8 MB actual)
[    0.000000@0]     memory  : 0xffffffc000000000 - 0xffffffc020000000   (   512 MB)
[... 155 lines cut ...]
[   10.313509@1] init: command 'exec' r=0
[   10.317623@1] init: executing '/system/bin/ifconfig'
[   11.174281@3] start_addr=(0x8000), end_addr=(0x10000), buffer_size=(0x8000), smp_number_max=(4096)
[   11.202171@3] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[   11.215558@2] init: '/system/bin/ifconfig' exited with status 0
[   11.215940@2] init: command 'exec' r=0
[   11.219884@2] init: executing '/system/bin/iw'
command failed: No such device (-19)
[   11.263601@0] init: '/system/bin/iw' exited with status 237
[   11.263747@0] init: command 'exec' r=237
[   11.268399@0] init: starting 'wpa_supplicant'
[   11.272302@0] init: command 'start' r=0
Serial Number: bin
Initializing random number generator...
[   12.041472@2] unifykey:  key_unify_init() already inited!
[   12.041624@2] unifykey: name_store() 1302, name deviceid, 8
[   12.046927@2] unifykey: name_store() 1311
[   12.067588@2] unifykey: name: deviceid, size 15
[   12.107315@2] capability: warning: `dhcpcd' uses 32-bit capabilities (legacy support in use)
[   12.620677@1] BT_RADIO going: off
[   12.620715@1] AML_BT: going OFF
[   12.847739@1] BT_RADIO going: on
[   12.847775@1] AML_BT: going ON
[   13.134872@0] configfs-gadget ff400000.dwc2_a: failed to start amlogic: -19
```

duhow, 2021-06-20:
```
axg_s420_v1_gva#env print jtag
jtag=disable
axg_s420_v1_gva#jtagon apao 1
efuse_pw_en: 0x0
WARNING! efuse bits is disabled
Enable A53 JTAG to AO
``` 

still nothing.

duhow, 2021-06-20:
Some other output, but still can't write to the console. I'm starting to think that this will require soldering an USB header and do via adb? Hope not :( 

```
axg_s420_v1_gva#env print bootargs
bootargs=init="/bin/busybox sh" loglevel=7 console=ttyS0,115200 no_console_suspend earlycon=aml_uart,0xff803000 emergency
axg_s420_v1_gva#run bootcmd
[imgread]szTimeStamp[2020090314331755]
[imgread]secureKernelImgSz=0x9a3000
aml log : R-2048 check pass!
aml log : R2048 check pass!
aml log : R2048 check pass!
aml log : R2048 check pass!
avb2: 0
save_power_post ...
avb2: 0
## Booting Android Image at 0x01080000 ...
reloc_addr =1dedc5d0
copy done
[store]Is good fdt check header, no need decrypt!
load dtb from 0x1000000 ......
      Amlogic Multi-DTB tool
      Single DTB detected
   Uncompressing Kernel Image ... OK
   kernel loaded at 0x01080000, end = 0x02121a00
   Loading Ramdisk to 1db41000, end 1deb0800 ... OK
   Loading Device Tree to 000000001db32000, end 000000001db401af ... OK
fdt_fixup_memory_banks, reg:0000000000000000

Starting kernel ...

uboot time: 154375038 us
[    0.000000@0] Booting Linux on physical CPU 0x0
[    0.000000@0] Linux version 4.9.113-04598-gde6945f522b7-dirty (ming.liu@droid16-sz) (gcc version 6.3.1 20170109 (Linaro GCC 6.3-2017.02) ) #81 SMP PREEMPT Thu Sep 3 14:18:06 CST 2020
[    0.000000@0] Boot CPU: AArch64 Processor [410fd034]
[    0.000000@0] earlycon: aml_uart0 at MMIO 0x00000000ff803000 (options '')
[    0.000000@0] bootconsole [aml_uart0] enabled
[    0.000000@0] efi: Getting EFI parameters from FDT:
[    0.000000@0] efi: UEFI not found.
[    0.000000@0]        07400000 - 07500000,     1024 KB, ramoops@0x07400000
[    0.000000@0] __reserved_mem_alloc_size, start:0x0000000005000000, end:0x0000000005400000, len:4 MiB
[    0.000000@0]        05000000 - 05400000,     4096 KB, linux,secmon
[    0.000000@0] cma: Reserved 8 MiB at 0x000000001f800000
[    0.000000@0] psci: probing for conduit method from DT.
[    0.000000@0] psci: PSCIv1.0 detected in firmware.
[    0.000000@0] psci: Using standard PSCI v0.2 function IDs
[    0.000000@0] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000@0] psci: SMC Calling Convention v1.1
[    0.000000@0] percpu: Embedded 26 pages/cpu @ffffffc01f750000 s68632 r8192 d29672 u106496
[    0.000000@0] Detected VIPT I-cache on CPU0
[    0.000000@0] CPU features: enabling workaround for ARM erratum 845719
[    0.000000@0] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 129024
[    0.000000@0] Kernel command line: init="/bin/busybox sh" loglevel=7 console=ttyS0,115200 no_console_suspend earlycon=aml_uart,0xff803000 emergency reboot_mode=cold_boot
[    0.000000@0]
[    0.000000@0] PID hash table entries: 2048 (order: 2, 16384 bytes)
[    0.000000@0] Dentry cache hash table entries: 65536 (order: 7, 524288 bytes)
[    0.000000@0] Inode-cache hash table entries: 32768 (order: 6, 262144 bytes)
[    0.000000@0] Memory: 480040K/524288K available (9468K kernel code, 1222K rwdata, 2508K rodata, 3776K init, 635K bss, 31960K reserved, 12288K cma-reserved)
[    0.000000@0] Virtual kernel memory layout:
[    0.000000@0]     modules : 0xffffff8000000000 - 0xffffff8008000000   (   128 MB)
[    0.000000@0]     vmalloc : 0xffffff8008000000 - 0xffffffbebfff0000   (   250 GB)
[    0.000000@0]       .text : 0xffffff8009080000 - 0xffffff80099c0000   (  9472 KB)
[    0.000000@0]     .rodata : 0xffffff80099c0000 - 0xffffff8009c40000   (  2560 KB)
[    0.000000@0]       .init : 0xffffff8009c40000 - 0xffffff8009ff0000   (  3776 KB)
[    0.000000@0]       .data : 0xffffff8009ff0000 - 0xffffff800a121a00   (  1223 KB)
[    0.000000@0]        .bss : 0xffffff800a121a00 - 0xffffff800a1c06dc   (   636 KB)
[    0.000000@0]     fixed   : 0xffffffbefe7fb000 - 0xffffffbefec00000   (  4116 KB)
[    0.000000@0]     PCI I/O : 0xffffffbefee00000 - 0xffffffbeffe00000   (    16 MB)
[    0.000000@0]     vmemmap : 0xffffffbf00000000 - 0xffffffc000000000   (     4 GB maximum)
[    0.000000@0]               0xffffffbf00000000 - 0xffffffbf00800000   (     8 MB actual)
[    0.000000@0]     memory  : 0xffffffc000000000 - 0xffffffc020000000   (   512 MB)
[... 185 lines cut ...]
[    9.652930@1] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[    9.666218@0] init: '/system/bin/ifconfig' exited with status 0
[    9.666587@0] init: command 'exec' r=0
[    9.670558@0] init: executing '/system/bin/iw'
command failed: No such device (-19)
[    9.714246@0] init: '/system/bin/iw' exited with status 237
[    9.714372@0] init: command 'exec' r=237
[    9.719102@0] init: starting 'wpa_supplicant'
[    9.722900@0] init: command 'start' r=0
[    9.723369@3] init: Created socket '/dev/socket/wpa_wlan0' with mode '660', user '1008', group '1008'
[    9.735438@0] init: executing '/sbin/populate_sn.sh'
[    9.788424@0] init: '/sbin/populate_sn.sh' exited with status 0
Serial Number: bin
Initializing random number generator...
[   10.495749@3] unifykey:  key_unify_init() already inited!
[   10.495901@3] unifykey: name_store() 1302, name deviceid, 8
[   10.501141@3] unifykey: name_store() 1311
[   10.517479@2] unifykey: name: deviceid, size 15
[   10.541806@1] capability: warning: `dhcpcd' uses 32-bit capabilities (legacy support in use)
[   11.213431@3] BT_RADIO going: off
[   11.213466@3] AML_BT: going OFF
[   11.442394@3] BT_RADIO going: on
[   11.442428@3] AML_BT: going ON
[   11.551645@1] configfs-gadget ff400000.dwc2_a: failed to start amlogic: -19
```

duhow, 2021-06-20:
Almost there! Looks like this time `/bin/sh` had some issue.

```
axg_s420_v1_gva#env set bootargs console=ttyAMA1,115200 console=ttyAMA0,115200 console=ttyS0,115200 init=/bin/sh loglevel=7 console=tty0
axg_s420_v1_gva#run bootcmd
[imgread]szTimeStamp[2020090314331755]
[imgread]secureKernelImgSz=0x9a3000
aml log : R-2048 check pass!
aml log : R2048 check pass!
aml log : R2048 check pass!
aml log : R2048 check pass!
avb2: 0
save_power_post ...
avb2: 0
## Booting Android Image at 0x01080000 ...
reloc_addr =1dedc5d0
copy done
[store]Is good fdt check header, no need decrypt!
load dtb from 0x1000000 ......
      Amlogic Multi-DTB tool
      Single DTB detected
   Uncompressing Kernel Image ... OK
   kernel loaded at 0x01080000, end = 0x02121a00
   Loading Ramdisk to 1db41000, end 1deb0800 ... OK
   Loading Device Tree to 000000001db32000, end 000000001db401af ... OK
fdt_fixup_memory_banks, reg:0000000000000000

Starting kernel ...

uboot time: 70490310 us
domain-0 init dvfs: 1
[    0.000000@0] Booting Linux on physical CPU 0x0
[    0.000000@0] Linux version 4.9.113-04598-gde6945f522b7-dirty (ming.liu@droid16-sz) (gcc version 6.3.1 20170109 (Linaro GCC 6.3-2017.02) ) #81 SMP PREEMPT Thu Sep 3 14:18:06 CST 2020
[    0.000000@0] Boot CPU: AArch64 Processor [410fd034]
[    0.000000@0] efi: Getting EFI parameters from FDT:
[    0.000000@0] efi: UEFI not found.
[    0.000000@0]        07400000 - 07500000,     1024 KB, ramoops@0x07400000
[    0.000000@0] __reserved_mem_alloc_size, start:0x0000000005000000, end:0x0000000005400000, len:4 MiB
[    0.000000@0]        05000000 - 05400000,     4096 KB, linux,secmon
[    0.000000@0] cma: Reserved 8 MiB at 0x000000001f800000
[    0.000000@0] psci: probing for conduit method from DT.
[    0.000000@0] psci: PSCIv1.0 detected in firmware.
[    0.000000@0] psci: Using standard PSCI v0.2 function IDs
[    0.000000@0] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000@0] psci: SMC Calling Convention v1.1
[    0.000000@0] percpu: Embedded 26 pages/cpu @ffffffc01f750000 s68632 r8192 d29672 u106496
[    0.000000@0] Detected VIPT I-cache on CPU0
[    0.000000@0] CPU features: enabling workaround for ARM erratum 845719
[    0.000000@0] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 129024
[    0.000000@0] Kernel command line: console=ttyAMA1,115200 console=ttyAMA0,115200 console=ttyS0,115200 init=/bin/sh loglevel=7 console=tty0 reboot_mode=cold_boot
[    0.000000@0]
[    0.000000@0] PID hash table entries: 2048 (order: 2, 16384 bytes)
[    0.000000@0] Dentry cache hash table entries: 65536 (order: 7, 524288 bytes)
[    0.000000@0] Inode-cache hash table entries: 32768 (order: 6, 262144 bytes)
[    0.000000@0] Memory: 480040K/524288K available (9468K kernel code, 1222K rwdata, 2508K rodata, 3776K init, 635K bss, 31960K reserved, 12288K cma-reserved)
[    0.000000@0] Virtual kernel memory layout:
[    0.000000@0]     modules : 0xffffff8000000000 - 0xffffff8008000000   (   128 MB)
[    0.000000@0]     vmalloc : 0xffffff8008000000 - 0xffffffbebfff0000   (   250 GB)
[    0.000000@0]       .text : 0xffffff8009080000 - 0xffffff80099c0000   (  9472 KB)
[    0.000000@0]     .rodata : 0xffffff80099c0000 - 0xffffff8009c40000   (  2560 KB)
[    0.000000@0]       .init : 0xffffff8009c40000 - 0xffffff8009ff0000   (  3776 KB)
[    0.000000@0]       .data : 0xffffff8009ff0000 - 0xffffff800a121a00   (  1223 KB)
[    0.000000@0]        .bss : 0xffffff800a121a00 - 0xffffff800a1c06dc   (   636 KB)
[    0.000000@0]     fixed   : 0xffffffbefe7fb000 - 0xffffffbefec00000   (  4116 KB)
[    0.000000@0]     PCI I/O : 0xffffffbefee00000 - 0xffffffbeffe00000   (    16 MB)
[    0.000000@0]     vmemmap : 0xffffffbf00000000 - 0xffffffc000000000   (     4 GB maximum)
[    0.000000@0]               0xffffffbf00000000 - 0xffffffbf00800000   (     8 MB actual)
[    0.000000@0]     memory  : 0xffffffc000000000 - 0xffffffc020000000   (   512 MB)
[    0.000000@0] can't find symbol:arm_dma_alloc
[    0.000000@0] can't find symbol:__alloc_from_contiguous
[... 154 lines cut ...]
[   10.099877@0] start_addr=(0x8000), end_addr=(0x10000), buffer_size=(0x8000), smp_number_max=(4096)
[   10.128907@1] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[   10.159425@2] init: '/system/bin/iw' exited with status 237
[   10.159569@2] init: command 'exec' r=237
[   10.164052@2] init: starting 'wpa_supplicant'
[   10.168290@2] init: command 'start' r=0
[   10.168824@3] init: Created socket '/dev/socket/wpa_wlan0' with mode '660', user '1008', group '1008'
[   10.180704@2] init: executing '/sbin/populate_sn.sh'
[   10.226831@2] init: '/sbin/populate_sn.sh' exited with status 0
[   10.227226@2] init: command 'exec' r=0
[   10.231082@2] init: executing '/bin/sh'
[   10.258101@3] meson_uart ff803000.serial: ttyS0 use xtal(8M) 24000000 change 115200 to 115200
Serial Number: bin
[   10.273672@1] init: '/bin/sh' exited with status 0
[   10.943376@1] unifykey:  key_unify_init() already inited!
[   10.943560@1] unifykey: name_store() 1302, name deviceid, 8
[   10.948798@1] unifykey: name_store() 1311
[   10.966653@2] unifykey: name: deviceid, size 15
[   11.008344@1] capability: warning: `dhcpcd' uses 32-bit capabilities (legacy support in use)
[   11.624240@1] BT_RADIO going: off
[   11.624299@1] AML_BT: going OFF
[   11.853059@1] BT_RADIO going: on
[   11.853114@1] AML_BT: going ON
[   11.985070@3] configfs-gadget ff400000.dwc2_a: failed to start amlogic: -19
```

mah8050, 2021-06-20:
Good progress
About usb header and getting access to adb
I realized somewhere in documents it said they use something similar to fastboot in android but with extra commands available
And that recovery partition is interesting, do you think we can boot into recovery like android devices? Even get access to shell?

duhow, 2021-06-20:
@mah8050 uboot mentions `fastboot` and there's also another possibility to insert data via USB.
I tried changing `boot_part` to `recovery` and `factory`, factory does not even work, recovery reboots itself after 2-3 minutes.
There's also a cmd to set "android debug" in uboot.

mah8050, 2021-06-21:
Today I had time to poke around my speaker a bit
and guess what, I have login prompt

``▒▒▒▒AXG:BL1:d1dbf2:a4926f;FEAT:E0DC318C:2000;POC:F;EMMC:800;NAND:0;READ:0;0.0;CHK:0;
sdio debug board detected
TE: 23724

BL2 Built : 15:46:00, Sep 29 2019. axg g8667414 - sentao.gao@droid08-bj

set vcck to 1140 mv
set vddee to 1070 mv
Board ID = 1
CPU clk: 1200MHz
DDR low power enabled
DDR DQS-calibration enabled
DDR scramble enabled
DDR3 chl: Rank0 16bit @ 792MHz - PASS
Rank0: 128MB-2T-11
DataBus test pass!
AddrBus test pass!
NAND init
chk page: 00000500
chk page: 00000540
chk page: 00000580
chk page: 000005c0
bbt blk:00000014
bbt page:00000000
0000000000000000000000000000000000000000000000000000000000000000
Load FIP HDR from NAND, src: 0x0000c000, des: 0x01700000, size: 0x00004000
Load BL3x from NAND, src: 0x00010000, des: 0x01704000, size: 0x00080000
NOTICE:  BL31: v1.3(release):87b5e26
NOTICE:  BL31: Built : 11:33:44, Apr 10 2020
NOTICE:  BL31: AXG normal boot!
NOTICE:  BL31: BL33 decompress pass
[Image: axg_v1.1.3268-b93dd79 2017-12-01 14:22:18 huan.biao@droid12]
OPS=0x43
9f e 90 e4 59 3 8b 90 f5 2c 1a d0 bl30:axg ver: 9 mode: 0
bl30:axg thermal0
[0.014405 Inits done]
secure task start!
high task start!
low task start!
ERROR:   Error initializing runtime service opteed_fast

U-Boot 2015.01 (Apr 26 2020 - 13:41:31), Build: jenkins-Mico_l09a_ota_publish-67

DRAM:  128 MiB
Relocation Offset is: 06f17000
register usb cfg[0][1] = 0000000007f89658
NAND:  nand id: 0xc8 0xd1
128MiB, SLC, page size: 2048, OOB size: 64
NAND device id: c8 d1 80 95 40 7f
NAND device: Manufacturer ID: 0xc8, Chip ID: 0xc8 (Gigadevice ESMT SLC 128MiB 3.3V 8-bit F59L1G81Mxxx)
oob avail size 6
Creating 1 MTD partitions on "ESMT SLC 128MiB 3.3V 8-bit F59L1G81Mxxx":
0x000000000000-0x000000200000 : "bootloader"
ESMT SLC 128MiB 3.3V 8-bit F59L1G81Mxxx initialized ok
nand id: 0xc8 0xd1
128MiB, SLC, page size: 2048, OOB size: 64
NAND device id: c8 d1 80 95 40 7f
NAND device: Manufacturer ID: 0xc8, Chip ID: 0xc8 (Gigadevice ESMT SLC 128MiB 3.3V 8-bit F59L1G81Mxxx)
PLANE change!
aml_nand_init :oobmul=1,oobfree.length=8,oob_size=64
oob avail size 8
bbt_start=20 env_start=24 key_start=32 dtb_start=40
nbbt: info size=0x400 max_scan_blk=24, start_blk=20
nbbt : phy_blk_addr=20, ec=0, phy_page_addr=0, timestamp=1
nbbt free list:
blockN=21, ec=-1, dirty_flag=0
blockN=22, ec=-1, dirty_flag=0
[... 143 lines cut ...]
ledserver[1165]: light_show_control 4

ledserver[1165]: open_light start

ledserver[1165]: ledtype 2

ledserver[1165]: light_show_effect 4

ledserver[1165]: open_light end

[   11.816224@0] adr fault isr
[   11.816254@0] BUG: sleeping function called from invalid context at kernel/locking/mutex.c:97
[   11.821857@0] in_atomic(): 1, irqs_disabled(): 128, pid: 1253, name: mipns-xiaomi
[   11.829271@0] Preemption disabled at:[<ffffff8009413740>] mpll_set_rate+0x60/0x1f0
[   12.424702@0] pdm_dclk is : 0
[   12.480875@3] configfs-gadget ff400000.dwc2_a: failed to start amlogic: -19
[   13.840958@3] pdm_dclk is : 0
[   13.841608@0] invalid toddr src

L09A login: root
magic[release]: 27446/A0QH497384206839F
password:
`
each time I enter a login name (like 'root') it makes different magic[release] code
I think each time it needs a different password to log in.

dgiron-midokura, 2021-06-21:
But you have this input from openwrt boot
```
Press the [1], [2], [3] or [4] key and hit [enter] to select the debug level
Please press Enter to activate this console.
```

How did you get this?

mah8050, 2021-06-21:
Just connected serial port and hit enter while booting.
It seems that the firmware is completely different in L09A as it has much less Storage and less RAM

dgiron-midokura, 2021-06-21:
Right. So maybe the system is also different, like... Maybe this is Google provided, not Xiaomi image or content at all?
If so, then there's nothing to do here - hard mode. 😂

mah8050, 2021-06-21:
clearly something is different:
mine is:
```tpl: off 8388608, size 8388608
Creating 6 MTD partitions on "ESMT SLC 128MiB 3.3V 8-bit F59L1G81Mxxx":
0x000000800000-0x000001000000 : "tpl"
0x000001000000-0x000001600000 : "boot0"
0x000001600000-0x000001c00000 : "boot1"
0x000001c00000-0x000004400000 : "system0"
0x000004400000-0x000006c00000 : "system1"
0x000006c00000-0x000008000000 : "data"
ESMT SLC 128MiB 3.3V 8-bit F59L1G81Mxxx initialized ok```
and yours is:
```[    2.362975@1] aml_nand_add_partition:392 factory bad addr=601
[    2.388135@1] Creating 7 MTD partitions on "nandnormal":
[    2.388165@1] 0x000000800000-0x000001000000 : "tpl"
[    2.396888@2] 0x000001000000-0x000001a00000 : "recovery"
[    2.402862@3] 0x000001a00000-0x000002600000 : "boot"
[    2.408643@3] 0x000002600000-0x000003600000 : "system"
[    2.415334@3] 0x000003600000-0x000007a20000 : "chrome"
[    2.423468@3]  NAND bbt detect factory Bad block at 5040000
[    2.438235@3] 0x000007a20000-0x000007e20000 : "factory"
[    2.445393@3] 0x000007e20000-0x000010000000 : "data"
[    2.465592@3]  NAND bbt detect factory Bad block at c000000
[    2.465660@3]  NAND bbt detect factory Bad block at c020000
[    2.492502@3] nandnormal initialized ok
```
That would make me sad, I really hope to make the speaker Speak in an understandable language :((

duhow, 2021-06-21:
Don't get confused, thing here is that L09A may work with this repo project, while L09G not! 😂 
Looks like the same structure of dual boot partitions. 
You may try this to get the root password.
[mi_passwd.html.txt](https://github.com/duhow/xiaoai-patch/files/6687576/mi_passwd.html.txt)

After that, dump/backup all the mtd partitions in your computer, and analyze whether they are signed or not with `binwalk`.

mah8050, 2021-06-22:
Ok I'm in
It was updated to the latest firmware (1.68.1) so it used that magic number to login but after some try and not letting it boot completely suddenly it gave up on giving magic number and I could log in using password provided by the html file, that file is a great work. after checking version number again realized that it changed into (1.64.4) and app says update available!
I've `binwak`ed through the mtd4 and other partitions and it seems that there is no DER Certificate.
So i think I can Flash :)))
If you needed more details about L09A I would be happy to help

duhow, 2021-06-22:
That's great news @mah8050 ! I suggest let's close this issue - no longer a "snapcast question", and let's work on this in a separate issue for L09A . Please create it and I'll detail you the operations to do.

closed 2021-06-22 by duhow

radekson, 2022-01-10:
Is there anyone else trying to break the L09G?

ds2k5, 2022-03-08:
Hi radekson,
I am trying.... but no success until yet

dragar, 2022-12-23:
Is there anything new about the L09G?

ds2k5, 2022-12-23:
>Is there anything new about the L09G?

No sorry not from my side!

My I will have time after XMas to "restart"

duhow, 2023-01-04:
The firmware is already located from several days ago.
`https://cdn.cnbj1.fds.api.mi-img[.]com/xiaoqiang/rom/l09g/mico_l09g_6c5c9_1.44.27.bin`

If any of you do any findings or progress in L09G, please open a new issue.

duhow, 2023-05-03:
For the people interested in L09G, you may want to have a look at https://github.com/duhow/xiaoai-patch/issues/49#issuecomment-1533440769 :)