duhow/xiaoai-patch #3: Help,how to install this patch?

Mybrc91, 2021-06-28:
I have a Xiaomi Mi AI Speaker Mini. Can you give me more detailed installation documentation? I don't know much about it. How can I login in to the device? Thank you.

duhow, 2021-06-28:
Hi! You'll need to disassemble it and connect via TTL pins. Pins are at the button-facing part, so you need to remove all parts. Also GND soldering is quite hard, does not stick as easy as the others.

Once you're there, login with `root / tina`. Make a full image backup of all partitions.

Let me know when you're done :)

Mybrc91, 2021-06-29:
> Hi! You'll need to disassemble it and connect via TTL pins. Pins are at the button-facing part, so you need to remove all parts. Also GND soldering is quite hard, does not stick as easy as the [...]

I may have to give up on LX01 because I don't have the tools, would LX06 be easier or something else?

duhow, 2021-06-29:
Same procedure. All speakers require physical disassemble and solder wires to get into shell and patch them.

Mybrc91, 2021-06-29:
> Same procedure. All speakers require physical disassemble and solder wires to get into shell and patch them.

Okay,i will try to do it.

closed 2021-06-29 by Mybrc91