duhow/xiaoai-patch #5: Question about the fallback partition on LX06

hillbicks, 2021-07-20:
Hey guys,

since I was not successful with the L09, I got my hands on an LX06. With the LX06 I was able to solder the UART, get the shell working, generate the root password from S/N with the help of the html file. So far so good. Here is where it gets interesting though. 

default boot_part is boot0, using mtdblock2 and mtdblock4 for system0. But both mtdblock3 for boot1 and mtdblock5 for system1 were completely empty. I confirmed it via xxd, but images only contained FF. So before flashing the patched image I thought I would clone mtd2 to mtd3 and mtd4 to mtd5 and see if I can boot from the other partition. Unfortunately, this doesn't work. If I stop uboot, set boot_part to boot1, I can see this error message and mtd4 is booted again. 

`NAND bbt detect factory Bad block at 6000000 `
`aml_nand_add_partition:1794 factory bad addr=300`
`ERROR: image is not a fdt - must RESET the board to recover.`  

I used dd to clone both mtd2 and mtd4 to their respective counterparts via dd with bs=4096. I was expecting this to work and am currently struggling to understand what exactly is wrong here. What I also don't get: Why were both fallback partitions empty to begin with. I also tried a factory reset in between, the error message remains.

Anybody got an idea what the problem is? Any hints would be appreciated. I can provide the full boot log if needed, for some reason github doesn't want to format it. 

Thanks!

duhow, 2021-07-20:
In fact I got the same issue at the beginning. Fresh LX06 speaker comes only with the first partition and the rest is empty, while other speakers may have a second partition, with a light system, for "Factory mode".

As you were doing, you can copy one partition into another and try to boot them, you'll need to copy both the **kernel** and the **rootfs** / system partitions.

On the other hand, NAND flash memory are meant to have bad blocks. You can try flashing with `mtd write image.img /dev/mtd5` instead of `dd`.

What's the version you're running? `cat /usr/share/mico/version`

hillbicks, 2021-07-20:
I'll trry the mtd write command tomorrow. You mean bootX and systemX partitions, right?

Version is 1.64.15

duhow, 2021-07-20:
Yes, both bootX and systemX.

You can also extract boot.img and root.squashfs files from the update file with `miso` tool or https://github.com/NyaMisty/mkxqimage_rev with `mkxq -r -x mico_firmware.bin`

https://bigota.miwifi.com/xiaoqiang/rom/lx06/mico_firmware_8b63c_1.58.13.bin

You can check more info in this forum thread:
https://bbs.hassbian.com/forum.php?mod=viewthread&tid=8754&page=1#pid283801

hillbicks, 2021-07-21:
Yes, success! Thanks!

Although I still had a bad block, but using mtd to write both images was indeed the solution. I was fist able to basically clone mtd4 to mtd5 and reboot, then I could flash the patched image and now have ssh access to the LX06!

Now the fun stuff begins 😁 

Closing this isssue

closed 2021-07-21 by hillbicks