duhow/xiaoai-patch #7: monitor.sh install is broken

hillbicks, 2021-07-23:
While going through the packages, I saw that you already build monitor.sh as one of the packages. Unfortunately, this script currently doesn't work. 

1) The symlinks for /usr/bin/monitor.sh /usr/share/monitor.sh both point to /usr/share/monitor/monitor.sh, but the actual script resides in /usr/share/monitor.sh/monitor.sh. The directory is missing the .sh 

I would have created a pull request for this, but even with this fix, this script needs a lot of overhaul to work. A lot of binaries missing, invalid options, errors due to ro FS

Couple of things that are obvious and can be removed immediately, but it wouldn't be compatible with upstream anymore. (not without major checks at least). I'll have a look over the weekeend and see if I make some dirty changes to get it running on my box at least, no promises though. 
```
/usr/share/monitor.sh/monitor.sh: line 49: git: command not found
/usr/share/monitor.sh/monitor.sh: line 64: ./support/argv: No such file or directory
/usr/share/monitor.sh/monitor.sh: line 65: ./support/init: No such file or directory
/usr/share/monitor.sh/monitor.sh: line 68: ./support/mqtt: No such file or directory
/usr/share/monitor.sh/monitor.sh: line 69: ./support/log: No such file or directory
/usr/share/monitor.sh/monitor.sh: line 70: ./support/data: No such file or directory
/usr/share/monitor.sh/monitor.sh: line 71: ./support/btle: No such file or directory
/usr/share/monitor.sh/monitor.sh: line 72: ./support/time: No such file or directory
pidof: invalid option -- 'x'
BusyBox v1.32.1 (2021-07-20 11:36:17 CEST) multi-call binary.

Usage: pidof [OPTIONS] [NAME]...

List PIDs of all processes with names that match NAMEs

        -s      Show only one PID
        -o PID  Omit given pid
                Use %PPID to omit pid of pidof's parent
mkfifo: main_pipe: Read-only file system
mkfifo: log_pipe: Read-only file system
mkfifo: packet_pipe: Read-only file system
/usr/share/monitor.sh/monitor.sh: line 150: /dev/fd/63: No such file or directory
/usr/share/monitor.sh/monitor.sh: line 151: /dev/fd/63: No such file or directory
/usr/share/monitor.sh/monitor.sh: line 150: : No such file or directory
/usr/share/monitor.sh/monitor.sh: line 152: /dev/fd/63: No such file or directory
/usr/share/monitor.sh/monitor.sh: line 151: : No such file or directory
/usr/share/monitor.sh/monitor.sh: line 152: : No such file or directory
rm: can't remove '.pids': No such file or directory
/usr/share/monitor.sh/monitor.sh: line 821: log_listener: command not found
/usr/share/monitor.sh/monitor.sh: line 823: .pids: Read-only file system
> log listener pid = 2001
/usr/share/monitor.sh/monitor.sh: line 827: btle_scanner: command not found
/usr/share/monitor.sh/monitor.sh: line 829: .pids: Read-only file system
> btle scan pid = 2002
/usr/share/monitor.sh/monitor.sh: line 833: btle_text_listener: command not found
/usr/share/monitor.sh/monitor.sh: line 835: .pids: Read-only file system
> btle text pid = 2003
/usr/share/monitor.sh/monitor.sh: line 839: btle_listener: command not found
/usr/share/monitor.sh/monitor.sh: line 841: .pids: Read-only file system
> btle listener pid = 2004
/usr/share/monitor.sh/monitor.sh: line 847: .pids: Read-only file system
/usr/share/monitor.sh/monitor.sh: line 845: mqtt_listener: command not found
> mqtt listener pid = 2005
/usr/share/monitor.sh/monitor.sh: line 853: .pids: Read-only file system
/usr/share/monitor.sh/monitor.sh: line 851: btle_packet_listener: command not found
> packet listener pid = 2006
/usr/share/monitor.sh/monitor.sh: line 859: .pids: Read-only file system
/usr/share/monitor.sh/monitor.sh: line 857: beacon_database_expiration_trigger: command not found
> beacon database time trigger pid = 2007
/usr/share/monitor.sh/monitor.sh: line 1990: main_pipe: No such file or directory
/usr/share/monitor.sh/monitor.sh: line 1987: [: : integer expression expected
```

hillbicks, 2021-07-23:
Well, let me take that back. Sort of. Apparently something got mixed up when building the image with monitor.sh. Currently rebuilding the package, will update once I've reflashed the image.

duhow, 2021-07-23:
Didn't test monitor.sh fully, but this errors were kind of expected.
You may try copying the `monitor.sh` folder and content to `/var` since that's writable.

hillbicks, 2021-07-25:
Sounds good, will have a look once I finished the music implementation via squeezelite.

hillbicks, 2021-08-03:
So, I gave it another try and copied monitor.sh to /var/ Couple of things to note: 
1) shebang points to /bin/bash, so this needs to be changed to /usr/bin/bash or symlinked to that place in order to keep changes in the actual script to a minimum. 
2) We're missing bluez_hcidump it seems. 

But with the folder on a rw partition, it already looks a lot cleaner: 

```
./monitor.sh
./monitor.sh: line 49: git: command not found
cat: can't open '.previous_version': No such file or directory
> updated monitor.sh (v. Unknown) -> (v. 0.2.200)...
> error: required package 'hcidump' not found. please install 'bluez-hcidump' (e.g., sudo apt-get install bluez-hcidump).
> warning: Recommended package 'git' not found. please consider installing for regular updates.
> mosquitto preferences file created. please customize.
> public MAC address list file created. please customize.
> beacon MAC address list file created. please customize.
```

I'll have a look at bluez_hcidump

hillbicks, 2021-08-03:
working bluez_hcidump package.mk file

```
PACKAGE_NAME="Bluetooth Linux"
PACKAGE_VERSION="2.5"
PACKAGE_SRC="http://www.kernel.org/pub/linux/bluetooth/bluez-hcidump-${PACKAGE_VERSION}.tar.xz"
PACKAGE_DEPENDS="kernel-headers glibc alsa-lib dbus sbc glib libical readline bluez"

configure_package() {
        CC="${BUILD_CC}" CFLAGS="${BUILD_CFLAGS}" LDFLAGS="${BUILD_LDFLAGS}" \
           CXX="${BUILD_CXX}" CXXFLAGS="${BUILD_CFLAGS}" CPPFLAGS="${BUILD_CFLAGS}" \
           PKG_CONFIG_LIBDIR="${BUILD_PKG_CONFIG_LIBDIR}" PKG_CONFIG_SYSROOT_DIR="${BUILD_PKG_CONFIG_SYSROOT_DIR}" \
           PKG_CONFIG_PATH="${BUILD_PKG_CONFIG_LIBDIR}" \
           ./configure --prefix=${INSTALL_PREFIX} --build=${MACHTYPE} --host=${BUILD_TARGET}
}

make_package() {
        make -j${MAKE_JOBS}
}

install_package() {
        make DESTDIR=${STAGING_DIR} install
}
```

So, that is working. Still a lot of other errors, some related to different functionality of busybox commands (timeout, pidof), which can be "resolved" quick and dirty. 

I'll try and establish a connection to my phone, so far I had no luck. Will try it on a pi to rule out that it is working with my phone.

duhow, 2022-12-01:
I've managed to build `hcidump` from "deprecated" `bluez` tools.
Still, there's plenty of work to do... https://github.com/duhow/xiaoai-patch/commit/e38f0f789ae36bfad0a54a655f0f575e42a796e2