duhow/xiaoai-patch #8: Requirements for new packages

hillbicks, 2021-07-23:
Since I ordered a couple more smart speakers for around the house, it was time for me to look at the multiroom audio solution again. Currently I'm using snapcast for two speakers and while the sync is fantastic, the usability and compatibility with home assistant and automations leaves a lot to desire imho. I want to test out Logitech media server again. The sync is not as perfect as snapcast, but it is way easier to handle I think. 

For that, I would need to build the squeezelite package into the image of course and if I'm doing that, it would make sense to provide the package.mk file to this project. Dependency wise it seems we're only missing libsoxr, everything else is already there judging from the [AUR package from arch]( https://aur.archlinux.org/packages/squeezelite/) 

I would probably use the snapcast package as a template and modify it for the squeezelite packages. I haven't looked at the contents for libsoxr yet, but one the other lib* packages should be provide a template as well. 

Any requirements from your side for the do's and don'ts of a new package? Could be a good exercise for any additional packages that might come in the future.

hillbicks, 2021-07-23:
Update so far. soxr is not missing, I just didn't see it because I was looking for libsoxr. 

By using the prebuilt binary from [sourceforge](https://sourceforge.net/projects/lmsclients/files/squeezelite/linux/squeezelite-1.9.9.1386-armhf.tar.gz/download) and copying it to tmp, squeezelite is working on the LX06. 

`./squeezelite -o pcm.default -s audio.localdomain-n Bathroom`
Next step is to write the packages.mk for it, so I can build it.

duhow, 2021-07-23:
No requirements "as-is", since the libs required are already built with other packages, best would be to build the package from source code in order to reuse same libraries and lower size of the binary program.

https://github.com/ralph-irving/squeezelite

hillbicks, 2021-07-25:
Alright, I tried to make progress with building squeezelite, but hit a wall for the moment. The output_alsa.c file contains 
`#include <alsa/asoundlib.h>`
which of course doesn't work, because it is in another directory. I have to do a bit of research to figure out how to include these directories for make. 

I'll report back once I made further progress.

duhow, 2021-07-30:
If you want, you can do a draft PR with what you have so far, and we can continue working there :)

hillbicks, 2021-08-02:
Well, to give an update on the squeezelite work. As it turns out, I have no idea what I'm doing. :joy: 

Currently make is failing because ld can't find lasound. I thought reusing the make_package section from another package like ec would solve that problem, but so far I haven't had any luck. I'll poke around a bit more and see if I can figure something out here.

PR #12 (2021-08-08): Add Squeezelite and Opusfile packages
Closes #8
closed 2021-08-08 by duhow