duhow/xiaoai-patch #10: libmad build fails

hillbicks, 2021-07-28:
While trying to get the squeezelite package to build, I saw that libmad is not actually build with the default packages. Trying to build it, fails at the moment. It seems the libmad package doesn't have support for newer arm architecture (no surprise there I guess). 

```
/tmp/ccAIzqCy.s: Assembler messages:
/tmp/ccAIzqCy.s:1378: Error: selected processor does not support `rsc lr,lr,#0' in Thumb mode
/tmp/ccAIzqCy.s:1546: Error: selected processor does not support `rsc r9,r9,#0' in Thumb mode
/tmp/ccAIzqCy.s:2044: Error: selected processor does not support `rsc lr,lr,#0' in Thumb mode
/tmp/ccAIzqCy.s:2206: Error: selected processor does not support `rsc lr,lr,#0' in Thumb mode
````

There are patches available for arch arm [arch arm](https://archlinuxarm.org/packages/arm/libmad) and [debian ](https://packages.debian.org/de/source/jessie/libmad)

I haven't been able to successfully apply the arch arm patch yet (or rather successfully build libmad), I'll give it another shot later. Also, I'm not entirely sure how to best proceed here. Might be easier to use the arch arm package as the source directly instead of the sourceforge one.

duhow, 2021-07-29:
The "arch arm" patch seems more like a pkg-config file already created, and removing optimize flags. Only the "frame_length" seems to be doing something. https://archlinuxarm.org/packages/arm/libmad/files/frame_length.diff

duhow, 2021-07-29:
Was a tricky one. Now I remember i removed as a dependency of MPD because of build problems as well... Try it and let me know. Maybe it will also need the pkgconfig / pc file?

hillbicks, 2021-07-29:
Compiles fine now, don't think we need anything else really. 

Thanks!

commit 974df9f9 (2021-07-29): Fix build libmad

Fixes #10