duhow/xiaoai-patch #17: [Workaround] hardcoded wifi mac address is used for LX01

hillbicks, 2021-08-08:
The image for the LX01 includes the firmware files nv_43436p.txt and nv_43438a1.txt. These include a hardcoded mac address for wifi, not sure what the reason is for that. I haven't looked that much into it, just encountered the behavior yesterday and  thought I share my findings. Both files include

```
macaddr=00:90:4c:c5:12:38
il0macaddr=00:90:4c:c5:12:38
```

If you search the internet for this mac address, there are a quite a few results for it. Anyway, using the hardcoded mac of course leads to problems when you have more than one LX01 in your network. (duplicate IP address for all devices with this mac). 

I created a script, well adapted really, from this [site](https://oper.io/?p=Fixing_Android_Mac_Address_Conflicts), which changes the mac address in both files to a random one. Thought I'd share in case anyone else runs into the same problem. 

28_generate_wifi_mac.sh in scripts

```
#!/bin/sh

nv_43438a1=$ROOTFS/lib/firmware/nv_43438a1.txt
nv_43436p=$ROOTFS/lib/firmware/nv_43436p.txt
echo "[*] Updating wifi mac address"
new_mac=$(printf '00:90:4c:%02x:%02x:%02x\n' $[RANDOM%256] $[RANDOM%256] $[RANDOM%256])

sed -i "s/macaddr=.*/macaddr=${new_mac}/" ${nv_43438a1}
sed -i "s/macaddr=.*/macaddr=${new_mac}/" ${nv_43436p}

sed -i "s/il0macaddr=.*/il0macaddr=${new_mac}/" ${nv_43438a1}
sed -i "s/il0macaddr=.*/il0macaddr=${new_mac}/" ${nv_43436p}

chmod 755 $nv_43438a1 $nv_43436p
chown root:root $nv_43438a1 $nv_43436p
```

duhow, 2021-08-08:
I don't think that's appropiate, since `nandg` **private** already contains the MAC address set. Should take it from there instead of generating a new random one. If it is not doing so, then that's another problem.

hillbicks, 2021-08-08:
Appropriate? Absolutely not! Does it work for me? Yes :p   

good point about nandg, just checked and the real mac is in there it seems, I just checkd nandg for two devices. Come to think of it, when setting up the devices before they were flashed, I'm pretty sure they had the original mac from nandg (can't confirm it though) Just after flashing ( and I didn't touch nandg apart from taking a backup) the modified image the problem appeared. I only noticed because the ssh host key kept changing when connecting to the same IP. 

Could you please check the mac on your lx01 so we can see if it's something on my end or with the creation of the image?
Thanks!

duhow, 2022-01-31:
Sorry for taking too long. I did have a deep look at what's going on.

When booting, `/etc/init.d/wireless` program loads the driver and tells to load config from `/data/wifi`.
Also some part is specified in `/etc/init.d/check_mac`.
That makes sense, otherwise having all devices to run with same MAC address on `/lib/firmware` would cause network issues, as you may have guessed.
In my case, field `macaddr` in `/data/wifi/` files contained the right MAC, while `il0macaddr` was set to this factory-default.

When device is booting up, it loads default values from `/lib`:
```
[   15.031432] Final fw_path=/lib/firmware/fw_bcm43438a1.bin
[   15.031440] Final nv_path=/lib/firmware/nv_43438a1.txt
[   15.031446] Final clm_path=/lib/firmware/clm_bcmdhd.blob
[   15.031452] Final conf_path=/lib/firmware/config.txt
```

But when I ask the device to restart wireless:
```
/etc/init.d/wireless restart
[  569.980478] Final fw_path=/lib/firmware/fw_bcm43438a1.bin
[  569.980486] Final nv_path=/data/wifi/nv_43438a1.txt
[  569.980494] Final clm_path=/lib/firmware/clm_bcmdhd.blob
[  569.980501] Final conf_path=/lib/firmware/config.txt
```

So here's where the problem comes from.
Although, MAC address shouldn't be changing constantly, neither the SSH key, as you reported.
That may be that `/data` partition is not mounted properly, check `nandi` and do `mkfs.ext4 nandi` if required.

commit f480f3b0 (2022-01-31): LX01: Ensure WiFi driver uses persistent config

Allows to use MAC address stored in /data/wifi configuration
Fixes #17
closed 2022-01-31 by duhow