duhow/xiaoai-patch #19: Xiaoai Speaker Play - L05B

danielk117, 2021-12-08:
Hi,
I just bought a "Xiaoai Speaker Play", but i seems to be a new revision. 😞 

The model number (l05b) is different from the one in the readme (lx05).
I found a TX0/RX0 and TX2/RX2. Any idea where i can get GND?

photos came here in a few minutes...

duhow, 2021-12-08:
Can't find it either. Try removing the 4 button rubber pad?
Also try to follow connectivity on TP43 as it seems to be power ground?

duhow, 2021-12-12:
@danielk117 any luck?

danielk117, 2021-12-13:
Unfortunately, I have not yet found the time to search further. ☹️
TP43 seems to be the power ground. I cant use it directly right?

duhow, 2021-12-13:
You should be able to use TP43 directly.
Also if you have a multimeter, you can try to find contact with any other point without powering the speaker, in order to identify any other GND.

danielk117, 2021-12-14:
> Also if you have a multimeter, you can try to find contact with any other point without powering the speaker, in order to identify any other GND.

Good idea. TP7 (next to TX0/RX0) and the screw holes seems to be ground. 

I soldered my adapter to TP7, TX0 and RX0. Which baudrate (or other settings) should work?

[image]

duhow, 2021-12-14:
Default baud rate is 115200 8n1. Also ensure to disable hardware flow control.
This garbage output may be cause due to GND missing contact.

danielk117, 2021-12-14:
nope, but 921600 8n1 seems to be working 👍 
[output_2021-12-14_09-08-28.log](https://github.com/duhow/xiaoai-patch/files/7709785/output_2021-12-14_09-08-28.log)

duhow, 2021-12-14:
NuttX? This doesn't seem to be it. Try with TX2/RX2 .

danielk117, 2021-12-14:
no, its the one without the clock, but i think they have simular model numbers (l05b and l05c, https://home.miot-spec.com/s/wifispeaker) and could have simular hardware.

connected TX2/RX2 now, but I didn't get any output. not even garbage output. 😞

danielk117, 2021-12-14:
I've done some quick research about xiaomi and NuttX... https://www.gizmochina.com/2020/11/05/xiaomi-launches-a-new-iot-software-platform-xiaomi-vela-based-on-nuttx-os/

danielk117, 2021-12-15:
switching back to TX0/RX0... It's already running a shell (NuttShell) and no login is needed

for example
```
free
             total       used       free    largest  nused  nfree
Umem:      6957696    1680368    5277328    5175360   4887     22
```

```
df -h
  Filesystem    Size      Used  Available Mounted on
  binfs           0B        0B         0B /bin
  littlefs        7M      468K      6700K /data
  romfs        1072K     1072K         0B /etc
  procfs          0B        0B         0B /proc
  tmpfs           1K        0B         1K /tmp
```

```
env
PWD=/
ALSA_CONFIG_PATH=/data/etc/alsaALSA_CONFIG_DIR=/data/etc/
SN=31834/B1VD98309
WMAC=5C:02:14:54:77:06
BMAC=5C:02:14:54:77:07
DID=504390216
KEY=##############
UID=0000000000000001
IMEI=000000000000000
TZ=Asia/Shanghai
```

```
ap> ps
  PID GROUP PRI POLICY   TYPE    NPX STATE    EVENT     SIGMASK   STACK   USED  FILLED    CPU COMMAND
    0     0   0 FIFO     Kthread N-- Ready              00000000 003052 001044  34.2%   76.0% Idle Task
    1     1 253 RR       Kthread --- Waiting  Semaphore 00000000 003044 001296  42.5%    0.0% hpwork 0x20120f3c
    2     2 253 RR       Kthread --- Waiting  Semaphore 00000000 003044 001296  42.5%    0.0% hpwork 0x20120f3c
    3     3 253 RR       Kthread --- Waiting  Semaphore 00000000 003044 001296  42.5%    0.0% hpwork 0x20120f3c
    4     4 110 RR       Kthread --- Waiting  Semaphore 00000000 008164 002256  27.6%    0.0% lpwork 0x20120f30
    5     5 110 RR       Kthread --- Waiting  Semaphore 00000000 008164 001232  15.0%    0.0% lpwork 0x20120f30
    6     6 110 RR       Kthread --- Waiting  Semaphore 00000000 008164 000896  10.9%    0.0% lpwork 0x20120f30
    8     8 101 RR       Kthread --- Waiting  Semaphore 00000000 004060 001896  46.6%    0.0% bes_main  0x34010a04
    9     9 252 RR       Kthread --- Waiting  MQ empty  00000000 006108 000528   8.6%    0.0% app_thread  0x3400fcd8
   10    10 253 RR       Kthread --- Waiting  Semaphore 00000000 008124 000856  10.5%    1.1% audio_flinger  0x20045ca8
   11    11 100 RR       Kthread --- Waiting  Signal    00000000 003028 000720  23.7%    0.0% apps_recover  0x3400fcf8
   12    12 252 RR       Kthread --- Waiting  MQ empty  00000000 010180 000944   9.2%    0.0% net_wq  0x200480c0
   13    13 252 RR       Kthread --- Waiting  MQ empty  00000000 008132 000520   6.3%    0.0% net_tasklet  0x20048088
   14    14 252 RR       Kthread --- Waiting  Semaphore 00000000 004036 001104  27.3%    0.0% cw1200_bh  0x20048174
   15    15 224 RR       Kthread --- Waiting  Semaphore 00000000 004060 001400  34.4%    0.0% rptun audio 0x2014b8c0
   16    16 100 RR       Task    --- Waiting  Semaphore 00000000 004060 002536  62.4%    0.0% rc_raw_event_thx 0x2014d5e0
   17    17 100 RR       Task    --- Waiting  Signal    00000000 004084 001848  45.2%    0.0% init
   18    16 100 RR       pthread --- Waiting  Semaphore 00000000 016388 000448   2.7%    0.0% custom_ditective_speech_log_0 0x20135560
   19    17 100 RR       pthread --- Waiting  Semaphore 00000000 016388 000680   4.1%    0.0% custom_ditective_speech_log_1 0x20155a20
   20    20 100 RR       Task    --- Waiting  Semaphore 00000000 004084 002408  58.9%    0.0% kvdbd
   22    22 252 RR       Task    --- Waiting  Semaphore 00000000 004060 001016  25.0%    1.1% rpmsg-uorb-audio 0x341ac9f0
   23    23 232 RR       Task    --- Waiting  Semaphore 00000000 006132 001904  31.0%    0.0% usrsock
   24    24 100 RR       Task    --- Waiting  Signal    00000000 004076 002844  69.7%    0.0% mico_fluorided
   25    24 100 RR       pthread --- Waiting  Semaphore 00000000 010244 006640  64.8%    0.0% bt_stack_manager_thread 0x341b3cc0
   26    24 246 FIFO     pthread --- Waiting  Semaphore 00000000 002052 000376  18.3%    0.0% alarm_deprecated 0
   27    24 246 FIFO     pthread --- Waiting  Semaphore 00000000 004100 001544  37.6%    0.0% alarm_default_ca 0x341bca28
   29    24 246 FIFO     pthread --- Waiting  Semaphore 00000000 004100 001912  46.6%    0.0% alarm_dispatcher 0x341bca28
   30    24 100 RR       pthread --- Waiting  Semaphore 00000000 004100 002276  55.5%    0.0% bt_jni_thread 0x341b9510
   32    24 246 FIFO     pthread --- Waiting  Semaphore 00000000 004100 002276  55.5%    0.0% bt_hci_thread 0x341c6220
   33    24 246 FIFO     pthread --- Waiting  Semaphore 00000000 004100 001096  26.7%    0.0% bt_rx_thread 0
   34    24 246 FIFO     pthread --- Waiting  Semaphore 00000000 012804 009008  70.3%    0.0% bt_main_thread 0x341cc110
   35    24 246 FIFO     pthread --- Waiting  Semaphore 00000000 008196 002276  27.7%    0.0% bt_a2dp_sink_worker_thread 0x341c4bc0
   36    36 101 RR       Kthread --- Waiting  MQ empty  00000000 004060 000880  21.6%    0.0% wifi_event  0x340108d8
   37    37 252 RR       Kthread --- Waiting  MQ empty  00000000 004052 000608  15.0%    0.0% bes_netdev_wq
[... 32 lines cut ...]
   62    62 100 RR       Task    --- Waiting  Signal    00000000 008172 001072  13.1%    0.0% micoplayer
   63    58 180 RR       pthread --- Waiting  Signal    00000000 016388 001792  10.9%    3.4% mico_led 0x34227310
   64    58 100 RR       pthread --- Waiting  Semaphore 00000000 016388 000448   2.7%    0.0% mico_touchpad_timer 0x34235df0
   65    65 100 RR       Task    --- Waiting  Semaphore 00000000 006116 001656  27.0%    0.0% mible_mesh_common_bt
   66    58 200 RR       pthread --- Waiting  Signal    00000000 010244 000880   8.5%    0.0% mico_touchpad 0x34235d4c
[... cut ...]
   82    44 100 RR       pthread --- Ready              00000000 016388 001680  10.2%    0.0% miot_helper 0x3429d720
   83    44 100 RR       pthread --- Waiting  Signal    00000000 006148 001608  26.1%    0.0% nightmode 0
   84    44 150 RR       pthread --- Waiting  Semaphore 00000000 016388 000536   3.2%    0.0% mico_iot_event 0x341ecbb0
   85    58 100 RR       pthread --- Waiting  Semaphore 00000000 006148 001152  18.7%    0.0% event_manager 0x3421b85c
   86    86 100 RR       Task    --- Running            00000000 002036 001552  76.2%    3.9% nsh
```

duhow, 2021-12-15:
As per output, applications seem to run directly in there...

If we're unable to get a Linux console, this may mean mean the board is using a microcontroller instead of ARM computer. Hence this wouldn't be supported here.

Unless there's another computer that can reach internally from here - like another serial port?

danielk117, 2021-12-17:
> As per output, applications seem to run directly in there...

Yes, it seems to be the new way Xiaomi is making the firmware for this device. 

I was able to set a wifi connection:
```
ifdown wlan0
ifup wlan0
wapi psk wlan0 PASSWORD 1
wapi essid wlan0 SSID 1
renew wlan0
```
and turning on telnet, by simply start the daemon:
```
telnetd
```

I browsed the whole filesystem (as NuttX/NuttShell hat all basic linux dir/file commands), but there isn't any file showing me the architecture of the chip. But I think I can confirm, that all the "magic" is done here: There are bluetooth, player or alarm daemons and I found config files for STT/TTS.

> If we're unable to get a Linux console, this may mean mean the board is using a microcontroller instead of ARM computer. Hence this wouldn't be supported here.
> 
> Unless there's another computer that can reach internally from here - like another serial port?

> https://www.gizmochina.com/2020/11/05/xiaomi-launches-a-new-iot-software-platform-xiaomi-vela-based-on-nuttx-os/

As my previous links show, wants Xaiomi using NuttX for all of his devices. I think they just started now with these speakers.
I don't know why they should run any different OS, as they new NuttX seems to do anything. 

So yes, in this state your patches can't be used anymore. You could update your README, that the new "Speaker Play" doesn't work anymore. 😃

[image]

I removed the metal housing, the rubbers and thermal paste from all chips. There are 4 "real" chips I can see:
- BES2300 (Bluetooth 5.0 Chip: https://airreps.info/files/datasheets/BES2300-IH_Datasheet_v0.25.pdf)
- ??? (? -> Next to the BES2300, I only can read "T1G" or "T1C"... I don't know what this could be)
- ESMT AD52580 (Audio Amplifier Chip: http://www.datasheet39.com/PDF/826755/AD52580-datasheet.html)
- JWH6J (? -> I only found it on AliExpress, which refers to a JW5052: https://www.joulwatt.com/en/proinfo_3407.html)

I don't which chip does the magic and is running the NuttX... 😞

And now?

When I bought the device, I was aware that I might not be able to hack it. My plan B was to remove the board and put a Pi Zero with a ReSpeaker 2-Mic in it. So just using the speaker and the housing, to build a real open source and good looking smart speaker. 😆

duhow, 2021-12-17:
Just sad news... Anyway, thanks a lot for the investigation!
I'll close this issue as there's not much to do at the moment, but feel free to continue adding any investigation details you may find, I'll keep looking for it :)

closed 2021-12-17 by duhow

danielk117, 2021-12-27:
@duhow what do you think, is the "Xiaoai Speaker Pro" still the old hackable one? https://www.aliexpress.com/item/1005002212462596.html
if yes, would you recommend it? 😄

duhow, 2021-12-27:
Yep, that one should be LX06 and with old firmware versions should be easy to get in.
If need any help with it open a new issue.

Eric9453, 2022-10-11:
> nope, but 921600 8n1 seems to be working 👍 [output_2021-12-14_09-08-28.log](https://github.com/duhow/xiaoai-patch/files/7709785/output_2021-12-14_09-08-28.log)
> 
> how to interrupt this and get a logon?

did you find teh way  to interrupt this and get a logon?

seems that you find the way to get the shell.

is there any key need to press duting startup? thank you.

danielk117, 2022-10-14:
Connect your adapter to TX0/RX0. There is no login needed. Just press Enter when all boot stuff is done and you will get a shell.

Eric9453, 2022-10-18:
> l05c,

Tnahks for your reply.  
I tried but it's still rolling and rollong. Nonstop rolling...

dissipator, 2025-04-17:
How know the username and password of the l05c?

fantomgs, 2025-05-18:
I got L05B from China this month. It has old HW but new FW (BUILD TIME: Jan 12 2024,17:20:12)
Now it ask for password at login. And it looks like the password is generated/verified using a DSA signature instead of just md5(sn+salt).
Is it possible to download FW from BES2300 chip?

[L05B_boot.log](https://github.com/user-attachments/files/20271676/L05B_boot.log)

fantomgs, 2025-05-20:
FW of version 1.62.28 https://cdn.cnbj1.fds.api.mi-img.com/xiaoqiang/rom/l05c/mico_6d9a0_1.62.28_nuttx.bin.lzma

[mico_6d9a0_1.62.28_nuttx.bin.lzma.txt](https://github.com/user-attachments/files/20331769/mico_6d9a0_1.62.28_nuttx.bin.lzma.txt)

fantomgs, 2025-12-02:
I made python utility to patch password verification (only for FW version 1.62.28, for other version it required to find offset to patch at)
https://github.com/fantomgs/bestool-py/
Attach UART dongle, run `./bestool.py patch /dev/ttyUSB0` and power on your device

Also I dumped the bootloader from address 0x0 (sems like mapped to/from 0x20000), size 0x10000, it have identification string "Sep 30 2019 18:42:32 2b0a177c1d" [l05b.bootrom.0x20000.bin.7z.txt](https://github.com/user-attachments/files/23877084/l05b.bootrom.0x20000.bin.7z.txt)

fantomgs, 2025-12-02:
Seems like Alibaba Tmall Genie TG_C5A01 smart speaker is brother of Xiaomi L05B/L05C by hardware (CPU, sound amplifier)
It uses SOC Tmall Genie+ 6000A which looks the same as BES2300

[image]
[image]

Borsh231, 2026-01-22:
> I got L05B from China this month. It has old HW but new FW (BUILD TIME: Jan 12 2024,17:20:12) Now it ask for password at login. And it looks like the password is generated/verified using a DSA signature instead of just md5(sn+salt). Is it possible to download FW from BES2300 chip?
> 
> upd. link to ~latest~ FW update https://api.miwifi.com/rs/grayupgrade/v2/l05b?s=26ec48ed85bbc1bf6937877865c1b442&channel=release&countryCode=WW&locale=en_US&time=1747557115&version=1.0.2
> 
> https://cdn.cnbj1.fds.api.mi-img.com/xiaoqiang/rom/l05c/mico_94327_1.62.26_nuttx.bin.lzma
> 
> [L05B_boot.log](https://github.com/user-attachments/files/20271676/L05B_boot.log)

Dude can you tell me how to make a more recent column using your script?

Faber-ge, 2026-07-09:
> > or explain how to flash it to the speaker?
> 
> Cut 0x30 bytes from the beginning of mico_6d9a0_1.62.28_nuttx.bin.lzma Then decompress it using 7zip or lzma -d or binwalk -e (i.e. `lzma -d -c mico_6d9a0_1.62.28_nuttx.bin.cut.lzma > mico_6d9a0_1.62.28_nuttx.bin`) And at end flash it using bestool-py via UART adapter `bestool.py burn_watch mico_6d9a0_1.62.28_nuttx.bin COM1` (replace COM1 with name of your UART adapter) If you want to flash it manually then you have to replace first 4 bytes from [FF FF FF FF] to [1C EC 57 BE] and flash it at address 0x2C080000. I have never checked that bestool's burn function works correctly so do it for your risk and make sure you have a full flash backup.

Good afternoon. I have an L05C speaker (xiaoi.wifispeaker.l05c) for the Chinese market. The firmware version shown in the Mi Home app is 1.0.7. Could you please advise me on how to read the firmware from it so that I can later flash it using your method? I'm not very strong in programming, but I can follow step-by-step instructions. Thank you.

zhizinan1997, 2026-11-16:
### Confirmed: Xiaomi XiaoAI Play Enhanced L05C (BES best2001) UART dump, authentication bypass, and persistent custom application

Hi everyone — I have now reproduced this on a **physical L05C** (not just L05B), running **NuttX MiAiSoundbox-L05C 1.68.2, built 2025-11-24**. Thanks to @fantomgs and the earlier BES work here.

**What works on the device:**

- CH340 USB-UART, **921600 8N1**; PCB **TX0 → adapter RXD, RX0 → adapter TXD, TP7 → GND**. The speaker uses its original 12 V supply; **do not connect adapter VCC**. Check UART logic voltage on your own board before connecting.
- Boot ROM synchronization is **two-stage**: first SYNC response `0x00` is *not yet ready*; send SYNC again until `0x02`, then load the **best2001-specific `programmer2001.bin` into SRAM**.
- Flash is **16 MiB**, readable at `0x28000000` via programmer `BULK_READ`. Runtime code addresses use a different mapping: `VA = 0x2C000000 + flash_offset`. Two full reads matched byte-for-byte (SHA-256 verified). Do **not** publish full dumps: the data/factory partitions include device-specific credentials.
- The factory NSH `login:` expects a **hex-encoded DSA signature**, not a plaintext password. For **this specific 1.68.2 image**, disassembly located the login call near `0x2C1153CA`. Replacing the four bytes at flash offset `0x1153CA` from `00 f0 bf f8` to `00 20 00 bf` skips that login call and sets the expected **success return 0**. The resulting UART shell prompt is **`ap>`**, not `nsh>`. **These bytes/addresses are version-specific: never use on a different dump without verifying its disassembly and byte signatures.**
- I also verified **32 KiB block-level write + readback** on this hardware. Programmer version `0x0103` requires an erase/burn block size of **`0x8000`**, not the `0x1000` assumption used by some tools. The `0x62` reply status `0x60` was normal on this unit, not an error.
- With the original NuttX and its existing Wi-Fi/audio drivers retained, a custom Thumb application is now persistent in flash at offset `0x870000` (VA `0x2C870000`), with a boot hook at `0x2C1153B4`; installed by writing only the `0x870000` application block and `0x110000` hook block, **application first and hook last**, each read back and verified. Current application size: **24,848 bytes**.
- A custom voice service is running with microphone PCM capture, WebSocket streaming, cloud ASR/LLM/TTS, playback through the original mainplayer, and an on-device local web configuration / RAM OTA interface. External speaker/microphone loopback testing passed; **this is not a from-scratch replacement for NuttX**, and IR/LED/buttons have not been fully regression-tested.

**Minimal binary patch example (only for the exact verified image):**

```python
# image must be the user's own 16 MiB, twice-verified backup
# Before writing, verify firmware version, chip, and exact original bytes.
off = 0x1153CA
assert image[off:off+4] == bytes.fromhex("00 f0 bf f8")
image[off:off+4] = bytes.fromhex("00 20 00 bf")
# This snippet only modifies an in-memory bytearray.
# Write only after preserving a recovery copy and validating the target block.
```

Relevant implementation files in my project: `scripts/xiaoai_dump.py` (read-only Boot ROM protocol, two-pass dump), `docs/FLASH_DUMP_RUNBOOK.md`, `scripts/flash_blocks.py` (two-pass pre-write check, readback and rollback), `fw/build_voice_firmware.py` (version-locked image builder / boot hook), `fw/voice_runtime.c` and `research_log/findings.md`.

Repository: https://github.com/zhizinan1997/xiaoai-l05c — **currently private**, so these file links are not yet publicly accessible; the details above are provided here directly. I have deliberately not posted factory data, the full Xiaomi firmware, API keys, or device-specific identifiers. Happy to compare reproducibility details with anyone working on L05B/L05C.