duhow/xiaoai-patch #20: Hack into my new LX06

danielk117, 2022-01-07:
Hi @duhow,

just received my LX06 and disassembled it. According to https://github.com/Jian-Xian/CVE-POC/blob/master/picture/CVE-2020-10263_1.jpg, the following three contacts are needed. 

[image]

Tx, Rx, GND (from left to right)?

duhow, 2022-01-07:
Contacts should be as in this pic: https://github.com/duhow/xiaoai-patch/blob/master/research/lx06/board.jpg

danielk117, 2022-01-07:
Ok, serial connection is working.

```
AXG:BL1:d1dbf2:a4926f;FEAT:E0DC318C:2000;POC:F;EMMC:800;NAND:0;READ:0;0.0;CHK:0;
sdio debug board detected
TE: 23660

BL2 Built : 15:46:00, Sep 29 2019. axg g8667414 - sentao.gao@droid08-bj

set vcck to 1140 mv
set vddee to 1070 mv
Board ID = 5
CPU clk: 1200MHz
DDR low power enabled
DDR DQS-calibration enabled
DDR scramble enabled
DDR3 chl: Rank0 16bit @ 792MHz - PASS
Rank0: 128MB(auto)-2T-11
DataBus test pass!
AddrBus test pass!
NAND init
chk page: 00000500
chk page: 00000540
chk page: 00000580
chk page: 000005c0
bbt blk:00000014
bbt page:00000000
0000000000000000000000000000000000000000000000000000000000000000
Load FIP HDR from NAND, src: 0x0000c000, des: 0x01700000, size: 0x00004000
Load BL3x from NAND, src: 0x00010000, des: 0x01704000, size: 0x00080000
NOTICE:  BL31: v1.3(release):87b5e26
NOTICE:  BL31: Built : 11:33:44, Apr 10 2020
NOTICE:  BL31: AXG normal boot!
NOTICE:  BL31: BL33 decompress pass
[Image: axg_v1.1.3268-b93dd79 2017-12-01 14:22:18 huan.biao@droid12]
OPS=0x43
68 77 5 fa 69 a2 2b b5 a1 c0 8a 51 bl30:axg ver: 9 mode: 0
bl30:axg thermal0
[0.014497 Inits done]
secure task start!
high task start!
low task start!
ERROR:   Error initializing runtime service opteed_fast

U-Boot 2015.01 (Jun 28 2021 - 02:53:53), Build: jenkins-Mico_lx06_ota_publish-251

DRAM:  128 MiB
Relocation Offset is: 06f17000
register usb cfg[0][1] = 0000000007f89648
NAND:  nand id: 0xec 0xf1
128MiB, SLC, page size: 2048, OOB size: 64
NAND device id: ec f1 0 95 42 c6
NAND device: Manufacturer ID: 0xec, Chip ID: 0xec (Samsung M Generation NAND 1Gib FS33ND01GS108TFI0)
oob avail size 6
Creating 1 MTD partitions on "M Generation NAND 1Gib FS33ND01GS108TFI0":
0x000000000000-0x000000200000 : "bootloader"
M Generation NAND 1Gib FS33ND01GS108TFI0 initialized ok
nand id: 0xec 0xf1
128MiB, SLC, page size: 2048, OOB size: 64
NAND device id: ec f1 0 95 42 c6
NAND device: Manufacturer ID: 0xec, Chip ID: 0xec (Samsung M Generation NAND 1Gib FS33ND01GS108TFI0)
PLANE change!
aml_nand_init :oobmul=1,oobfree.length=8,oob_size=64
oob avail size 8
bbt_start=20 env_start=24 key_start=32 dtb_start=40
nbbt: info size=0x400 max_scan_blk=24, start_blk=20
nbbt : phy_blk_addr=20, ec=0, phy_page_addr=0, timestamp=1
nbbt free list:
blockN=21, ec=-1, dirty_flag=0
blockN=22, ec=-1, dirty_flag=0
[... 159 lines cut ...]
[   13.454752@2] wlan: Driver loaded successfully
[   13.752292@0] wlan: Starting AP
[   13.753717@0] fw doesn't support 11ax
[   13.764983@0] wlan: AP started
[   13.767998@0] Set AC=3, txop=47 cwmin=3, cwmax=7 aifs=1
[   13.770465@3] Set AC=2, txop=94 cwmin=7, cwmax=15 aifs=1
[   13.775309@1] Set AC=0, txop=0 cwmin=15, cwmax=63 aifs=3
[   13.780862@0] Set AC=1, txop=0 cwmin=15, cwmax=1023 aifs=7
[   13.932336@0] HCI UART driver ver 2.2-M2614100[   13.932420@3] HCI H4 protocol initialized
[   16.409959@3] ps_init_work...
[   16.409988@3] ps_init_timer...
[   16.411017@0] ps_init...
[   20.478884@0] ps_init_work...
[   20.478915@0] ps_init_timer...
[   20.479800@3] ps_init...
[   22.269431@3] ps_init...
```

```
LX06 login: root
magic[release]: 23948/*********EFC06027
password:
```

How to get the root password?

duhow, 2022-01-08:
Looks like this speaker has a new firmware version, IIRC this `magic` string appear starting in `1.74.1` version.
Try to run U-boot by pressing any key in serial port, gather as much information as possible (maybe dump NAND via `rx`?).
You can also try to change `env` data to have Linux kernel boot directly `/bin/sh` or something like that.
Try to downgrade to this firmware update: https://bigota.miwifi.com/xiaoqiang/rom/lx06/mico_firmware_8b63c_1.58.13.bin
(and block internet whenever possible to avoid OTA)

duhow, 2022-01-08:
Check more information in Hassbian forums: https://bbs.hassbian.com/thread-8754-12-1.html

danielk117, 2022-01-09:
> and block internet whenever possible to avoid OTA

Yes. I didn't setup the speaker, so it isn't able to connect to the internet for now. 

I will try to get into uboot tomorrow. A dump of the NAND is a good idea. 

Any idea how to downgrade the speaker? I only found the following method: http://javabin.cn/2021/xiaoai_fm.html

danielk117, 2022-01-11:
I'm still not able to get into uboot... at least not the "normal" way 😉 

I looked into my previous posted log and googled some of the commands. 

"amlogic" and "InUsbBurn" seems interesting to me. 

- http://openlinux.amlogic.com:8000/download/A113/DOC/Amlogic_update_usb_tool_user_guide.pdf
- http://openlinux.amlogic.com:8000/download/A113/Tool/windows/USB_Burning_Tool_v2.1.3.zip
- http://openlinux.amlogic.com:8000/download/A113/Tool/flash-tool-v4.7/flash-tool/tools/windows/update.exe (and the 4 dll in the same folder, webserver allow DirectoryListening)

So I connected a cable to the microUSB port on that board, installed the driver and played around with the update utility. Running `update.exe identify` while starting the device:
```
> update.exe identify
AmlUsbIdentifyHost
This firmware version is 0-7-0-16-0-0-0-0
```

So I tried...
```
> update.exe bulkcmd "printenv"
AmlUsbBulkCmd[printenv]
```
While the serial connection shows me the output!!!
```
BULKcmd[printenv]
baudrate=115200
board_id=5
boot_failcnt=1
boot_failed=if itest ${boot_failcnt} == 1; then setenv boot_failcnt 2; setenv boot_part boot1; else if itest ${boot_failcnt} == 2; then setenv boot_failcnt 1; setenv boot_part boot0; else run set_boot_flag;fi;fi;
boot_part=boot0
bootargs=rootfstype=ramfs init=/init console=ttyS0,115200 no_console_suspend quiet earlycon=aml_uart,0xff803000 jtag=apao reboot_mode=cold_boot uboot=U-Boot 2015.01 (Jun 28 2021 - 02:53:53)
bootcmd=run storeboot
bootdelay=0
dtb_mem_addr=0x1000000
factory_detect=echo no need detect now.
fdt_high=0x20000000
firstboot=1
identifyWaitTime=1000
initargs=rootfstype=ramfs init=/init console=ttyS0,115200 no_console_suspend quiet earlycon=aml_uart,0xff803000
jtag=apao
loadaddr=1080000
preboot=run storeargs;if test ${reboot_mode} = cold_boot; then run try_auto_burn; fi;
product_model=lx06
reboot_mode=cold_boot
rpmb_state=0
set_boot_flag=if test ${boot_part} = boot0; then setenv boot_failcnt 1; else setenv boot_failcnt 2; fi;
silent_boot=0
stderr=serial
stdin=serial
stdout=serial
storeargs=setenv bootargs ${initargs} jtag=${jtag}; setenv bootargs ${bootargs} reboot_mode=${reboot_mode} uboot=${version};
storeboot=if test ${reboot_mode} = cold_boot; then run set_boot_flag; else run boot_failed; fi; saveenv; if imgread kernel ${boot_part} ${loadaddr}; then bootm ${loadaddr}; fi; reset;
try_auto_burn=update 500 1000;
ubootenv_version=1
upgrade_step=2
version=U-Boot 2015.01 (Jun 28 2021 - 02:53:53)

Environment size: 1536/65532 bytes
[MSG]ret = 0
[info]success
```

because some of these things are new to me, i play very cautiously. but i think i could dump AND flash everything with the update tool... or what do you think? 😆

duhow, 2022-01-11:
Interesting... You may try increasing the bootdelay? Should be something like:
```
printenv bootdelay
setenv bootdelay 3
saveenv
```

danielk117, 2022-01-11:
```
BULKcmd[printenv bootdelay]
bootdelay=0
[MSG]ret = 0
[info]success
BULKcmd[setenv bootdelay 3]
[MSG]ret = 0
[info]success
BULKcmd[saveenv]
Saving Environment to aml-storage...
uboot env amlnf_env_save : ####
aml_nand_save_rsv_info:656, nenv: valid=1, pages=32
aml_nand_save_rsv_info:716,save info to 330000
aml_nand_write_rsv:520,write info to 330000
[MSG]ret = 0
[info]success
```
Restart device and again no real countdown. Cant interupt uboot...
```
Hit Enter or space or Ctrl+C key to stop autoboot -- :  0
```
But I realize a really really small delay after this line. So I tried again and was able to stop it! 
I set bootdelay to 30 now. Now I had enough time to stop autoboot. It is counting really fast, so in my case 30 are not 30 seconds, it's only around 5 seconds... I think thats the reason why I can't access the uboot on `bootdelay=0`.

So, trying to unterstand the google-translated version of http://javabin.cn/2021/xiaoai_fm.html 😆 

What you think, should I flash an old untouched (like you posted 1.58.13) or an touched image (https://bbs.hassbian.com/forum.php?mod=redirect&goto=findpost&ptid=8754&pid=368909 1.70.4)? Old one should be fine, because I want build my own patched firmware using your scripts, right? 😉

duhow, 2022-01-11:
I'm running my software in the old firmware, then I just keep patching it.
Ensure to backup all content (dump MTD) before writing!
Also you may want to flash system1 (second partition), extract kernel from update file or just clone the same boot0 to boot1.
Then just change uboot env "boot_part".

danielk117, 2022-01-12:
So, I created dumps from all partitions.
Can i flash the *.bin directly or do I need a *.img?

danielk117, 2022-01-13:
Used binwalk before you answered and flashed it. I was able to boot. But...
```
mico login: root
Password:
Login incorrect
```
Created a password using the mi_speaker_pwd.html file, but v1 or v2 aren't working...

duhow, 2022-01-14:
Yay! Congrats!
By the way, if you can share the latest squash firmware it will be appreciated, I can try to analyze some content there to see how it works.

danielk117, 2022-01-16:
> By the way, if you can share the latest squash firmware it will be appreciated, I can try to analyze some content there to see how it works.

Here is my [mtd4](https://we.tl/t-IMtXom3jB1). It seems to be a "1.74.10".

I've build the docker (had some errors with the latest wget2, so I set version to 1.99.2) and patched the 1.74.10. It's running my speaker now.
What I have noticed until now:
- Using wget throws some errors and has problem with certificates. 
```
root@LX06:/tmp# wget https://www.dreamweaver.at/music-free/dreamweaver_-_Inspiring-Guitar-Song_01.mp3
Unknown configuration key 22 (maybe this config value should be of another type?)
Unknown configuration key 21 (maybe this config value should be of another type?)
Could not complete TLS handshake: certificate verify failed
Failed to connect: Certificate error
Failed to open tmpfile '/root/.local/share/wget/.wget-ocsp_hostsIIfcR6' (2)
Failed to write to OCSP hosts to '/root/.local/share/wget/.wget-ocsp_hosts'
Failed to open tmpfile '/root/.local/share/wget/.wget-ocspAVkEz3' (2)
Failed to write to OCSP fingerprints to '/root/.local/share/wget/.wget-ocsp'
```
- Can't discover bluetooth. Daemon is running and config seems to ok.

duhow, 2022-01-16:
For SSL, check it isn't 2021 (`date`). `ntpd -q -p pool.ntp.org`. Still need to fix this for LX06.
Regarding bluetooth, you should connect to `bluetoothctl` and ensure it is visible and discoverable. Didn't mess with it too much, but I remember having some headaches when trying to play audio... -.-

I'll try to check this firmware to see what changes.
As you shared this is 1.74.10 `28 Jun 2021 10:31:51 +0800`. The Git tag in there does not change for any firmware, so I'm skipping it. `uci -c /usr/share/mico show`

duhow, 2022-01-16:
As now you already have access to the speaker, I'll close this issue. If you face any other errors or want to improve some building stuff, feel free to open new issues or contritube back! :D

closed 2022-01-16 by duhow

danielk117, 2022-01-16:
> For SSL, check it isn't 2021 (`date`). `ntpd -q -p pool.ntp.org`. Still need to fix this for LX06. 

Good idea, but it didn't solved it. As said before, I've changed the build for wget2 from `latest` to `1.99.2` (because of an build error). Might be a reason for the error. 
Should i open a new issue for this?
```
root@LX06-9341:/tmp# date
Sun Jan 16 22:17:41 +01 2022
root@LX06-9341:/tmp# wget https://www.dreamweaver.at/music-free/dreamweaver_-_Inspiring-Guitar-Song_01.mp3
Unknown configuration key 22 (maybe this config value should be of another type?)
Unknown configuration key 21 (maybe this config value should be of another type?)
Could not complete TLS handshake: certificate verify failed
Failed to connect: Certificate error
Failed to open tmpfile '/root/.local/share/wget/.wget-ocsp_hostsAMiolM' (2)
Failed to write to OCSP hosts to '/root/.local/share/wget/.wget-ocsp_hosts'
Failed to open tmpfile '/root/.local/share/wget/.wget-ocspGOxKvo' (2)
Failed to write to OCSP fingerprints to '/root/.local/share/wget/.wget-ocsp'
```
Should i open a new issue

duhow, 2022-01-16:
Yep, please open a new issue. Also `curl` should work?

danielk117, 2022-01-16:
`curl` is working. i will create a new issue.

BTW, thanks for your help 😃 
I will write an short instruction how to hack lx06 with the amlogic update tool. I think there is no need to solder TX/RX/GND.

danielk117, 2022-01-21:
any idea where to place my instruction? i could contribute it in a makdown file to your research directory...

duhow, 2022-01-22:
Sure, do a PR in a file `research/lx06/install.md` or similar. Thanks!

Hcy1142, 2023-11-17:
i cant find the http://openlinux.amlogic.com:8000/download/A113/Tool/flash-tool-v4.7/flash-tool/tools/windows/update.exe tool, can you share this with me?

duhow, 2023-11-18:
https://github.com/osmc/aml-flash-tool

jingyibo123, 2024-01-25:
I think some recent patch on LX06 also blocked access with Amlogic tool vis USB. 

Does anyone have any idea on how to bypass that?

duhow, 2024-01-25:
@jingyibo123 please open a new issue 🙏🏻