duhow/xiaoai-patch #37: Build for package 'sndio' not completed

Eric9453, 2022-10-18:
I try to run docker, wait for a long time. Then got this error...
Seems https://github.com/ratchov/sndio/archive/refs/tags/release_1_8_1.tar.gz is unavailable any more.
Is there any alternative link? Thank you.
---
Entering build for package 'sndio' 
Entering directory /xiaoai/build-packages/src/sndio...
Downloading package source...
--2022-10-18 16:02:59--  https://github.com/ratchov/sndio/archive/refs/tags/release_1_8_1.tar.gz
Resolving github.com (github.com)... 20.27.177.113
Connecting to github.com (github.com)|20.27.177.113|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://codeload.github.com/ratchov/sndio/tar.gz/refs/tags/release_1_8_1 [following]
--2022-10-18 16:03:00--  https://codeload.github.com/ratchov/sndio/tar.gz/refs/tags/release_1_8_1
Resolving codeload.github.com (codeload.github.com)... 20.27.177.114
Connecting to codeload.github.com (codeload.github.com)|20.27.177.114|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2022-10-18 16:03:00 ERROR 404: Not Found.

Build for package 'sndio' not completed

Eric9453, 2022-10-18:
I modified the [sndio packege.mk](https://github.com/duhow/xiaoai-patch/blob/master/packages/sndio/package.mk)  as  

--  
PACKAGE_VERSION="1.8.1"  
PACKAGE_SRC="https://github.com/ratchov/sndio/archive/refs/tags/v${PACKAGE_VERSION}.tar.gz"  

It's OK.

duhow, 2022-10-18:
Feel free to do a pull request with the fix :)

PR #42 (2022-12-01): Update packages
Fixes #37
commit 22100de6 (2022-12-01): Update packages (#42)

* packages: fix package sndio and update to 1.9.0
Fixes #37

* downgrade libconfig from master to stable v1.7.3
* Update busybox patches
closed 2022-12-01 by duhow