duhow/xiaoai-patch #38: AS05G

ItsTheEdgy, 2022-10-29:
[image]
[image]
[image]
[image]

I recently bought the new xioami smart speaker with ir controll, which is a google home device, features the same clock as the L05C(?).
I couldn't make out any uart rx/tx pads. The processor is a Amlogic A113X Quad-core ARM Cortex-A53. My guess it's running some version of android and the desoldered micro-usb plug could be used to get adb running, this being only whild guesses as the echo 2nd gen is featuring a similar amlogic processor running android 5/6. Has anyone else taken a look at it yet?

ItsTheEdgy, 2022-10-29:
[image]
[image]
[image]
[image]

duhow, 2022-10-29:
Thanks for the info! Can you share the link where you purchased this one?
Can't find any UART reference at the moment.

ItsTheEdgy, 2022-10-30:
@duhow i ordered it via www.mi.com, might be a region locked product asxiaomi tends to do that, the Xiaoai Speaker Pro LX06 wasn't/isn't even available in germany.

duhow, 2022-10-30:
I will assume this is https://www.mi.com/es/product/xiaomi-smart-speaker-ir-control/specs .
This model `L05G` (Google support) might not be valid for this patch... But need to jack in first.

ItsTheEdgy, 2022-10-30:
Most likely it wont be applicable, I mainly added this here as this seems to be the most active place to talk about xiaomi smart speaker reverse engineering. If you know of someplace more fitting i will move it there.

danielk117, 2022-11-01:
> @duhow i ordered it via www.mi.com, might be a region locked product as xiaomi tends to do that, the Xiaoai Speaker Pro LX06 wasn't/isn't even available in europe.

I ordered my LX06's at AliExpress. The devices are not designed for europe. Chinese language and socket only.

ktkd, 2022-11-03:
oh i have the same speaker. I couldn't find a ttl pins on the board.
hope for your idea with micro usb.

ktkd, 2022-11-03:
`�XG:BL1:d1dbf2:a4926f;FEAT:F0DC31BC:2000;POC:F;EMMC:800;NAND:0;READ:0;0.0;0.0;CHK:0;
sdio debug board detected 
TE: 136498

BL2 Built : 10:06:34, Jan 19 2021. axg g796c27d - jenkins@walle02-sh

set vcck to 1050 mv
set vddee to 950 mv
Board ID = 0
CPU clk: 1200MHz
DDR low power enabled
DDR3 chl: Rank0 16bit @ 912MHz
bist_test rank: 0 20 00 40 2e 10 4c 1c 00 39 31 13 4f 00 00 00 00 00 00 00 00 00 00 00 00 637   - PASS
Rank0: 512MB(auto)-2T-13
AddrBus test pass!
NAND init
scan bbt ecc error happen:
storage init finish
Load FIP HDR from NAND, src: 0x0000c000, des: 0x01700000, size: 0x00004000, part: 0
Load BL3x from NAND, src: 0x00010000, des: 0x01704000, size: 0x000b2800, part: 0
NOTICE:  BL31: v1.3(release):d3a620ec3
NOTICE:  BL31: Built : 10:32:40, Jan 20 2021
NOTICE:  BL31: AXG secure boot!
NOTICE:  BL31: BL33 decompress pass
OPS=0x43
[Image: axg_v1.1.3516-f91e350f3 2020-12-11 20:41:05 jenkins@walle02-sh]
25 0c 43 00 c8 cc f9 9b 0f 3a e1 55 9b 2c 97 9e 
bl30:axg ver: 9 mode: 0
bl30:axg thermal0
[0.016061 Inits done]
secure task start!
high task start!
low task start!
ERROR:   Error initializing runtime service opteed_fast

U-Boot 2015.01-g35e62defed-dirty (Nov 18 2021 - 15:48:24)

DRAM:  512 MiB
Relocation Offset is: 1eebd000
mmu cfg end: 0x20000000
mmu cfg end: 0x20000000
register usb cfg[0][1] = 000000001ff72df0
aml_i2c_init_port init regs for 0
NAND:  get_sys_clk_rate_mtd() 281, clock setting 200!
bus cycle0: 6,timing: 7
NAND device id: 98 da 90 15 76 16 
NAND device: Manufacturer ID: 0x98, Chip ID: 0x98 (Toshiba A revision NAND 2Gib TC58NVG1S3HBAI4 )
get_sys_clk_rate_mtd() 281, clock setting 200!
m3_nand_adjust_timing() sys_clk_rate 200, bus_c 6, bus_t 7
oob_fill_cnt =32 oob_size =64, bch_bytes =14
ecc mode:6 ecc_page_num=2 eep_need_oobsize=16
plane_num=1 writesize=0x800 ecc.size=0x200 bch_mode=1
oob avail size 6
Creating 1 MTD partitions on "A revision NAND 2Gib TC58NVG1S3HBAI4 ":
0x000000000000-0x000000200000 : "bootloader"
A revision NAND 2Gib TC58NVG1S3HBAI4  initialized ok
get_sys_clk_rate_mtd() 281, clock setting 200!
bus cycle0: 6,timing: 7
NAND device id: 98 da 90 15 76 16 
NAND device: Manufacturer ID: 0x98, Chip ID: 0x98 (Toshiba A revision NAND 2Gib TC58NVG1S3HBAI4 )
get_sys_clk_rate_mtd() 281, clock setting 200!
m3_nand_adjust_timing() sys_clk_rate 200, bus_c 6, bus_t 7
oob_fill_cnt =32 oob_size =64, bch_bytes =14
ecc mode:6 ecc_page_num=2 eep_need_oobsize=16
PLANE change!
plane_num=1 writesize=0x800 ecc.size=0x200 bch_mode=1
aml_nand_init :oobmul=1,oobfree.length=8,oob_size=64
oob avail size 8
nbbt=20
nenv=24
[... 513 lines cut ...]
[   10.179404@3]- enter vm_cfg80211_get_channel, no support yet
[   10.179453@3]- vm_cfg80211_get_tx_power 2243
[   10.183729@3]- ---drv_curtxpower:0
[   10.187540@3]- enter vm_cfg80211_get_channel, no support yet
[   10.192684@3]- vm_cfg80211_get_tx_power 2243
[   10.196907@3]- ---drv_curtxpower:0
[   10.200811@3]- enter vm_cfg80211_get_channel, no support yet
[   10.205868@3]- vm_cfg80211_get_tx_power 2243
[   10.210091@3]- ---drv_curtxpower:0
[   10.252511@3]- aml_cfg80211_flush_pmksa
[   10.273428@3]- <wlan0>:vm_cfg80211_scan vm_opmode:1, vm_state:1, connect_status:0, nssids:2
[   10.276942@3]- <wlan0> vm_cfg80211_scan ssid=Bee, len=3
[   10.281407@3]- <wlan0> vm_cfg80211_scan ssid=, len=0
[   10.286321@3]- vm_cfg80211_scan lnssids 2, lssidlen 3, n_channels:51
[   10.292614@3]- wifi_mac_start_scan vm_scanchan_rssi:0 
[   10.297653@3]- wifi_mac_get_channel_rssi_before_scan last rssi:0
[   10.303629@3]- wifi_mac_get_channel_rssi_before_scan current rssi:-65
[   10.309999@3]- wifi_mac_set_channel_rssi channel rssi:191
[   10.315385@3]- <running> wifi_mac_scan_flush 441 
[   10.315609@0]- phy_set_channel_rssi channel rssi:191
[   10.324898@3]- wifi_mac_start_scan vid:0---> scan start, CfgFlags is:0000008f, ss->ss_nssid:2
[   10.333413@0]- scan_start_task wm_nrunning:0
[   10.351965@0]s FUNCTION: wifi_mac_hardstart LINE: 409:WARNNING: no connection vif_id 0
[   10.790776@1]- configfs-gadget ff400000.dwc2_a: failed to start amlogic: -19
`

duhow, 2022-11-03:
Great finding @ktkd !
Can you detail which ports are?
TP65 - ?
TP63 - ?
TP?? - GND

Also, are you able to stop uboot and edit run args?
I haven't tried with USB debugging via fastboot or adb yet, so let me know if you get a way into a shell.

ktkd, 2022-11-03:
TP63 - RX
TP65 - TX
GND from any GND point on board ( microusb gnd is ok )

u-boot can be stopped by ctrl+c  `Hit Enter or space or Ctrl+C key to stop autoboot -- :   `

ItsTheEdgy, 2022-11-03:
Well, I soldered the usb port, but it seems usb debugging is off by default. Therfore that's a dead end.

duhow, 2022-12-01:
@ktkd @ItsTheEdgy can you list Uboot commands and `printenv` available?
Also I'm interested if it's possible to dump the partitions (specially `system`), try the following:
https://cybergibbons.com/hardware-hacking/recovering-firmware-through-u-boot/
https://github.com/gmbnomis/uboot-mdb-dump

ItsTheEdgy, 2022-12-01:
@duhow 

```
?       - alias for 'help'
aml_bcb - aml_bcb
aml_sysrecovery- Burning with amlogic format package from partition sysrecovery
amlmmc  - AMLMMC sub system
amlnf   - aml mtd nand sub-system
autoscr - run script from memory
avb     - avb
base    - print or set address offset
bcb     - bcb
boot_cooling- cpu temp-system
booti   - boot arm64 Linux Image image from memory
bootm   - boot application image from memory
chpart  - change active partition
clkmsr  - Amlogic measure clock
cmp     - memory compare
cp      - memory copy
crc32   - checksum calculation
dcache  - enable or disable data cache
defenv_reserv- reserve some specified envs after defaulting env
echo    - echo args to console
efuse   - efuse commands
efuse_user- efuse user space read write ops
emmc    - EMMC sub system
env     - environment handling commands
exit    - exit script
false   - do nothing, unsuccessfully
fastboot- use USB Fastboot protocol
fatinfo - print information about filesystem
fatload - load binary file from a dos filesystem
fatls   - list files in a directory (default /)
fatsize - determine a file's size
fdt     - flattened device tree utility commands
forceupdate- forceupdate
get_avb_mode- get_avb_mode
get_bootloaderversion- print bootloader version
get_rebootmode- get reboot mode
get_slot_state- get_slot_state a suc_stete
get_system_as_root_mode- get_system_as_root_mode
get_valid_slot- get_valid_slot
gpio    - query and control gpio pins
help    - print command description/usage
i2c     - I2C sub-system
icache  - enable or disable instruction cache
imgread - Read the image from internal flash with actual size
itest   - return true/false on integer compare
jtagoff - disable jtag
jtagon  - enable jtag
keyman  - Unify key ops interfaces based dts cfg
keyunify- key unify sub-system
loop    - infinite loop on address range
md      - memory display
mm      - memory modify (auto-incrementing address)
mmc     - MMC sub system
mmcinfo - display MMC info
mtdparts- define flash/nand partitions
mw      - memory write (fill)
mwm     - mw mask function
nand    - NAND sub-system
nboot   - boot from NAND device
nm      - memory modify (constant address)
open_scp_log- print SCP messgage
printenv- print environment variables
query   - SoC query commands
readMetadata- readMetadata
read_temp- cpu temp-system
reboot  - set reboot mode and reboot system
reset   - Perform RESET of the CPU
ringmsr - Amlogic measure ring
rpmb_state- RPMB sub-system
rsvmem  - reserve memory
run     - run commands in an environment variable
saradc  - saradc sub-system
saradc_12bit- saradc sub-system
saveenv - save environment variables to persistent storage
sdc_burn- Burning with amlogic format package in sdmmc
sdc_update- Burning a partition with image file in sdmmc card
set_active_slot- set_active_slot
set_trim_base- cpu temp-system
set_usb_boot- set usb boot mode
setenv  - set environment variables
showvar - print local hushshell variables
sleep   - delay execution for some time
store   - STORE sub-system
systemoff- system off
tee_log_level- update tee log level
temp_triming- cpu temp-system
test    - minimal test like /bin/sh
true    - do nothing, successfully
ubi     - ubi commands
ubifsload- load file from an UBIFS filesystem
ubifsls - list files in a directory
ubifsmount- mount UBIFS volume
ubifsumount- unmount UBIFS volume
unpackimg- un pack logo image into pictures
update  - Enter v2 usbburning mode
usb     - USB sub-system
usb_burn- Burning with amlogic format package in usb
usb_update- Burning a partition with image file in usb host
usbboot - boot from USB device
version - print monitor, compiler and linker version
vpp     - vpp sub-system
watchdog- enable or disable watchdog
write_trim- cpu temp-system
write_version- cpu temp-system
```
here's the whole command list

I'm going to try dumping the firmware fith md later

pabsi, 2024-01-16:
Hello everyone,
I have an L05G I am willing to open up and connect to via UART for testing purposes.
Is there anything I can help with to get this project going on this device?
Thank you in advance :)

ItsTheEdgy, 2024-01-16:
@pabsi, hey iirc I tested around some more and it seemed like the usb is active. However sadly i broke my device when testing some things so I didn't rly test around a lot more. Getting a micro/mini usb soldered on there and trying adb/fastboot is your best bet imo.