duhow/xiaoai-patch #45: Docker container exits with error "Build for package 'glibc' not completed"

hdh75, 2023-02-10:
Thanks a lot David for this awesome project! 

I am trying to create an image now for my L09A and am runnining in to the following issue. I have downloaded the git structure to my windows computer and created a docker image and a container with Docker Desktop. When I try to run the container it exits after about 20 mins with the error in the issue title. I have included an excerpt from the logs that might help narrowing down the issue:

`2023-02-10 15:54:12 
2023-02-10 15:54:12 Host arch         :  x86_64
2023-02-10 15:54:12 Target arch       :  armv7
2023-02-10 15:54:12 Model detected    :  
2023-02-10 15:54:12 Prefix            :  /usr
2023-02-10 15:54:12 Make concurrency  :  8
2023-02-10 15:54:12 
2023-02-10 15:54:12 Note: Prefix will be the path to all binaries in target device 
2023-02-10 15:54:12 
2023-02-10 15:54:12 tput: No value for $TERM and no -T specified
2023-02-10 15:54:12 tput: No value for $TERM and no -T specified
2023-02-10 15:54:12 Try 'timeout --help' for more information.
2023-02-10 15:54:13 Entering build for package 'toolchain' 
2023-02-10 15:54:13 Entering directory /xiaoai/build-packages/src/toolchain...
2023-02-10 15:54:13 Entering directory /xiaoai/build-packages/build/armv7/toolchain...
2023-02-10 15:54:14 Generated: /xiaoai/build-packages/build/armv7/toolchain/config/cmake-toolchain.txt
2023-02-10 15:54:14 Generated: /xiaoai/build-packages/build/armv7/toolchain/config/cmake-toolchain-sysroot.txt
2023-02-10 15:54:14 Generated: /xiaoai/build-packages/build/armv7/toolchain/config/meson-cross-file.txt
2023-02-10 15:54:14 tput: No value for $TERM and no -T specified
2023-02-10 15:54:14 
2023-02-10 15:54:14 TOOLCHAIN_DIR="/xiaoai/build-packages/build/armv7/toolchain"
2023-02-10 15:54:14 BUILD_TARGET="arm-linux-gnueabihf"
2023-02-10 15:54:14 BUILD_CC="/xiaoai/build-packages/build/armv7/toolchain/bin/arm-linux-gnueabihf-gcc"
2023-02-10 15:54:14 BUILD_CXX="/xiaoai/build-packages/build/armv7/toolchain/bin/arm-linux-gnueabihf-g++"
2023-02-10 15:54:14 BUILD_AR="/xiaoai/build-packages/build/armv7/toolchain/bin/arm-linux-gnueabihf-ar"
2023-02-10 15:54:14 BUILD_RANLIB="/xiaoai/build-packages/build/armv7/toolchain/bin/arm-linux-gnueabihf-ranlib"
2023-02-10 15:54:14 BUILD_OBJCOPY="/xiaoai/build-packages/build/armv7/toolchain/bin/arm-linux-gnueabihf-objcopy"
2023-02-10 15:54:14 BUILD_STRIP="/xiaoai/build-packages/build/armv7/toolchain/bin/arm-linux-gnueabihf-strip"
2023-02-10 15:54:14 BUILD_CFLAGS="-I/xiaoai/build-packages/staging/armv7//usr/include -Os"
2023-02-10 15:54:14 BUILD_LDFLAGS="-L/xiaoai/build-packages/staging/armv7/lib -L/xiaoai/build-packages/staging/armv7//usr/lib -Wl,--rpath-link=/xiaoai/build-packages/staging/armv7//usr/lib -Wl,--rpath-link=/xiaoai/build-packages/staging/armv7//usr/lib/pulseaudio -Wl,--rpath-link=/xiaoai/build-packages/staging/armv7//usr/lib/private -Wl,--rpath=/usr/arm-linux-gnueabihf/lib -Wl,--rpath=/usr/lib -Wl,--dynamic-linker=/lib/ld-linux-armhf.so.3 -Os"
2023-02-10 15:54:14 BUILD_PKG_CONFIG_LIBDIR="/xiaoai/build-packages/staging/armv7//usr/lib/pkgconfig"
2023-02-10 15:54:14 BUILD_PKG_CONFIG_SYSROOT_DIR="/xiaoai/build-packages/staging/armv7"
2023-02-10 15:54:14 PATH="/xiaoai/build-packages/build/armv7/toolchain/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin"
2023-02-10 15:54:14 TOOLCHAIN_CMAKE="/xiaoai/build-packages/build/armv7/toolchain/config/cmake-toolchain.txt"
2023-02-10 15:54:14 
2023-02-10 15:54:14 tput: No value for $TERM and no -T specified
2023-02-10 15:54:14 Leaving build for package 'toolchain' 
2023-02-10 15:54:14 Entering build for package 'glibc' 
2023-02-10 15:54:14 Entering directory /xiaoai/build-packages/src/glibc...
2023-02-10 15:55:18 Using cached download
2023-02-10 15:55:18 Unpacking /xiaoai/build-packages/src/glibc/glibc-2.27.tar.gz...
2023-02-10 15:56:58 Entering directory /xiaoai/build-packages/build/armv7/glibc-build...
2023-02-10 15:56:58 Entering configure_package... 
2023-02-10 15:56:59 checking build system type... x86_64-pc-linux-gnu
2023-02-10 15:56:59 checking host system type... arm-unknown-linux-gnueabihf
2023-02-10 15:56:59 checking for arm-linux-gnueabihf-gcc... /xiaoai/build-packages/build/armv7/toolchain/bin/arm-linux-gnueabihf-gcc
2023-02-10 15:57:06 checking for suffix of object files... o
2023-02-10 15:57:07 checking whether we are using the GNU C compiler... yes
2023-02-10 15:57:08 checking whether /xiaoai/build-packages/build/armv7/toolchain/bin/arm-linux-gnueabihf-gcc accepts -g... yes
2023-02-10 15:57:08 checking for gcc... gcc
2023-02-10 15:57:08 checking for arm-linux-gnueabihf-readelf... arm-linux-gnueabihf-readelf
2023-02-10 15:57:08 checking for arm-linux-gnueabihf-g++... arm-linux-gnueabihf-g++
2023-02-10 15:57:12 checking whether we are using the GNU C++ compiler... yes
2023-02-10 15:57:12 checking whether arm-linux-gnueabihf-g++ accepts -g... yes
2023-02-10 15:57:38 checking whether arm-linux-gnueabihf-g++ can link programs... yes
2023-02-10 15:57:38 checking for sysdeps preconfigure fragments... aarch64 alpha arm configure: Found compiler is configured for armv7
2023-02-10 15:57:38 hppa i386 m68k microblaze mips nios2 powerpc riscv s390 sh sparc tile x86_64 
2023-02-10 15:57:39 checking for use of fpu sysdeps directories... yes
2023-02-10 15:57:39 checking for -fstack-protector... yes
2023-02-10 15:57:40 checking for -fstack-protector-strong... yes
2023-02-10 15:57:40 checking for -fstack-protector-all... yes
2023-02-10 15:57:41 checking for assembler and linker STT_GNU_IFUNC support... yes
2023-02-10 15:57:42 checking for gcc attribute ifunc support... yes
2023-02-10 15:57:43 checking if compiler warns about alias for function with incompatible types... no
2023-02-10 15:57:44 checking sysdep dirs... sysdeps/unix/sysv/linux/arm sysdeps/arm/nptl sysdeps/unix/sysv/linux sysdeps/nptl sysdeps/pthread sysdeps/gnu sysdeps/unix/inet sysdeps/unix/sysv sysdeps/unix/arm sysdeps/unix sysdeps/posix sysdeps/arm/armv7/multiarch sysdeps/arm/armv7 sysdeps/arm/armv6t2 sysdeps/arm/armv6 sysdeps/arm sysdeps/wordsize-32 sysdeps/ieee754/flt-32 sysdeps/ieee754/dbl-64 sysdeps/ieee754 sysdeps/generic
2023-02-10 15:57:44 checking for a BSD-compatible install... /usr/bin/install -c
[... 86 lines cut ...]
2023-02-10 15:58:45 mv -f /xiaoai/build-packages/build/armv7/glibc-build/sysd-rulesT /xiaoai/build-packages/build/armv7/glibc-build/sysd-rules
2023-02-10 15:58:45 gawk -f scripts/gen-sorted.awk \
2023-02-10 15:58:45        -v subdirs='csu assert ctype locale intl catgets math setjmp signal stdlib stdio-common libio malloc string wcsmbs time dirent grp pwd posix io termios resource misc socket sysvipc gmon gnulib iconv iconvdata wctype manual shadow gshadow po argp crypt localedata timezone rt conform debug mathvec support dlfcn elf' \
2023-02-10 15:58:45        -v srcpfx='' \
2023-02-10 15:58:45        sysdeps/nptl/Subdirs sysdeps/unix/inet/Subdirs sysdeps/posix/Subdirs assert/Depend catgets/Depend debug/Depend iconvdata/Depend intl/Depend libio/Depend localedata/Depend malloc/Depend mathvec/Depend posix/Depend rt/Depend stdio-common/Depend stdlib/Depend string/Depend time/Depend wcsmbs/Depend > /xiaoai/build-packages/build/armv7/glibc-build/sysd-sorted-tmp
2023-02-10 15:58:45 mv -f /xiaoai/build-packages/build/armv7/glibc-build/sysd-sorted-tmp /xiaoai/build-packages/build/armv7/glibc-build/sysd-sorted
2023-02-10 15:58:46 { echo '#include "posix/bits/posix1_lim.h"';\
2023-02-10 15:58:46 } |\
2023-02-10 15:58:46 /xiaoai/build-packages/build/armv7/toolchain/bin/arm-linux-gnueabihf-gcc -E -dM -MD -MP -MF /xiaoai/build-packages/build/armv7/glibc-build/bits/stdio_lim.dT -MT '/xiaoai/build-packages/build/armv7/glibc-build/bits/stdio_lim.h /xiaoai/build-packages/build/armv7/glibc-build/bits/stdio_lim.d' \

[...]

2023-02-10 16:18:02 /xiaoai/build-packages/build/armv7/toolchain/bin/arm-linux-gnueabihf-gcc sdlopen.c -c -std=gnu11 -fgnu89-inline  -Os -Wall -Wundef -Wwrite-strings -fmerge-all-constants -fno-stack-protector -frounding-math -Wstrict-prototypes -Wold-style-definition      -ftls-model=initial-exec      -I../include -I/xiaoai/build-packages/build/armv7/glibc-build/dlfcn  -I/xiaoai/build-packages/build/armv7/glibc-build  -I../sysdeps/unix/sysv/linux/arm  -I../sysdeps/arm/nptl  -I../sysdeps/unix/sysv/linux/include -I../sysdeps/unix/sysv/linux  -I../sysdeps/nptl  -I../sysdeps/pthread  -I../sysdeps/gnu  -I../sysdeps/unix/inet  -I../sysdeps/unix/sysv  -I../sysdeps/unix/arm  -I../sysdeps/unix  -I../sysdeps/posix  -I../sysdeps/arm/armv7/multiarch   [... cut ...]
2023-02-10 16:18:02 make[2]: *** Waiting for unfinished jobs....
2023-02-10 16:18:06 make[2]: Leaving directory '/xiaoai/build-packages/build/armv7/glibc/dlfcn'
2023-02-10 16:18:06 make[1]: *** [Makefile:215: dlfcn/subdir_lib] Error 2
2023-02-10 16:18:06 make[1]: Leaving directory '/xiaoai/build-packages/build/armv7/glibc'
2023-02-10 16:18:06 make: *** [Makefile:9: all] Error 2
2023-02-10 16:18:06 make_package returned error
2023-02-10 16:18:06 Build for package 'glibc' not completed 
`

Any help is greatly appreciated 🙏

hdh75, 2023-02-12:
I figured it out. The problem was that glibc requires a case-sensitive filesystem. In WSL that needs to be enabled for the folder which is mounted to the Docker container. Here's how:

1. run Windows PowerShell as Administrator
2. create a new folder (it needs to be empty)
3. run `fsutil.exe file setCaseSensitiveInfo <path to folder> enable`
4. copy the files from the git repo into the folder

However, Ithink in the instruction it was written it could take up to one hour to build. With Docker Desktop on Windows it took around 9h to compile the libraries, so I would not recommend going this route.

duhow, 2023-02-12:
You can check that Action Build took a bit more than 1h:
https://github.com/duhow/xiaoai-patch/actions/runs/4146380579

I'll add a note regarding WSL, although the notes already state "requires Linux" (just to avoid this kind of issues...)

hdh75, 2023-02-12:
You can make that note stating that WSL will not work as long as https://github.com/libffi/libffi/issues/552 is not resolved.

hdh75, 2023-02-12:
Now I was tying to compile it on a Raspberry Pi 3 with Raspberry Pi OS 11 "Bullseye". It is throwing the following error:

------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Host arch         :  armv7l
Target arch       :  armv7
Model detected    :  
Prefix            :  /usr
Make concurrency  :  4
[1;33m
Note: Prefix will be the path to all binaries in target device [0m
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Process will begin shortly. Press enter or break run (Ctrl+C). 
Try 'timeout --help' for more information.
[1;32mEntering build for package 'toolchain' [0m
Entering directory /xiaoai/build-packages/src/toolchain...
Entering directory /xiaoai/build-packages/build/armv7/toolchain...
Generated: /xiaoai/build-packages/build/armv7/toolchain/config/cmake-toolchain.txt
Generated: /xiaoai/build-packages/build/armv7/toolchain/config/cmake-toolchain-sysroot.txt
Generated: /xiaoai/build-packages/build/armv7/toolchain/config/meson-cross-file.txt
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
TOOLCHAIN_DIR="/xiaoai/build-packages/build/armv7/toolchain"
BUILD_TARGET="arm-linux-gnueabihf"
BUILD_CC="/xiaoai/build-packages/build/armv7/toolchain/bin/arm-linux-gnueabihf-gcc"
BUILD_CXX="/xiaoai/build-packages/build/armv7/toolchain/bin/arm-linux-gnueabihf-g++"
BUILD_AR="/xiaoai/build-packages/build/armv7/toolchain/bin/arm-linux-gnueabihf-ar"
BUILD_RANLIB="/xiaoai/build-packages/build/armv7/toolchain/bin/arm-linux-gnueabihf-ranlib"
BUILD_OBJCOPY="/xiaoai/build-packages/build/armv7/toolchain/bin/arm-linux-gnueabihf-objcopy"
BUILD_STRIP="/xiaoai/build-packages/build/armv7/toolchain/bin/arm-linux-gnueabihf-strip"
BUILD_CFLAGS="-I/xiaoai/build-packages/staging/armv7//usr/include -Os"
BUILD_LDFLAGS="-L/xiaoai/build-packages/staging/armv7/lib -L/xiaoai/build-packages/staging/armv7//usr/lib -Wl,--rpath-link=/xiaoai/build-packages/staging/armv7//usr/lib -Wl,--rpath-link=/xiaoai/build-packages/staging/armv7//usr/lib/pulseaudio -Wl,--rpath-link=/xiaoai/build-packages/staging/armv7//usr/lib/private -Wl,--rpath=/usr/arm-linux-gnueabihf/lib -Wl,--rpath=/usr/lib -Wl,--dynamic-linker=/lib/ld-linux-armhf.so.3 -Os"
BUILD_PKG_CONFIG_LIBDIR="/xiaoai/build-packages/staging/armv7//usr/lib/pkgconfig"
BUILD_PKG_CONFIG_SYSROOT_DIR="/xiaoai/build-packages/staging/armv7"
PATH="/xiaoai/build-packages/build/armv7/toolchain/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin"
TOOLCHAIN_CMAKE="/xiaoai/build-packages/build/armv7/toolchain/config/cmake-toolchain.txt"
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
[1;34mLeaving build for package 'toolchain' [0m
[1;32mEntering build for package 'glibc' [0m
Entering directory /xiaoai/build-packages/src/glibc...
Using cached download
Unpacking /xiaoai/build-packages/src/glibc/glibc-2.27.tar.gz...
Entering directory /xiaoai/build-packages/build/armv7/glibc-build...
[36mEntering configure_package... [0m
checking build system type... arm-unknown-linux-gnueabihf
checking host system type... arm-unknown-linux-gnueabihf
checking for arm-linux-gnueabihf-gcc... /xiaoai/build-packages/build/armv7/toolchain/bin/arm-linux-gnueabihf-gcc
checking for suffix of object files... configure: error: in `/xiaoai/build-packages/build/armv7/glibc-build':
configure: error: cannot compute suffix of object files: cannot compile
See `config.log' for more details
configure_package returned error
[1;31mBuild for package 'glibc' not completed [0m

Any idea what could be wrong this time? 😕

duhow, 2023-02-12:
Cannot use Raspberry to build. Requires `amd64` host.

closed 2023-02-12 by duhow