duhow/xiaoai-patch #49: L05G

mrmariosgithub, 2023-04-28:
Hi,

I recently got my hands  on a L05G and started poking around using the informations available in issue #48 and the [research](https://github.com/duhow/xiaoai-patch/tree/master/research/l05g) folder.

I managed to connect an USB stick to the board by soldering to the test pins on the underside of the board. Running `usb start` recognises the connected usb stick:
```
axg_s420_v1_gva#usb start
(Re)start USB...
USB0:   USB3.0 XHCI init start
Register 1000140 NbrPorts 1
Starting the controller
USB XHCI 1.00
scanning bus 0 for devices... 2 USB Device(s) found
       scanning usb for storage devices... init_part() 282: PART_TYPE_DOS
1 Storage Device(s) found
```

I tried both `fatinfo` and `fatls` to confirm, that the USB stick can be accessed correcty:
```
axg_s420_v1_gva#fatinfo usb 0
Interface:  USB
  Device 0: Vendor: VendorCo Rev: 2.00 Prod: ProductCode
            Type: Removable Hard Disk
            Capacity: 3840.0 MB = 3.7 GB (7864320 x 512)
Filesystem: FAT32 "NO NAME    "
axg_s420_v1_gva#
axg_s420_v1_gva#
axg_s420_v1_gva#
axg_s420_v1_gva#fatls usb 0
  5570560   DSC02072.JPG
            System Volume Information/

1 file(s), 1 dir(s)
```

But now I have no idea how to continue. Can I dump the flash to the USB stick? Or can I flash a modified firmware from the USB stick?

KInd regards
Mario

mrmariosgithub, 2023-04-28:
Can we somehow unpack the update file mentioned here? https://github.com/duhow/xiaoai-patch/issues/1#issuecomment-1371028610

mrmariosgithub, 2023-04-28:
I didn't manage to get a flash backup. I tried `nand read.raw` followed by `md.b` afterwards, but I got some strange results. I copied two flash pages (zero and one) to RAM and printed them using `md.b`. Then, to verfiy that successive read of flash pages work, I read only flash page one and printed it again. But the second read did not print the expected results, there was a gap of 0x40 bytes that were missing. In case someone wants to try it, here are the commands I used:

```
axg_s420_v1_gva#nand read.raw 0x02000000 0 0x1000
axg_s420_v1_gva#md.b 0x02000000 0x1000

axg_s420_v1_gva#nand read.raw 0x02000000 0x800 0x800
axg_s420_v1_gva#md.b 0x02000000 0x800
```

Maybe I am using the wrong aproach to dump the flash, then I would appreciate a hint how to do it correctly.

Anyways, I managed to unpack the update file and extract the boot.img file. I copied it to the flash drive and was able to boot from the flash drive using the following commands:
```
usb start
fatload usb 0 ${loadaddr} boot.img
bootm ${loadaddr}
```

Then I tried to boot a modified boot.img (with the boot commandline embedded into the boot.img patched) but that failed because of the signature check:
```
aml log : Sig Check 1423
```

Do you have any ideas what to try next?

mrmariosgithub, 2023-04-30:
I managed to dump the flash by dumping all nand pages and writing the result to a file. This was an incredible slow process which - fully automated - took around 23 hours.

I verified the dump by comparing the `binwalk` with the output of `mtdparts`:
```
DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
1157          0x485           MySQL ISAM compressed data file Version 4
16777216      0x1000000       Android bootimg, kernel size: 6453248 bytes, kernel addr: 0x1080000, ramdisk size: 3143680 bytes, ramdisk addr: 0x1000000, product name: ""
27262976      0x1A00000       Android bootimg, kernel size: 6453248 bytes, kernel addr: 0x1080000, ramdisk size: 3602432 bytes, ramdisk addr: 0x1000000, product name: ""
39845888      0x2600000       Squashfs filesystem, little endian, version 4.0, compression:gzip, size: 11356998 bytes, 1086 inodes, blocksize: 131072 bytes, created: 1970-01-01 00:00:00
56623104      0x3600000       Squashfs filesystem, little endian, version 4.0, compression:gzip, size: 51200242 bytes, 175 inodes, blocksize: 131072 bytes, created: 1970-01-01 00:00:00
107954016     0x66F3F60       Zlib compressed data, best compression
128057344     0x7A20000       UBI erase count header, version: 1, EC: 0x1, VID header offset: 0x800, data offset: 0x1000
```

```
axg_s420_v1_gva#mtdparts
0x000000000000-0x000000200000 : "bootloader"
0x000000800000-0x000001000000 : "tpl"
0x000001000000-0x000001a00000 : "recovery"
0x000001a00000-0x000002600000 : "boot"
0x000002600000-0x000003600000 : "system"
0x000003600000-0x000007a20000 : "chrome"
0x000007a20000-0x000007e20000 : "factory"
0x000007e20000-0x000010000000 : "data"
```

I extracted the data partition from the dump file using `dd`:
```
dd if=dev_1.bin bs=1k skip=129152 count=132992 of=data.img
```

The extracted file seems to be a valid UBI image:
```
data.img: UBI image, version 1
```

As I can access an USB stick from u-boot, I should be able to update the nand with modified data.

For now I am out of ideas. Does anybody else know how to continue from here?

mrmariosgithub, 2023-05-01:
Ping @ds2k5

duhow, 2023-05-01:
The `boot.img` (kernel) shouldn't be needed to change.
Ideally you'll need to:
- update the `system` partition and change the init running programs: disable **google** assistant, run **sshd** ?, update `/etc/inittab` to run a shell, update **root** password... 
- check `Makefile` for the commands to generate an `squashfs` file and ensure to match the same compression settings (compare with original with `file` command)
- flash the new `system` partition at the specific position
- additionally, update the **uboot** `bootarg` (kernel boot options) and run a program after kernel load, but maybe this shouldn't be required.

What I'm concerned is whether `uboot` or the `kernel` (`boot.img`) will verify the `system` squashfs partition with a certificate / signature or not... That means the whole system might be signed and may need another way of running this.

ds2k5, 2023-05-02:
@mrmariosgithu

Hi,

nice thanks for shareing

did you have a USB Port soldered ? could you access with adb ?

I am seaching for a friend that have a smd soldering station to
try to use adb for more possibilities

mrmariosgithub, 2023-05-03:
**Before I post my updates, let my answer your questions first:**
> additionally, update the uboot bootarg (kernel boot options) and run a program after kernel load, but maybe this shouldn't be required.

I did not manage to start any processes using kernel boot parameters, I tried differend commandlines to spawn a shell on the serial console but without success.

> how did you the dump?

I wrote a small programm that repetitively executed the `nand dump` command on the u-boot shell. This way I dumped each flash page and wrote the results to a dump file. This is an increadible slow process as each flash page takes around one second to dump and there are 131072 of them. In the end it took around 23h to dump the entire flash chip that way. The `binwalk` results of that files are posted above. I tried `nand read` and `nand read.raw` first but the results did not match the page dump results.

> could you access with adb ?

No, unfortunatelly not. As it later turned out, the ADB service is disabled and cannot be activated using kernel boot parameters. And I found no other command to enable/start ADB from u-boot, which made sense to me.

**Here are my new findings:**
TL;DR;
Good news: I got SSH working
Bad news: I ripped off the USB pads from the PCB
Good news: I made the USB stick work again

Long story:
I managed to solder on an micro USB socket. Unfortunatelly, my soldering iron was not beefy enough to heat through the THT holes which use used to mount the micro USB socket to the board as the holes are connected to massive ground planes. So I simply glued the socket in place. Using the `fastboot` command from `u-boot`, I got a new USB device showing up on the hub and responding to the `fastboot` tool provided by the Android platform-tools. I tried flashing the `system.img` file from the mico update file but that did not went well, the device failed to boot afterwards as it could not find the binaries on the system partition. While moving the board around on my desc, I (obviously) applied to much force to the micro USB socket and ripped it off. Unfortunatelly, the solder pads were gone, too. And the test pads on the bottom of the board - where I soldered the USB A socket to - did not work anymore. Luckily I was able to solder jumper wires to the USB ESD protection device pads and got the USB A socket working again.

Now that I got the USB stick working again, I started messing aroung with the NAND flash partitions. Turns out, that replacing the `boot.img` does not work because it is signed and kernel, ramdisk and second stage are encrypted. The encryption is not a problem - at least compared to the signature check. So replacing the kernel or ramdisk is a dead end (for now). However, I was able to restore the system partition using the extracted mico update file and the USB stick. Next, I tried modifying the `system.img` and it turned out that it's not signed so I was able to flash a modified system partition.

I noticed, that during boot the script `set_usb_serialnumber.sh` is executed which is part of the `system.img`. So I used that file as an entry point to the device started exploring by adding commands to the end of the file. Here are the most interesting results I got that way:

<details>
<summary>mount</summary>
rootfs / rootfs rw 0 0
tmpfs /dev tmpfs rw,relatime,size=4096k,mode=755 0 0
devpts /dev/pts devpts rw,relatime,mode=600,ptmxmode=000 0 0
proc /proc proc rw,relatime 0 0
sysfs /sys sysfs rw,relatime 0 0
tmpfs /tmp tmpfs rw,nosuid,nodev,noexec,relatime,size=32768k 0 0
configfs /sys/kernel/config configfs rw,relatime 0 0
tmpfs /dev/shm tmpfs rw,nosuid,nodev,noexec,relatime 0 0
/dev/block/mtdblock4 /system.ro squashfs ro,nodev,noatime 0 0
/dev/block/mtdblock5 /chrome squashfs ro,nodev,noatime 0 0
/dev/ubi6_0 /factory ubifs rw,relatime 0 0
/dev/ubi7_0 /cache ubifs rw,relatime 0 0
overlay /system overlay rw,relatime,lowerdir=/system.ro,upperdir=/cache/.system.overlay,workdir=/cache/.system.work 0 0
/dev/ubi7_0 /data ubifs rw,relatime 0 0
none /sys/kernel/debug debugfs rw,relatime 0 0
tmpfs /data/chrome/tmp tmpfs rw,relatime,size=1024k 0 0
tmpfs /data/share/chrome/metrics tmpfs rw,relatime,size=1024k 0 0
tmpfs /data/chrome/metrics tmpfs rw,relatime,size=1024k 0 0
adb /dev/usb-ffs/adb functionfs rw,relatime 0 0
</details>

<details>
<summary>find / | grep conf</summary>
/chrome/setup/http/audio/data/config.json
/system.ro/bin/ifconfig
/system.ro/bin/third_party/btconfig
/system.ro/chrome/setup/http/audio/data/config.json
/system.ro/etc/asound.conf
/system.ro/etc/bluetooth/bt_did.conf
/system.ro/etc/bluetooth/bt_stack.conf
/system.ro/etc/bluetooth/bt_stack.conf.debug
/system.ro/etc/bluetooth/rtkbt.conf
/system.ro/etc/bluetooth/rtl8821cs_config
/system.ro/etc/dbus-1/system.conf
/system.ro/etc/dhcpcd/dhcpcd-hooks/20-dns.conf
/system.ro/etc/dhcpcd/dhcpcd-hooks/95-configured
/system.ro/etc/dnsmasq.conf
/system.ro/etc/fonts/fonts.conf
/system.ro/etc/fonts/local.conf
/system.ro/etc/fw_env.config
/system.ro/etc/nsswitch.conf
/system.ro/etc/ota_domain_default.conf
[... 1084 lines cut ...]

service cast_auth_check /bin/cast_auth_check
    user root
    oneshot
</details>

In the `init.rc` file, the SSH service is disabled, as well as ADB. But I grabbed the ssh service commandline from there and pasted it to the end of the `set_usb_serialnumber.sh` file: `/usr/bin/sshd -G -r /etc/rsa_host_key`

[... cut ...]
1) Block the speaker from phoning home
Most likely by blocking a bunch of hosts in the `hosts` files. Persistance would not be needed for that as we could write that file on each boot.
1) Connecting the device to my home wifi and configuring it
I currently have no idea how to do that
1) Disable unwanted services
E.g. the updater. If changes on `init.rc` are persistant, this could be done there
1) Install additional services and run them at boot
Should be no problem as we can modify the system partition now.

duhow, 2023-05-03:
That's great news!! I'll need to gain some more trust to solder the USB header then...
What I'd suggest is:

Using `1.44.27` firmware version as `system` base image, create `patches` for the `l09g` model:
- remove OTA run
- remove chrome run (Google Assistant)
- add ssh with persistence key in `/data`
- allow console shell with root (`/etc/inittab`, or other?)

You could then create 2 images based from the same `system`, one more "light" (less size) to boot, and other with all programs (MPD, snapcast) to be flashed at `chrome` partition (large).
Maybe remount everything from `/` to the `chrome` partition.

duhow, 2023-05-03:
I understand that "keeping Google services" may be desired, but my main objective with this project is to fully replace all the propietary software with open source alternatives, and having Home Assistant project that is promoting Assistant and Voice, I'm very confident we can come up with something for it.
Also, we're here to experiment :D

duhow, 2023-05-03:
Configuring wifi "afterwards" requires some software / script to allow doing so. That's still pending to do.
What I'm using now is the script [wifi_connect](https://github.com/duhow/xiaoai-patch/blob/master/bin/wifi_connect) I wrote, but this is meant for `mico` images, not this Google one, so maybe it won't fully work for this case.
Still, you should be able to follow similar commands to configure `wpa_supplicant`.

mrmariosgithub, 2023-05-04:
> I'll need to gain some more trust to solder the USB header then...

I am currently working on a small program to flash the system partition using only the serial interface thus making the USB port obsolete. Once SSH is enabled flashing can be done wirelessly.

I initially tried to use a bash script but that was horribly slow (I estimated around 36 hours for the 9MB system partition).

Using my c program I was able to write the system partition in around 90 minutes but it turned out that I was writing data to fast and crashing the u-boot that way. I hope that I can fix the bug by combining the write and the verification process. I hope this way u-boot has enough time to process thw commands.

I‘ll keep you updated on my progress.

mrmariosgithub, 2023-05-06:
I finally managed to finialize and test the serial flash program. As stated before, I had to implement is in `c` because the shell script was terribly slow. I also created a small script to generate a modified `system.img` file with SSH enabled. I uploaded all my code to a dedicated repo: https://github.com/mrmariosgithub/l09g

Flashing the system partion takes about 6h to complete. To further speedup the process in the future we can generate a heavily stripped down system image.

mrmariosgithub, 2023-05-07:
The following idea just came to my mind: Why not create a minimal `system.img` that contains all required binaries like the ssh server, busybox etc. (and all symlinks to busybox) that can be used to initially flash the L09G using the serial port. But, here comes the twist: besides starting the ssh server, the modified system image also checks for a custom script on the chrome partition. If the script exists, it executes that script.

That said script is our loader to start all required programs, kill google and xiaomi programs and patches things where needed. As discussed before, we want to flash the chrome partition with our programs anyways. By using this loader-on-the-chrome-partition approach, we only need to flash the system partition once and everything else can we done using ssh.

What do you think about that? If you agree, we should start with the minimal system image first before starting to work on the chrome partition. I think developing the system image should not take that long.

duhow, 2023-05-10:
@mrmariosgithub I had to reimplement the same code in Python, I was facing some "line unmatch issues" (guessing due to the speed of serial processing), but got the same SSH working! 💪🏻 (`root` no password, lol)
I'll spend some time investigating this and see what we can do...

mrmariosgithub, 2023-05-10:
Very nice, good to hear that! Feel free to ping me if you want to discuss or test something :-)

mrmariosgithub, 2023-05-27:
Hi, did you made any progress with the L09G?

ds2k5, 2023-05-27:
> Very nice, good to hear that! Feel free to ping me if you want to discuss or test something :-)

sorry not now ...... still waiting for a fried he will buy  soldering station

hope end of june he have an can bring me the  micro usb port on

mrmariosgithub, 2023-06-04:
@duhow Can't we use [this](https://haxx.in/posts/dumping-the-amlogic-a113x-bootrom/) to dump the bootrom and OPT keys to flash a modified bootloader that does not check the signatures anymore?

The article is based on the sonos one gen2 which uses the same processor as the L09G but has more DRAM, so we might need to adjust some adresses. But as we have access to u-boot, we should be able to dump the FIP and BL3X from there. And as we have root access, we could use the kernel module as well.

duhow, 2023-06-04:
@mrmariosgithub as the speaker components contain propietary modules for wifi, sound, LEDs and so, I don't know if replacing the kernel / bootloader may remove this functionality.
I'm still on hold for the L09G due to work, but I plan to dedicate some time this summer.
My main focus for this project is to "patch base system to add new applications", but of course feel free to test anything that can improve functions or upgrade components version.

duhow, 2023-06-19:
File /[init.rc](https://github.com/duhow/xiaoai-patch/files/11789959/init.rc.txt) contains all the command run execution , but I cannot find where this file is located at.

What I've been playing so far is to execute `packages` built for LX06 successfully - though Speaker (ALSA) needs some investigation work (can't get sound atm).

Also I couldn't find any A/B system partition, which may make this a little bit more complex than expected...

mrmariosgithub, 2023-06-19:
> File /[init.rc](https://github.com/duhow/xiaoai-patch/files/11789959/init.rc.txt) contains all the command run execution , but I cannot find where this file is located at.

The `init.rc` file is part of the initramfs which is part of the bootloader. That’s why I wanted to have the possibility, to resign the bootloader or patch the signature verification.

> What I've been playing so far is to execute `packages` built for LX06 successfully - though Speaker (ALSA) needs some investigation work (can't get sound atm).

Great!!

> Also I couldn't find any A/B system partition, which may make this a little bit more complex than expected...

There is no A/B partition, there is just the system partition.

ds2k5, 2023-06-26:
@mrmariosgithub 

Hi,
had luck my friend could soldering the micoUSB-Port to the PCB.

Now I need helping hand to check if the port is working....

Tryed with "adb" but  "no devices"
so How can I check in the bootloader or somthing else if the USB-Port is working ?

duhow, 2023-06-26:
@ds2k5 `adb` might not be right available while the speaker is running in normal mode, but still you may be able to access it in `u-boot` - also `fastboot` may be available, so you can dump and flash content from USB directly.
Last, LX06 can be flashed with `amlogic-flash-tool`, so you can give it a try as well: https://github.com/duhow/xiaoai-patch/blob/master/research/lx06/install.md

ds2k5, 2023-06-26:
hi @duhow,

thanks......

 I tore off the mico-USB Port..... damned

Hope got help for repair....

mrmariosgithub, 2023-06-26:
Using the `usb start` command from uboot enables you to use USB mass storage device to load stuff from there. You can use the `fastboot` command from u-boot, if your solder connections are good you should see a new USB device on your PC and you should be able to connect to uboot using the fastboot windows tool.

Unfortunately, you cannot dump the NAND flash using fastboot as u-boot misses some functions required to dump the flash. But you can flash the NAND flash using fastboot, which is a lot better than having to use an USB mass storage device.

If you ripped off the micro USB pads from the PCB, there is still hope. Next to the micro USB pads there are not populated ESD protection devices. You can use those pads to solder on jumper wires which you can then solder to an USB socket. The marked pads are labeled on the silk screen on the bottom of the PCB, so you know which pad is which. When connecting the USB socket you basically do not need the +5V connection from the socket.
[image]

I ripped off my micro USB pads too, so I am using an USB A socket now with jumper wires and a lot of hot glue to keep it in place.

ds2k5, 2023-06-26:
@mrmariosgithub   Thank you!

Gaisberg, 2023-08-07:
Any updates on this?

ds2k5, 2023-08-07:
from my side - sorry no.... need to check after my friend holliday

ds2k5, 2023-12-27:
sorry no from my side could not fix the usb-port welding

psolyca, 2024-01-08:
Following this ^^ thanks guys for all this information.
I'm really interested. Already hacked some other devices ^^
It's just not my priority yet.

G33KatWork, 2024-01-10:
I know this might be naive, because they just might give a flying fuck about the GPL, but did anybody ever request the GPL sources for this thing from Xiaomi? We should at least get u-boot and the kernel which is everything we need.

psolyca, 2024-01-10:
You can try but I already know the answer.
I have done it for a project and people I know for others projects and it is really really hard to get them. If there are some proprietaries drivers it's impossible.

cosash, 2024-01-10:
@mrmariosgithub Would you share your "small programm that repetitively executed the nand dump command on the u-boot shell"? I have serial access but no usb stick soldered (my solder skills are really bad...), and I want a backup before I flash you modified system.img with ssh enabled.

cosash, 2024-01-10:
> @duhow Can't we use [this](https://haxx.in/posts/dumping-the-amlogic-a113x-bootrom/) to dump the bootrom and OPT keys to flash a modified bootloader that does not check the signatures anymore?
>
> The [...]

@G33KatWork Nice to see you here. Did you notice the linked article? I bet you are able to apply this..Or you know some hacker friend who is 😉

mrmariosgithub, 2024-01-10:
> @mrmariosgithub Would you share your "small programm that repetitively executed the nand dump command on the u-boot shell"?

The `nand dump` command dumps flash pages, you cannot (directly) use the dump to restore the partition. However, the system partition is part of the update file and you can simply extract the update and use the system partition image from there to reflash in case something went wrong.

> @G33KatWork Nice to see you here. Did you notice the linked article? I bet you are able to apply this..Or you know some hacker friend who is 😉

I tried to go that path but with no success. But I am no hacker, so someone else might succeed there.

There is an other approach to use a custom kernel module to dump the secure keys from within the booted OS. I managed to compile and load the module, but I did not manage to make it work as described.

However, there seems to be a way to patch/disable the u-boot signature verification. I first heard about that in a talk of Dennis Giese on how they hacked vacuum robots. The way it works is the following: u-boot is loaded to RAM and is executed from there. As the addresses of u-boot in RAM is fixed, the address of the signature verification routine is fixed as well. By patching a few bytes in RAM the verification can be disabled and u-boot no longer refuses to boot unsigned kernel images. Once the patch (both the bytes to patch and the address to patch) we can add the patch to one of the commands executed by u-boot at bootup and the signature validation is permanently turned off.

psolyca, 2024-01-10:
I will give it a try also. Same process than tegra hack (Switch)

psolyca, 2024-01-10:
Ok, just made the link with @G33KatWork... I'm please to meet you ^^
I watch many times your CCC video for glitching the Switch to apply this to an old Tegra T20 but did not have all the hardware to do it. This SOC is the only one we can not get the key ^^ and Fusee Gelee does not work on it.
So more skill than me on this ^^

duhow, 2024-01-15:
Bad news. 
I am stuck in a booloop from u-boot.

> AXG:BL1:d1dbf2:a4926f;FEAT:F0DC31BC:2000;POC:F;EMMC:800;NAND:0;READ:0;CHK:1F;READ:0;CHK:1F;READ:0;CHK:1F;READ:0;CHK:1F;READ:0;CHK:1F;READ:83;READ:0;CHK:1F;READ:0;CHK:1F;SD:800;USB:8;

jingyibo123, 2024-01-18:
> Bad news. I am stuck in a booloop from u-boot.
> 
Not sure if it's related, but seems like the updated firmware blocked serial and USB access? Last tested on LX06@1.84.39

duhow, 2024-01-18:
@jingyibo123 it's unrelated, this issue is for L09G. Each speaker has different firmware versions.

phste, 2024-01-25:
At the moment I'm trying to patch the signature verification. Loading of the kernel overrides the init bootarg, with the value in boot.img. I can preload the image with `imgread kernel boot 1080000` and then modify the init arg in memory and then boot with `bootm 1080000`. This change then triggers the signature check, but is only temporary and lost after a reboot. U-Boot is loaded into memory by relocation to 0x1fec3000, therefore I dumped a lot of memory from there and searched in the code for the signature check. At the moment I'm able to ignore the check for the boot if I call `mw.l 1fec8e40 d503201f` before `bootm`. Unfortunately, this somehow bricks the decompression of the kernel image. I then get this log 
```
aml log : R-2048 check fail with ERR = 1376

aml log : Sig Check 1376
avb2: 0
save_power_post ...
avb2: 0
## Booting Android Image at 0x01080000 ...
Kernel command line: rootfstype=ramfs init=/bin/sh console=ttyS0,115200 no_console_suspend earlycon=aml_uart,0xff803000
[store]Is good fdt check header, no need decrypt!
load dtb from 0x1000000 ......
      Amlogic Multi-DTB tool
      Single DTB detected
   Loading Kernel Image(COMP_NONE) ... OK
   kernel loaded at 0x01080000, end = 0x016a7800
Bad Linux ARM64 Image magic!(Maybe unsupported zip mode.)
```
If anybody has some ideas. Fixing the signature check would definitely allow us to modify the images as we please.

duhow, 2024-01-25:
> I can preload the image with `imgread kernel boot 1080000` and then modify the init arg in memory and then boot with `bootm 1080000`. 

If so, you can write a new `env` run command and use it by default as persistent, right?

phste, 2024-01-25:
> If so, you can write a new `env` run command and use it by default as persistent, right?
I was able to modify env variables at least in memory, I did not try saveenv yet. As I'm worried that it won't boot after that anymore.

duhow, 2024-01-25:
As long as you can return back to uboot, `saveenv` does not imply any risks.
Make sure to have a backup just in case, but if you're not deleting any entry should be fine :)

phste, 2024-01-26:
@mrmariosgithub I believe I managed to patch the signature check in memory. I tried to modify party of the boot partition but this does not work as it is encrypted. Maybe my work can help you modify the initrc as you were thinking about.

>  The encryption is not a problem - at least compared to the signature check.

Also I was wondering why the encryption is a smaller problem than the signature check?

mrmariosgithub, 2024-01-27:
> I believe I managed to patch the signature check in memory.

That’s great news, good job!

> > The encryption is not a problem - at least compared to the signature check.
> 
> Also I was wondering why the encryption is a smaller problem than the signature check?

Once the signature check is gone we can flash and boot our own kernel/system.

mrmariosgithub, 2024-02-01:
@phste Is the command `mw.l 1fec8e40 d503201f` still correct to disable the signature verification?

phste, 2024-02-01:
@mrmariosgithub Yes. I can check later this evening. It disables that U-Boot checks the signature before trying to boot, but one problem is the method which I patched is doing a systemcall to the secure call monitor. A bit deeper into the bootloader a stage below U-Boot, its BL31. What I could not verify is whether it would boot an unencrypted boot.img. As the system call is causing the encryption as far as I could see.

FloMaetschke, 2024-02-05:
Hi guys,
I also received an L09G today. I actually thought I had ordered an L09A, but that wasn't the case.

Unfortunately, I don't have much experience in device hacking. But I have already hacked a few devices following the instructions, connected TTL-to-serial, used uboot etc. My last project was a Kubernetes cluster running on several Android TV boxes.

So I'm not completely inexperienced, technically I could also understand what has been achieved so far and what you still have planned.

I am happy to offer my help if you want to check things again, e.g. run a tool for patching to see if a script runs as desired, etc. I still have the Smartspeaker here in its original packaging and will do my best to avoid setting it up.

My goal is also to get rid of the Google Assistant and use the whole thing as a Home Assistant Speaker/ MultiRoom Audio at some point.

Thank you for the good work you have already done.

FloMaetschke, 2024-02-17:
> @mrmariosgithub Yes. I can check later this evening. It disables that U-Boot checks the signature before trying to boot, but one problem is the method which I patched is doing a systemcall to the [...]

Have you found a solution to this?

noah-frank, 2024-03-22:
I have 3 L09G. If I can be any help let me know. I would love to use these speaker like my roborock with valetudo. I would donate if someone could make this happen.

mrmariosgithub, 2024-04-29:
I've got some spare minutes a few days ago and revisisted this issue.

I downloaded the firmware 1.44.21 for the L09G and extracted the `boot.img`. Then I cloned into `https://github.com/CyanogenMod/android_system_core.git` and unpacked the `boot.img`:

```
$ python3 android_system_core/mkbootimg/unpackbootimg -i boot.img -o tmp
Android magic found at: 0
BOARD_KERNEL_CMDLINE rootfstype=ramfs init=/init console=ttyS0,115200 no_console_suspend earlycon=aml_uart,0xff803000
BOARD_KERNEL_BASE 01080000
BOARD_RAMDISK_OFFSET 01000000
BOARD_SECOND_OFFSET 00f00000
BOARD_TAGS_OFFSET 00000100
BOARD_PAGE_SIZE 2048
BOARD_SECOND_SIZE 47104
BOARD_DT_SIZE 0
```

I immediatelly repacked it to get an unsigned boot image:

```
python3 android_system_core/mkbootimg/mkbootimg \
--base "0x$(cat tmp/boot.img-base)" \
--cmdline "$(cat tmp/boot.img-cmdline)" \
--kernel tmp/boot.img-zImage \
--kernel_offset "0x$(cat tmp/boot.img-kernel_offset)" \
--ramdisk tmp/boot.img-ramdisk.gz \
--ramdisk_offset "0x$(cat tmp/boot.img-ramdisk_offset)" \
--second tmp/boot.img-second \
--second_offset "0x$(cat tmp/boot.img-second_offset)" \
--output ./new-boot.img
```

Now I have "the same" boot image twice:
```
$ file boot.img new-boot.img
boot.img:     Android bootimg, kernel (0x1080000), ramdisk (0x1000000), second stage (0xf00000), page size: 2048, cmdline (rootfstype=ramfs init=/init console=ttyS0,115200 no_console_suspend earlycon=aml_uart,0xff803000)
new-boot.img: Android bootimg, kernel (0x1080000), ramdisk (0x1000000), second stage (0xf00000), page size: 2048, cmdline (rootfstype=ramfs init=/init console=ttyS0,115200 no_console_suspend earlycon=aml_uart,0xff803000)
$
$ ls -la --time-style=+"" boot.img new-boot.img
-rw-rw-r-- 1 mario mario 9148928  boot.img
-rw-rw-r-- 1 mario mario 9148416  new-boot.img
```

I copied both files to an USB drive and tried loading them with signature check on and patched. The stock `boot.img` worked in both cases.

Loading the `new-boot.img` with signature check on obviously makes the check fail:
```
axg_s420_v1_gva#usb start
(Re)start USB...
USB0:   USB3.0 XHCI init start
Register 1000140 NbrPorts 1
Starting the controller
USB XHCI 1.00
scanning bus 0 for devices... 2 USB Device(s) found
       scanning usb for storage devices... init_part() 282: PART_TYPE_DOS
1 Storage Device(s) found
axg_s420_v1_gva#fatload usb 0 ${loadaddr} new-boot.img
9148416 bytes read in 421 ms (20.7 MiB/s)
axg_s420_v1_gva#bootm ${loadaddr}

aml log : Sig Check 1423
axg_s420_v1_gva#
```

Patching the signature check leads to the error described above:
```
axg_s420_v1_gva#mw.l 1fec8e40 d503201f
axg_s420_v1_gva#usb start
(Re)start USB...
USB0:   USB3.0 XHCI init start
Register 1000140 NbrPorts 1
Starting the controller
USB XHCI 1.00
scanning bus 0 for devices... 2 USB Device(s) found
       scanning usb for storage devices... init_part() 282: PART_TYPE_DOS
1 Storage Device(s) found
axg_s420_v1_gva#fatload usb 0 ${loadaddr} new-boot.img
9148416 bytes read in 421 ms (20.7 MiB/s)
axg_s420_v1_gva#bootm ${loadaddr}

aml log : Sig Check 1423
avb2: 0
save_power_post ...
avb2: 0
## Booting Android Image at 0x01080000 ...
Kernel command line: rootfstype=ramfs init=/init console=ttyS0,115200 no_console_suspend earlycon=aml_uart,0xff803000
[store]Is good fdt check header, no need decrypt!
load dtb from 0x1000000 ......
      Amlogic Multi-DTB tool
      Single DTB detected
Host not halted after 16000 microseconds.
   Loading Kernel Image(COMP_NONE) ... OK
   kernel loaded at 0x01080000, end = 0x016a6000
Bad Linux ARM64 Image magic!(Maybe unsupported zip mode.)
axg_s420_v1_gva#
```

There are two possiblities that can cause the issue:
1. The signature patch is not working as intended
2. The repacked `new-boot.img` is corrupted

If someone comes up with an idea what to test next (new patch or fixed repacking) I can easily try that.

mrmariosgithub, 2024-05-03:
I was able to compile and "boot" a custom kernel on the L09G 😀

Unfortunately the kernel gets stuck at a kernel panic. **If someone can help me to get past that issue and drop into a busybox initramfs shell that would be awesome!**

(You need both UART and USB soldered to the board for this)

Here is what I did:
### Download and extract the kernel
The L09G uses a 4.9.113 kernel with custom/proprietary code. I decided to go for the 5.19.9 kernel as it is more recent and I thougt it could be better suited. I downloaded the kernel sources from here: https://cdn.kernel.org/pub/linux/kernel/v5.x/linux-5.19.9.tar.xz

### Create default config
```
make ARCH=arm64 CROSS_COMPILE=aarch64-linux-gnu- mrproper
make ARCH=arm64 CROSS_COMPILE=aarch64-linux-gnu- defconfig
```

### Tweak config
```
make ARCH=arm64 CROSS_COMPILE=aarch64-linux-gnu- menuconfig

Main menu -> Virtualization ==> disable
Main menu -> Platform selection ==> disable all and enable "Amlogic Platforms"
Main menu -> Boot options -> UEFI runtime support ==> disable
```

### Create and configure internal initramfs

See https://trac.gateworks.com/wiki/linux/initramfs

### Compile kernel and generate boot image
```
make ARCH=arm64 CROSS_COMPILE=aarch64-linux-gnu- Image -j8
mkimage -A arm64 -O linux -T kernel -C none -a 0x1080000 -e 0x1080000 -n l09g-custom -d arch/arm64/boot/Image ../uImage
```

### Load and boot the kernel on the L09G
```
axg_s420_v1_gva#mw.l 1fec8e40 d503201f
axg_s420_v1_gva#usb start
(Re)start USB...
USB0:   USB3.0 XHCI init start
Register 1000140 NbrPorts 1
Starting the controller
USB XHCI 1.00
scanning bus 0 for devices... 2 USB Device(s) found
       scanning usb for storage devices... init_part() 282: PART_TYPE_DOS
1 Storage Device(s) found
axg_s420_v1_gva#fatload usb 0 ${loadaddr} uImage
22464584 bytes read in 1006 ms (21.3 MiB/s)
axg_s420_v1_gva#setenv bootargs "keep_bootcon loglevel=15 debug noinitrd console=ttyS0,115200 no_console_suspend earlycon=meson,0xff803000"
axg_s420_v1_gva#bootm ${loadaddr}
```

### Boot log
```
aml log : Sig Check 1423
avb2: 0
save_power_post ...
avb2: 0
## Booting kernel from Legacy Image at 01080000 ...
   Image Name:   l09g-custom
   Image Type:   AArch64 Linux Kernel Image (uncompressed)
   Data Size:    22464520 Bytes = 21.4 MiB
   Load Address: 01080000
   Entry Point:  01080000
   Verifying Checksum ... OK
[store]Is good fdt check header, no need decrypt!
load dtb from 0x1000000 ......
      Amlogic Multi-DTB tool
      Single DTB detected
[... 219 lines cut ...]
[    0.899799] Kernel panic - not syncing: Attempted to kill init! exitcode=0x00000000
[    0.901855] CPU: 3 PID: 1 Comm: init Not tainted 5.19.9 #2
[    0.907362] Hardware name: Amlogic (DT)
[    0.911214] Call trace:
[    0.913662]  dump_backtrace.part.0+0xc8/0xd4
[    0.917949]  show_stack+0x18/0x5c
[    0.921274]  dump_stack_lvl+0x68/0x84
[    0.924949]  dump_stack+0x18/0x34
[    0.928274]  panic+0x16c/0x32c
[    0.931336]  do_exit+0x8c4/0x8cc
[    0.934574]  do_group_exit+0x34/0xa0
[    0.938161]  __arm64_sys_exit_group+0x18/0x20
[    0.942536]  invoke_syscall+0x48/0x114
[    0.946299]  el0_svc_common.constprop.0+0x44/0xfc
[    0.951024]  do_el0_svc+0x30/0xc0
[    0.954349]  el0_svc+0x34/0xb0
[    0.957411]  el0t_64_sync_handler+0xbc/0x140
[    0.961699]  el0t_64_sync+0x18c/0x190
[    0.965379] SMP: stopping secondary CPUs
[    0.969319] Kernel Offset: 0x80000 from 0xffff800008000000
[    0.974824] PHYS_OFFSET: 0x0
[    0.977711] CPU features: 0x0000,00120010,00001082
[    0.982524] Memory Limit: none
[    0.985593] ---[ end Kernel panic - not syncing: Attempted to kill init! exitcode=0x00000000 ]---
```

phste, 2024-05-03:
@mrmariosgithub Unfortunately, I was unable to devote a lot of time to it the last few weeks.  But its amazing that you are able to get a kernel to start booting up with the signature patch! What I was suspecting is that the kernel image is somehow encrypted, and by skipping the signature check (as my code is doing), the decryption is hindered. One aspect I'm not sure about is whether the kernel has some binary blobs in it for the audio processing and whether a 100% custom kernel destroys the audio recording quality. But that's a thing for testing. What also could be possible to replace the uboot with a custom compiled uboot that does not check for signature... I'll be checking out your kernel build, hopefully today or the next few days, to see whether I can somehow help.

FloMaetschke, 2024-09-20:
Heya, any progress on this?

vpeers, 2025-10-18:
@mrmariosgithub thanks, i want to try your serial method, but my shows firmware 1.52.54 and i'm not sure if https://cdn.cnbj1.fds.api.mi-img.com/xiaoqiang/rom/l09g/mico_l09g_6c5c9_1.44.27.bin system.img will be compatible with it. i tried to find newer download, but not luck

FloMaetschke, 2026-09-15:
Thx to @omnight for this.

I used claude to get a bit further:

# Rooting the Xiaomi LG09 / L09G Smart Speaker — persistent root SSH

Guide to get permanent root access over SSH on the Xiaomi **LG09 (L09G)** speaker —
**without** the Xiaomi app, using only the UART / U-Boot console.

- **SoC:** Amlogic `axg_s420_v1_gva`
- **Firmware:** Google Cast / Chromecast (Android `init`), kernel 4.9.113 aarch64
- **WLAN/BT:** Realtek RTL8821CS (SDIO)
- **Result:** `ssh -p 2222 root@<IP>` with an empty password, surviving reboots

> ⚠️ Your own device only. Flashing a signed partition (boot/kernel) wrong can brick the
> speaker. This guide writes **only** to the writable data partition (ubifs, mtd7) and its
> `/system` overlay — the signed partitions are **never** touched.

---

## 0. Prerequisites

- 3.3 V USB-TTL adapter (e.g. PL2303/CP2102) showing up as `/dev/ttyUSB0`
- Terminal program: `tio`, `picocom`, or `screen`
- Access to the speaker's UART pins (GND / box-TX / box-RX), **115200 8N1**

Open the serial console (example `tio`, with per-character delay so long pasted lines are
not dropped):

```bash
tio -b 115200 -o 5 /dev/ttyUSB0
```

If you lack permission on `/dev/ttyUSB0` (group `uucp`):

```bash
sudo setfacl -m u:$USER:rw /dev/ttyUSB0
```

---

## 1. Interrupt the U-Boot console

Unplug power, keep the terminal open, plug power back in and **immediately hit Enter
repeatedly** until the prompt stops at:

```
axg_s420_v1_gva#
```

---

## 2. Boot into a root ramfs shell

At the U-Boot prompt paste this **single** line (sets the kernel bootargs in RAM only —
**no `saveenv`**, so the original is left intact):

```
setenv bootargs 'rootfstype=ramfs rdinit=/sbin/busybox console=ttyS0,115200 no_console_suspend earlycon=aml_uart,0xff803000 -- ash -c "/sbin/busybox mkdir -p /proc /sys /dev /system /bin; /sbin/busybox mount -t proc proc /proc; /sbin/busybox mount -t sysfs sysfs /sys; /sbin/busybox mount -t devtmpfs devtmpfs /dev; /sbin/busybox mount -t squashfs -o ro /dev/mtdblock4 /system; /sbin/busybox --install -s /bin; export LD_LIBRARY_PATH=/system/lib; exec /bin/ash"'
```

Verify and boot:

```
printenv bootargs      # must end with ... exec /bin/ash"
run storeboot
```

You drop into a **root shell** (prompt `#`). The normal kernel/init is bypassed via
`rdinit`/`ash`; the signed partitions are mounted read-only only.
[... 372 lines cut ...]
ssh SPEAKER "setsid sh -c 'while :; do killall update_engine 2>/dev/null; sleep 3; done' </dev/null >/dev/null 2>&1 &"
```

---

## 20. Roadmap to the finished satellite

1. **Set up the HA host:** Home Assistant + Wyoming services **faster-whisper** (ASR),
   **Piper** (TTS), **openWakeWord** (wake word) + **wyoming-satellite** (e.g. as Docker
   containers).
2. **Make the speaker persistent:** add the cast shutdown (section 16) + audio bridge
   (section 18) to the boot hook.
3. Wire `wyoming-satellite` on the host to the speaker's TCP streams via
   `--mic-command`/`--snd-command`. Optional later: on-device wake word as a native ONNX
   binary.

## Appendix B: Additional paths / services

| Purpose | Value |
|---|---|
| speaker / mic | `hw:AMLAUGESOUND,2` / `hw:AMLAUGESOUND,3` |
| ALSA config | `/etc/asound.conf` (overlay: `/cache/.system.overlay/etc/asound.conf`), backup `/data/asound.conf.bak` |
| cast processes | `cast_shell`, `process_manager` (kill -9); services `cast_receiver`, `process_monitor`, `mute_service`, `ota_service` (`setprop ctl.stop/start`) |
| watchdog (keep running!) | service `watchdog` = `/bin/watchdog -F` |
| busybox (incl. nc) | `/sbin/busybox` |

sh4tteredd, 2026-11-05:
Hello, new here. Tried to follow step by step what @FloMaetschke wrote (many many many time), on my unit I could not get ssh work that way. Instead I was getting 

```
[   79.762673@0]- UBIFS (ubi0:0): reserved for root: 4952683 bytes (4836 KiB)
[   79.769488@0]- UBIFS (ubi0:0): media format: w4/r0 (latest is w4/r0), UUID 1382B989-657B-4EBF-ACF8-FE541ECF13A0, small LPT model
cat /per# ls -l /pers/.data/ssh/
/.-rw-rw-r-- root     root           27 2015-01-01 00:13 dropbear.log
-rw------- root     root          242 2015-01-01 00:11 dropbear_ecdsa_host_key
-rwxr-xr-x root     root          210 2015-01-01 00:12 start_ssh.sh
data# cat /pers/.data/ssh/dropbear.log 2>&1
disabled on stable-channel
#
```

so, without loosing my mind, I first enabled telnet (that is included) and starts flawlessly. Then there are two patches for this "disabled on stable-channel". I hope that somebody else is having this issue and wasn't just my skill issue. I'm pushing some docs regarding it into my fork

weuxstudio, 2026-11-10:
Since the ramfs bootargs from issue 49 boot straight off the original boot ramdisk, could you post /lib/firmware/msp430.txt from that shell together with its sha256 and the build number of your unit?

sh4tteredd, 2026-11-10:
As already stated in my last message, i have build 1.52.54. At os already booted, i don't have such txt

```
╰─$ ssh -p 2222 root@192.168.178.99
# ls /lib/firmware/
/lib/firmware/: No such file or directory
# cat /lib/firmware/msp430.txt
/lib/firmware/msp430.txt: No such file or directory
#
```