duhow/xiaoai-patch #53: Trying on L09b china vers.

Joshicopter, 2024-01-04:
Hello everyone,

I bought a l09b (Think thats the second version of L09a) and managed to get into the speaker over serial and flashed the firmware of the l09a. The Firmwareversion is 1.48.10. 

With only ssh patch activated I get into it with ssh over the created hotspot of the speaker. 

When using 96_compact_libs.sh or some of the other lib scripts, wifi stop working and bluetooth gets me an error.

Wifi:
```
error while loading shared libraries: libcurl.so.4: cannot open shared object file: No such file or directory
micocfg_default_volume_absolute: error while loading shared libraries: libcurl.so.4: cannot open shared object file: No such file or directory
```

```
wpa_cli
wpa_cli v2.7-devel
Copyright (c) 2004-2017, Jouni Malinen <j@w1.fi> and contributors

This software may be distributed under the terms of the BSD license.
See README for more details.

Interactive mode

Could not connect to wpa_supplicant: (nil) - re-trying

``` 

Bluetooth:
```
bluetoothctl: error while loading shared libraries: libgthread-2.0.so.0: cannot open shared object file: No such file or directory
```

Without the lib scripts wifi is working. Bluetooth not, but then without error.
The Buttons on the top also didnt work for me.

Tried the 1.69.54 which is the newest firmware, but this also didnt work for me. You can download the firmware here: https://cdn.cnbj1.fds.api.mi-img.com/xiaoqiang/rom/l09b/mico_firmware_8cdd1_1.69.54.bin

I hope someone can help :)

[image]

duhow, 2024-01-04:
Hi! Just to confirm, this is the model with built-in battery?
As you upgrade your firmware version, did you also flash new kernel partition? You can extract it with the tools available in repo.

Joshicopter, 2024-01-04:
Hey, 
thanks for reply.
Yes thats the model with the battery.
I download the image and use the mico_firmware.py to extract the update and the boot.img.
This version boots up, but nothing is really working with the l09a patches and scripts.

duhow, 2024-01-04:
The error you report (libraries not found) does not make much sense...
Same patching as L09A should work (maybe small diferences can be found)
Please confirm the following:
- The libraries exist in `squashfs-root` folder after `patching` the image? `/usr/lib`
- Running command `file` to library file reports as 
> ELF 32-bit LSB shared object, ARM, EABI5 version 1 (SYSV), dynamically linked

**NOTE:** You may want to use the script `wifi_connect` to reconfigure wireless network, since wpa_supplicant configuration may not be working initially after patching the speaker, so you can reconfigure with this script.

Also please share the `patch.log` .

Joshicopter, 2024-01-05:
I took a closer look at the log file and saw that it contains some errors.
[patch.log](https://github.com/duhow/xiaoai-patch/files/13842494/patch.log)

And not surprising that some of the symlinks are broken.
[libfiles.log](https://github.com/duhow/xiaoai-patch/files/13842517/libfiles.log)

duhow, 2024-01-05:
There seems to be an issue with `rsync` (app not installed or not running properly),
or your `build-packages` staging folder is empty, so the new built app files are not being copied (including libraries), that's why it does not work.

I have added a small check for this error copy in 73b2f35 .

Joshicopter, 2024-01-05:
Thank you for your help.
I switched from Debian to my Ubuntu PC and now it works. Rsync no longer has any problems. No idea what exactly it was on Debian.
I can use Bluetooth and play audio, the buttons are working too. Only the Shareplay connection still has problems. I can connect, but I can't get any sound. Do you know the best way to activate the log output for sharepoint on the speaker?

Joshicopter, 2024-01-05:
Sorry, I mean shairport-Sync with AirPlay2. I can connect to the speaker but get no sound.

Joshicopter, 2024-01-06:
After doing some research, I discovered that there is a problem with shairport-sync and nqptp. The latest version of nqptp (1.2.5) and shairport-sync (4.2) don't seem to be compatible. 
So I used the latest version of shairport-sync (4.3.2) and now it works fine. :tada:

duhow, 2024-01-06:
Usually the speakers may need tweak / patching for ALSA, this may be the case.

I uploaded several updates and fixes recently, since I don't have that model I can't test, but I trust everything else should be working as well.

Thanks and enjoy!

closed 2024-01-06 by duhow