duhow/xiaoai-patch #70: Lx01

al-mark-net, 2024-02-16:
hi, I just used the following command to flash lx01: 
`fastboot flash rootfs1 mico_firmware_7206d_240108_lx01.img`

when i reboot, The following information keeps recurring:
```
[   83.544666] can not support 4 channels 
[   83.549088] ac108 1-003b: can't set ac108-pcm0 hw params: -22
[   85.603248] can not support 4 channels 
[   85.607689] ac108 1-003b: can't set ac108-pcm0 hw params: -22
[   87.661938] can not support 4 channels 
[   87.666368] ac108 1-003b: can't set ac108-pcm0 hw params: -22
```

duhow, 2024-02-16:
Did you also flash the `kernel` partition matching the original firmware version?
What version were you previously?

al-mark-net, 2024-02-16:
Sorry, I didn't know I needed to flash the kernel.
Please tell me how to do it, I don't know how to flash the kernel.

al-mark-net, 2024-02-16:
`uname -a`
Linux LX01 3.4.39 #1 SMP PREEMPT Tue Nov 28 14:33:17 UTC 2023 armv7l GNU/Linux
I think the kernel version I was using before was 3.4.39.
I don't know how to compile and flash the kernel. Can you give me some tips? Thank you so much

duhow, 2024-02-16:
@al-mark-net that's not how it works.

Did you backup the `rootfs` partition previously before flashing?
Check for the `ROM version`  in `/usr/share/mico/version` file.

duhow, 2024-02-16:
Steps to do:
- ❕ backup original data partitions before flashing any other changes
- download update file version `1.32.6`
- extract update files
- flash `kernel.img` to the kernel partition (same number as the rootfs is running)
```
./tools/mico_upgrade.py lx01 release 1.32.6 | tail -n1 | jq -r .data.currentInfo.link
./tools/mico_firmware.py -e mico_all_*****_1.32.6.bin
```

al-mark-net, 2024-02-16:
`cat version`
```
config core 'version'
	# ROM ver
	option ROM '1.56.1'
	# channel
	option CHANNEL 'release'
	# hardware platform
	option HARDWARE 'LX01'
	# Bootload
	option UBOOT '1.56.1'
	# Linux Kernel ver
	option LINUX '0.0.1'
	# RAMFS ver
	option RAMFS '0.0.1'
	# SQUASHFS ver
	option SQAFS '0.0.1'
	# ROOTFS ver
	option ROOTFS '0.0.1'
	#build time
	option BUILDTIME 'Thu, 28 Oct 2021 21:33:21 +0800'
	#build timestamp
	option BUILDTS '1635428001'
	#build git tag
	option GTAG 'commit b9e9b6640c2491c7a77a22612e47790e6c8c0356'

config miio 'miio'
    #product id from iot.mi.com
	option product_id '0000'
	option module 'xiaomi.wifispeaker.lx01'
	option ssid_prefix 'xiaomi-wifispeaker-lx01_miap'

config player 'player'
	option VOLUME '200'
```
Thank you, bro
Nice project, looks a bit complicated, I will give it a try if necessary. thank you again

DevFlorian, 2024-04-03:
How do you Open LX01? There are on screws.

closed 2024-04-03 by duhow