duhow/xiaoai-patch #72: Xiaoai Speaker Pro Not recoginzed

DevFlorian, 2024-06-04:
I'm trying to flash the Xiaoai Speaker Pro, but it doesn't Work. How did you get Access to the device to flash it. It seems that my PC doesn't recoginze it.
I've installed amlogic flashtool and Worldcup Driver.

duhow, 2024-06-05:
You need to quickly interrupt the boot process by using the `update.exe identify` command.
It may not work at the first attempt.

> if you don't see this message then you wasn't successful and you need to try it again (power on -> wait -> run update identify)
📝 [Docs](https://github.com/duhow/xiaoai-patch/blob/master/research/lx06/install.md)

DevFlorian, 2024-06-05:
Now, i´m getting this error: AmlUsbIdentifyHost
This firmware version is [update]ERR(L783):fw ver is error! Maybe SOC wrong state

duhow, 2024-06-05:
Try with another USB cable?

DevFlorian, 2024-06-05:
It worked. But now it doesn't Show Up on https://www.improv-wifi.com/, when i Press Play 5 Times to Setup WiFi.
It just changes colors.

duhow, 2024-06-05:
Please confirm:
- what firmware version have you flashed?
- did you also flash kernel partition?
- does your host have Bluetooth enabled? (you can try Improv on your mobile phone as well)

DevFlorian, 2024-06-06:
I've used your Release v2024.1.8
I'm building a new Image now

duhow, 2024-06-06:
`v2024.1.8` uses https://voice-engine.github.io/hey-wifi/ , not `improv-wifi`.
Since the original issue is solved, I'm closing this one.

closed 2024-06-06 by duhow

Dinth, 2024-08-19:
> You need to quickly interrupt the boot process by using the `update.exe identify` command. It may not work at the first attempt.

Sorry for reopening a new ticket, but just trying to set up a brand new LX01 (purchased on AE a few months ago) and my speaker does never get recognized.

Im following the guide, but never even got a beep from Windows recognizing the device, i have also tried restarting the speaker and constantly running update.exe identify command. To troubleshoot i have also restarted the speaker on Linux, but even dmesg never sees anything connected to the usb port

duhow, 2024-08-19:
That process does not apply for LX01. However you can turn on the "download mode" by pressing [-] while powering on the first 2 seconds.

Haven't found a way to alter boot, but you can still unassemble it and jack into serial port. (Be patient, it's badly glued and fit into pressure...)

Dinth, 2024-08-19:
Apologies, this was a typo - ive got LX06.