duhow/xiaoai-patch #75: Connection Issue with Xiaomi Mi AI Speaker Model: MDZ-25-DA

dth97410, 2024-06-14:
Hello,I am facing an issue with my Xiaomi Mi AI Speaker. I have followed the steps to flash a new firmware, but I am stuck at the login step via the serial terminal where it asks for a login. 
Here are the details of my issue:
Steps Followed So Far:
Opened the Speaker and Connected the TTL Serial Pins: I identified the UART pins and connected a TTL-to-USB adapter for serial communication.Transferred the Firmware File: I used netcat to transfer the file mico_firmware_778f8_240108_lx06.img to the speaker.Verified the Transferred File: The file is present in the /tmp/ directory on the speaker.Attempted to Connect via Serial Terminal: At this step, the serial terminal asks for a login, but I do not know the default credentials.
Login Attempts:
I have tried the following combinations without success:Username: root, adminPassword: admin, root, 1234, passwordAdditional
 Information:
The boot logs show configuration operations and read/write attempts on configuration files, but there is a device initialization error (err=-22).I tried connecting using PuTTY, but each attempt fails at authentication.
I am looking to find out the default login credentials for this speaker or any other solution that might help me get past this step and complete the firmware flashing.Thank you in advance for your help!

duhow, 2024-06-14:
First of all, why would you flash a firmware from a different device?

I'm not sure if it is booting from the factory/recovery partition or from the newly flashed, but in case it's the new one, password should be `root`, unless it already copied a "generated" one to `/data/console/shadow` ...

dth97410, 2024-06-15:
I have tried to flash the firmware using netcat from another device, but have not been successful. It seems that the speaker boots from the factory/recovery partition since we failed to transfer the firmware.When it boots, it should ideally use "root" as the password unless a "generated" password has already been copied to /data/console/shadow.Also, I can access failsafe mode by pressing f while the speaker is rebooting. In this mode, I can change the root password. However, even after changing it, the new password doesn't seem to work as it still indicates that the password is incorrect.Can you tell me which is the correct firmware to use for my device? Also, if there is a specific procedure to make sure the password works correctly after resetting it, I'd be grateful if you could help me.

duhow, 2024-06-15:
This device is currently **not supported**, see the list in the readme file.

Can I ask you to backup all flash partitions and share the files here?

closed 2024-06-27 by duhow