duhow/xiaoai-patch #76: lx06: Patchset is out of date

nekromant, 2024-06-22:
When building from the current master, there are a bunch of (ignored) errors about failed patches. 
As a result, bluetooth_aplay service is not enabled, nor is started from /etc/init.d/bluetooth. 

```
>> scripts/10_apply_patches.sh
[*] Applying patches
>> patches/10_ssh.patch
patching file etc/config/dropbear
Reversed (or previously applied) patch detected!  Assume -R? [n] 
Apply anyway? [n] 
Skipping patch.
1 out of 1 hunk ignored -- saving rejects to file /dev/null
patching file etc/init.d/dropbear
Hunk #1 FAILED at 118.
1 out of 1 hunk FAILED -- saving rejects to file /dev/null
patching file etc/init.d/dropbear
Hunk #1 FAILED at 118.
1 out of 1 hunk FAILED -- saving rejects to file /dev/null
patching file etc/init.d/dropbear
patching file etc/init.d/dropbear
Hunk #1 succeeded at 63 (offset -1 lines).
Hunk #2 succeeded at 111 (offset -1 lines).
>> patches/11_autologin_serial.patch
patching file etc/inittab
>> patches/20_bluetooth.patch
patching file etc/init.d/bluetooth
Hunk #1 succeeded at 57 with fuzz 1 (offset 4 lines).
patching file etc/init.d/bluetooth
Hunk #1 FAILED at 56.
1 out of 1 hunk FAILED -- saving rejects to file /dev/null
patching file etc/init.d/bluetoothd
can't find file to patch at input line 38
Perhaps you used the wrong -p or --strip option?
The text leading up to this was:
--------------------------
|
|--- a/etc/init.d/check_mac     2019-12-25 04:18:35.000000000 +0100
|+++ b/etc/init.d/check_mac     2021-01-03 12:47:21.625258992 +0100
--------------------------
File to patch: 
Skip this patch? [y] 
Skipping patch.
3 out of 3 hunks ignored
can't find file to patch at input line 73
Perhaps you used the wrong -p or --strip option?
The text leading up to this was:
--------------------------
|--- a/etc/init.d/check_mac     2021-06-04 04:34:20.000000000 +0200
|+++ b/etc/init.d/check_mac     2024-01-05 17:31:52.984413254 +0100
--------------------------
File to patch: 
Skip this patch? [y] 
Skipping patch.
1 out of 1 hunk ignored
>> patches/21_bluetooth.patch
patching file etc/init.d/bluetoothd
patching file etc/init.d/bluetoothd
Reversed (or previously applied) patch detected!  Assume -R? [n] 
Apply anyway? [n] 
Skipping patch.
2 out of 2 hunks ignored -- saving rejects to file /dev/null
>> patches/21_disable_services.patch
patching file etc/init.d/dlnainit
Hunk #1 FAILED at 37.
1 out of 1 hunk FAILED -- saving rejects to file /dev/null
patching file etc/init.d/dlnainit
patching file etc/init.d/mitv-disc
Hunk #1 FAILED at 7.
1 out of 1 hunk FAILED -- saving rejects to file /dev/null
patching file etc/init.d/mitv-disc
can't find file to patch at input line 47
Perhaps you used the wrong -p or --strip option?
The text leading up to this was:
--------------------------
|--- a/etc/init.d/mdplay        2019-12-25 04:18:35.000000000 +0100
|+++ b/etc/init.d/mdplay        2021-01-03 12:47:22.005261065 +0100
--------------------------
File to patch: 
Skip this patch? [y] 
Skipping patch.
1 out of 1 hunk ignored
patching file etc/init.d/pns
Hunk #1 FAILED at 197.
1 out of 1 hunk FAILED -- saving rejects to file /dev/null
patching file etc/init.d/pns
>> patches/22_dhcp_hostname.patch
patching file etc/init.d/dhcpc
Hunk #1 FAILED at 11.
1 out of 1 hunk FAILED -- saving rejects to file /dev/null
patching file etc/init.d/dhcpc
>> patches/24_alsa_change_default.patch
patching file etc/asound.conf
>> patches/26_timezone.patch
can't find file to patch at input line 3
Perhaps you used the wrong -p or --strip option?
The text leading up to this was:
--------------------------
|--- a/etc/config/system        2019-12-25 04:18:35.000000000 +0100
|+++ b/etc/config/system        2021-01-03 12:47:22.305262701 +0100
--------------------------
File to patch: 
Skip this patch? [y] 
Skipping patch.
1 out of 1 hunk ignored
patching file usr/sbin/boot_function.sh
Hunk #1 succeeded at 221 (offset 30 lines).
>> patches/27_bootled.patch
patching file etc/init.d/done
Hunk #1 FAILED at 24.
1 out of 2 hunks FAILED -- saving rejects to file /dev/null
>> patches/30_libmico-pam.patch
patching file etc/pam.d/common-auth
patching file etc/pam.conf
>> patches/lx06/25_alsa_remove_dts.patch
patching file etc/asound.conf
Reversed (or previously applied) patch detected!  Assume -R? [n] 
Apply anyway? [n] 
Skipping patch.
1 out of 1 hunk ignored -- saving rejects to file /dev/null
>> patches/lx06/30_lx06_sysfixtime.patch
patching file usr/sbin/boot_function.sh
patch: **** malformed patch at line 13:  

>> patches/lx06/30_update_time.patch
patching file etc/init.d/wireless
Hunk #1 FAILED at 1179.
1 out of 1 hunk FAILED -- saving rejects to file /dev/null
patching file etc/init.d/wireless
Hunk #1 FAILED at 1156.
1 out of 1 hunk FAILED -- saving rejects to file /dev/null
>> patches/lx06/31_lx06_hostname.patch
patching file usr/sbin/boot_function.sh
Hunk #1 succeeded at 236 with fuzz 1 (offset 30 lines).
>> patches/lx06/31_lx06_persistent_shadow.patch
patching file etc/init.d/boot
Hunk #1 succeeded at 224 (offset 21 lines).
>> patches/lx06/40_bluealsa-aplay.patch
patching file etc/init.d/bluetooth
Hunk #2 FAILED at 27.
Hunk #3 FAILED at 35.
2 out of 3 hunks FAILED -- saving rejects to file /dev/null
----

```

SUGGESTION: Fail build and stop if any of the patches fail.

duhow, 2024-06-22:
@nekromant can you share what firmware version and speaker model are you using to build the image?

nekromant, 2024-06-23:
I'm using the firmware image recommended here: https://github.com/duhow/xiaoai-patch/blob/master/research/lx06/install.md 

My device is LX06, chinese mainland version.

duhow, 2024-12-18:
Since the reported issue is related to `blue-alsa` not starting, this was already fixed in 00d29c68c7c9234f29ec65f4d8486767fdfa45b4 .

The failure is unrelated to the patching itself.

closed 2024-12-18 by duhow