duhow/xiaoai-patch #83: Something wrong with the libraries

firedevel, 2024-07-06:
I didn't need extra features, so I modified the packages.sh: PACKAGES_TO_BUILD="update-libs update-binaries support jq avahi improv getevent ffmpeg shairport-sync"

but it gives an error when i run something on the speaker:

root@LX01-8083:~# shairport-sync
shairport-sync: error while loading shared libraries: /usr/lib/libc.so: invalid ELF header

root@LX01-8083:~# wifi_connect xxx xxx
...
Error relocating /usr/lib/libmico-common.so: EVP_CIPHER_CTX_cleanup: symbol not found
Error relocating /lib/libuuid.so.1: getrandom: symbol not found
Error relocating /usr/lib/libcrypto.so.1.1: getcontext: symbol not found
Error relocating /usr/lib/libcrypto.so.1.1: secure_getenv: symbol not found
Error relocating /usr/lib/libcrypto.so.1.1: makecontext: symbol not found
Error relocating /usr/lib/libcrypto.so.1.1: setcontext: symbol not found
Error relocating /usr/lib/libcrypto.so.1.1: __register_atfork: symbol not found
Error relocating /usr/lib/libmico-common.so: EVP_CIPHER_CTX_init: symbol not found
Error relocating /usr/lib/libmico-common.so: EVP_CIPHER_CTX_cleanup: symbol not found
Error relocating /lib/libuuid.so.1: getrandom: symbol not found
Error relocating /usr/lib/libcrypto.so.1.1: getcontext: symbol not found
Error relocating /usr/lib/libcrypto.so.1.1: secure_getenv: symbol not found
Error relocating /usr/lib/libcrypto.so.1.1: makecontext: symbol not found
Error relocating /usr/lib/libcrypto.so.1.1: setcontext: symbol not found
Error relocating /usr/lib/libcrypto.so.1.1: __register_atfork: symbol not found
Error relocating /usr/lib/libmico-common.so: EVP_CIPHER_CTX_init: symbol not found
Error relocating /usr/lib/libmico-common.so: EVP_CIPHER_CTX_cleanup: symbol not found
Error relocating /lib/libuuid.so.1: getrandom: symbol not found
Error relocating /usr/lib/libcrypto.so.1.1: getcontext: symbol not found
Error relocating /usr/lib/libcrypto.so.1.1: secure_getenv: symbol not found
Error relocating /usr/lib/libcrypto.so.1.1: makecontext: symbol not found
Error relocating /usr/lib/libcrypto.so.1.1: setcontext: symbol not found
Error relocating /usr/lib/libcrypto.so.1.1: __register_atfork: symbol not found
{ "name": "wireless" }
{ }
Error relocating /usr/lib/libmico-common.so: EVP_CIPHER_CTX_init: symbol not found
Error relocating /usr/lib/libmico-common.so: EVP_CIPHER_CTX_cleanup: symbol not found
Error relocating /lib/libuuid.so.1: getrandom: symbol not found
Command failed: Invalid argument

But the architecture seems correct: 
$ file '/home/ubuntu/ai/build/xiaoai-patch-master/squashfs-root/lib/libc.so' 
/home/ubuntu/ai/build/xiaoai-patch-master/squashfs-root/lib/libc.so: ELF 32-bit LSB shared object, ARM, EABI5 version 1 (SYSV), dynamically linked, no section header

firedevel, 2024-07-06:
And LX01 has bluetooth but why it ignores building bluetooth support

packages/update-libs/package.mk:
if [ "${BUILD_MODEL}" != "LX01" ]; then
   PACKAGE_DEPENDS="${PACKAGE_DEPENDS} bluez"
fi

firedevel, 2024-07-08:
# nm '/home/ubuntu/ai/build/xiaoai-patch-master/build-packages/staging/armv7/usr/lib/libcrypto.so.1.1'  | grep setcontext
         U setcontext@@GLIBC_2.4

# nm '/home/ubuntu/ai/build/xiaoai-patch-master/build-packages/s2t/armv7/usr/lib/libcrypto.so.1.1' 
nm: /home/ubuntu/ai/build/xiaoai-patch-master/build-packages/s2t/armv7/usr/lib/libcrypto.so.1.1：无符号(no symbol)

# nm '/home/ubuntu/ai/build/xiaoai-patch-master/squashfs-root/usr/lib/libcrypto.so.1.1' 
nm: /home/ubuntu/ai/build/xiaoai-patch-master/squashfs-root/usr/lib/libcrypto.so.1.1：无符号(no symbol)

What's happens when generating dictionary s2t?

duhow, 2024-07-08:
@firedevel have you extracted LX01 `squashfs-root` before building packages? This should define that the `packages.sh MODEL` is LX01 and will trigger the build for it, otherwise it will build for **LX06**, which has newer component versions and thus is incompatible with it.

Additionally, please confirm what firmware version are you using, and if you flashed the matching `kernel` partition as well.

firedevel, 2024-07-08:
I tried extract squashfs to xiaoai-patch-master/squashfs-root first, but there is still the same problem

$ sudo docker run -it -v $PWD:/xiaoai xiaoai-patch
------------------------------------------------------------------------------
Host arch         :  x86_64
Target arch       :  armv7
Model detected    :  LX01
Prefix            :  /usr
Make concurrency  :  8

Note: Prefix will be the path to all binaries in target device 
------------------------------------------------------------------------------
Process will begin shortly. Press enter or break run (Ctrl+C). 

I've flashed the kernel image in ota file to kernel1, the xiaoai-patch-master/release/lx01/latest to rootfs1 and boot currently

firedevel, 2024-07-08:
ROM Ver:1.56.1
OTA file: https://cdn.cnbj1.fds.api.mi-img.com/xiaoqiang/rom/lx01/mico_firmware_c1cac_1.56.1.bin
```
root@LX01-8083:~# uname -a
Linux LX01-8083 3.4.39 #1 SMP PREEMPT Tue Nov 28 14:33:17 UTC 2023 armv7l GNU/Linux
```

duhow, 2024-07-08:
Got it, please note that current target recommended version for LX01 is `1.32.6` .
During the build `packages.sh` process, if the model was previously wrong, you may need to rebuild all packages.

firedevel, 2024-07-09:
Thanks, I tried to use 1.32.6, `wifi_connect` seems works fine, but there is still the same problem with shairport-sync 
```
root@LX01-8083:/# shairport-sync 
shairport-sync: error while loading shared libraries: /usr/lib/libc.so: invalid ELF header
```

firedevel, 2024-07-09:
```
$ cat 'squashfs-root/usr/lib/libc.so' 
/* GNU ld script
   Use the shared library, but some functions are only in
   the static library, so try that secondarily.  */
OUTPUT_FORMAT(elf32-littlearm)
GROUP ( /lib/libc.so.6 /usr/lib/libc_nonshared.a  AS_NEEDED ( /lib/ld-linux-armhf.so.3 ) )
```
```
$ file 'squashfs-root/usr/lib/libc.so' 
squashfs-root/usr/lib/libc.so: ASCII text

$ file 'squashfs-root/lib/libc.so.6' 
squashfs-root/lib/libc.so.6: symbolic link to libc-2.27.so

$ file 'squashfs-root/lib/libc-2.27.so' 
squashfs-root/lib/libc-2.27.so: ELF 32-bit LSB shared object, ARM, EABI5 version 1 (GNU/Linux), dynamically linked, interpreter /lib/ld-linux-armhf.so.3, BuildID[sha1]=241476950cd0fe031d1460dcc35aed847010919a, for GNU/Linux 3.2.0, with debug_info, not stripped
```

firedevel, 2024-07-09:
Is the problem here?
```
$ sudo docker run -it -v $PWD:/xiaoai xiaoai-patch bash
root@438907228db9:/xiaoai# ld -T squashfs-root/usr/lib/libc.so --shared -o libc.so
ld: target elf32-littlearm not found
```

firedevel, 2024-07-28:
Any solutions?

duhow, 2024-09-08:
In short, you must keep the original `/lib/libc.so` from the firmware, and do not let the `patch` override that file with the text file you show. Additionally, the scripts will copy new libs built, but those should not override any of the existing (old) libraries.

What I don't understand is why you are seeing that text file...

firedevel, 2024-10-07:
I found that `squashfs-root/usr/lib/libc.so` was fine in the latest build, but it was a text file in the previous build, maybe it was caused by an interruption during previous building?

I'll test the latest firmware when I have some free time, thanks for your reply.

duhow, 2024-12-23:
I was able to reproduce the issue. Basically, the content is mainly stored in `/lib`, but system wants to check it in `/usr/lib` , which contains previous version and is not compatible.
Workaround found is to override on run, but I will prepare some patching for this.

```sh
mount --bind /lib/libc-2.27.so /usr/lib/libc.so
mount --bind /lib/libuuid.so.1.3.0 /usr/lib/libuuid.so.1.3.0
```

duhow, 2024-12-24:
Fixed by b9d3e8d4699433acfa7256591610751c5cb4ced9 .

closed 2024-12-24 by duhow