duhow/xiaoai-patch #86: squeezelite not working

DevFlorian, 2024-09-22:
when i build and Install squeezelite.
I get this Error when i Run it on my lx06.

==2498==ASan runtime does not come first in initial library list; you should either link runtime to your application or manually preload it with LD_PRELOAD.

Is there an error in package.mk?

duhow, 2024-12-19:
Not reproducible.

duhow, 2024-12-24:
Note for the future, several mentions of the reported issue:

> You should add `-fsanitize=address` to LDFLAGS as well.

> get rid of `/etc/ld.so.preload` on your test machine

> FWIW non-empty `/etc/ld.so.preload` was an exact reason why GCC 6 didn't work (ASan runtime indeed wasn't first in initial library list). GCC 4.9 works just because it's too old and doesn't have this sanity check.

> `-lasan`  is never the right way to fix this.
> Most likely your gcc installation is broken and I can hardly help here.
> But I would suggest to link asan statically (which is the default with clang).
> With gcc, use `-static-libasan` .

Still, without any additional changes I was able to build in several speaker platforms and runs properly.
Closing.

closed 2024-12-24 by duhow