duhow/xiaoai-patch #88: Recovering UDISK partition of an LX01

bench2012, 2024-10-23:
Hi,

I somehow manage to corrupt the UDISK partition of the LX01, now it just boot but can't configure anything. Is there anyway to recover this partition ? Fsck doesn't work with an error:

Error relocating /lib/libuuid.so.1: getrandom: symbol not found

and I can't read out the partition using fastboot:

couldn't parse max-fetch-size 'not supported'
fastboot: error: Unable to get max-fetch-size. Device does not support fetch command.

Any help appreciated since my LX01 is now a paper weight.

Many thx

duhow, 2024-12-03:
Sorry for the delay, `UDISK` partition is meant for "configuration" data, so the system should continue to work as usual.
It is mounted as `Linux EXT filesystem, rev 1.0, ext4 filesystem data` so you could format with `mkfs.ext4`, despite `lib/preinit/79_format_partition` should be already doing this by you?

bench2012, 2024-12-13:
Thanks for the reply. Will give this a try. BTW, any update on getting LX01 release on the latest version. I try compiling it, but as stated the image got too big for the partition.

Agani, thanks for your reply

duhow, 2024-12-13:
During build testing I did lock myself out as well, so I'm out of spare LX01 to test, but I still can build the image.

duhow, 2024-12-18:
Unfortunately I'm facing the same issue, the image generated is too large to build... 😮‍💨
I'll try to optimize it, but in the mean time I'm afraid next release will not contain this version, since I am currently unable to test it.

You can build a custom image, I suggest you remove the following: `mpd`, `upmpdcli`, `screen`, `python3`, `core_api` and `improv`.

duhow, 2024-12-18:
Also you still have previous version [2024.1.8](https://github.com/duhow/xiaoai-patch/releases/tag/2024.1.8).

bench2012, 2024-12-20:
Yes, that’s the version I end up with, since I can’t install a compile latest version.

closed 2024-12-20 by duhow