duhow/xiaoai-patch #93: Flashing on linux

Dinth, 2024-12-19:
Hi. Has anyone managed to flash the device on linux? Im trying to follow the LX06 guide, but "update.exe identify" crashes with an error message or even whole computer when im trying it on Windows.
So i thought i will give it a try on Linux using this tool: https://github.com/radxa/aml-flash-tool/

Here the update tool seems to work in general, but still im getting an error when it tries to connect:
```
dinth@dinth-mint:~/Downloads/aml-flash-tool-1.0.0/tools/linux-x86$ sudo ./update identify
[update]No [WorldCup Device] device after scan
[update]ERR(L1094):can not find device
dinth@dinth-mint:~/Downloads/aml-flash-tool-1.0.0/tools/linux-x86$ sudo ./update identify
^[[A[update]No [WorldCup Device] device after scan
[update]ERR(L1094):can not find device
dinth@dinth-mint:~/Downloads/aml-flash-tool-1.0.0/tools/linux-x86$ sudo ./update identify
AmlUsbIdentifyHost
IOCTL_IDENTIFY_HOST_Handler ret=-71 error_msg=error sending control message: Protocol error
ERR: get info from device failed
dinth@dinth-mint:~/Downloads/aml-flash-tool-1.0.0/tools/linux-x86$ sudo ./update identify
[update]No [WorldCup Device] device after scan
[update]ERR(L1094):can not find device
dinth@dinth-mint:~/Downloads/aml-flash-tool-1.0.0/tools/linux-x86$ sudo ./update identify
[update]No [WorldCup Device] device after scan
[update]ERR(L1094):can not find device
dinth@dinth-mint:~/Downloads/aml-flash-tool-1.0.0/tools/linux-x86$ sudo ./update identify
[update]No [WorldCup Device] device after scan
[update]ERR(L1094):can not find device
dinth@dinth-mint:~/Downloads/aml-flash-tool-1.0.0/tools/linux-x86$ sudo ./update identify
```
dmesg shows the following:
```
[  439.787267] usb 1-5.2: new high-speed USB device number 6 using xhci_hcd
[  439.893152] usb 1-5.2: New USB device found, idVendor=1b8e, idProduct=c003, bcdDevice= 0.07
[  439.893163] usb 1-5.2: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[  441.399363] usb 1-5.2: USB disconnect, device number 6
[  467.882855] usb 1-5.2: new high-speed USB device number 7 using xhci_hcd
[  467.989382] usb 1-5.2: New USB device found, idVendor=1b8e, idProduct=c003, bcdDevice= 0.07
[  467.989390] usb 1-5.2: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[  469.558657] usb 1-5.2: USB disconnect, device number 7
[  509.690219] usb 1-5.2: new high-speed USB device number 8 using xhci_hcd
[  509.798192] usb 1-5.2: New USB device found, idVendor=1b8e, idProduct=c003, bcdDevice= 0.07
[  509.798202] usb 1-5.2: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[  511.144243] usb 1-5.2: usbfs: USBDEVFS_CONTROL failed cmd update rqt 192 rq 32 len 4 ret -71
[  511.285439] usb 1-5.2: USB disconnect, device number 8

```

duhow, 2024-12-19:
So far I was only able to run this on Windows 10. Haven't tried Windows 11.
Any new findings are welcome.

Dinth, 2024-12-19:
Thanks, now i have managed to do that on Windows 10 **and** with cmd run with admin privileges. 

Regarding the next steps, what toolset should i use for building the firmware? Im kinda n00b when it comes to windows, should i build it using WSL2 and Docker for Desktop? 

Also, im not sure if i get it correctly, after backing up the original firmware i should pair my speaker with the phone app to get latest (?) updates and setup? As far as i understand the documentation xiaoai-patch needs to work on an image of the speakers partitions taken from the device or firmware updates - so should i make another backup after upgrading it on the app?

duhow, 2024-12-19:
If you apply the patch as-is without any customization on your side, the Mi Speaker will not work with Xiaoai app.
Instead, it is repurposed to work with **Home Assistant**, from there you can use it as a **Assist Satellite** (Voice Assistant).

Having the full backup of the flash will be useful in case you need to restore its functionality.

From recent release, you can download the prebuilt image, and flash it via the `aml-flash-tool` to the running partition (boot0 and system0).

Dinth, 2024-12-19:
I have tried flashing that with update.exe partition:
1. flashed boot file to boot1 partition
2. flashed system.squashfs to system0 and system1 as per guide
Unfortunately nothing has changed. Should i also flash boot0?

What would be the default admin password for ssh?

duhow, 2024-12-19:
What do you mean "nothing has changed"? Can you provide the commands you run?

Unless there's any other password previously persisted,
https://github.com/duhow/xiaoai-patch/blob/ba061387fdff683b018f29474d77bdc638a1e421/scripts/12_root_password_factory.sh#L7

Dinth, 2024-12-19:
i have downloaded the prebuilt image: https://github.com/duhow/xiaoai-patch/releases/download/2024.12.18/mico_firmware_c313391_241218_lx06.tar, unpacked it and run the following
```
update.exe partition boot1 ./boot
update.exe partition system0 ./root.squashfs
update.exe partition system1 ./root.squashfs 
```
(commands taken from: https://github.com/duhow/xiaoai-patch/blob/master/research/lx06/install.md )
Now i have also tried 
`update.exe partition boot0 ./boot`

after that i am restarting the speaker, connecting to its own wifi network and trying to:
`ssh root@10.0.0.1 `
The IP is the gateway ip from dhcp provided by the speaker.
Unfortunately this fails with "Connection refused" without even asking for a password.

duhow, 2024-12-19:
File is named `boot.img` , not `boot`. Looks like Windows is hiding the extension.
If flashed correctly, press 5 times ⏯️ to start [improv-wifi](https://www.improv-wifi.com/) and use your phone to connect it to Wifi.

closed 2024-12-19 by Dinth

duhow, 2024-12-19:
I will update the docs at some point, since the prebuilt images are somewhat new, and they were not available at that time.
Glad to know everything went fine! 😊