duhow/xiaoai-patch #97: LX01: shairport-sync still doesn't work

firedevel, 2024-12-29:
Thanks for the patch! But there's still problems with lx01 using `shairport-sync`

`shairport-sync` starts up normally, but an error occurs when trying to connect: 
```
root@LX01-XXXX:/# shairport-sync 
Fatal: getentropy is not supported: Function not implemented
Aborted (core dumped)
```

I think it might be caused by the OpenSSL version being too low, this is the related issue:  
https://github.com/openssl/openssl/issues/13002

firedevel, 2024-12-29:
After testing, it works normally when airplay2 support is turned off, so the issue might be with OpenSSL.

firedevel, 2024-12-29:
I tried using https://github.com/openssl/openssl/tree/OpenSSL_1_1_1-stable, but it didn't work.

duhow, 2024-12-29:
Thanks for the report, can you confirm what firmware version of LX01 are you using?

firedevel, 2024-12-29:
1.32.6 from ota package.

```
# uname -a
Linux LX01-2982 3.4.39 #1 SMP PREEMPT Tue Nov 28 14:33:17 UTC 2023 armv7l GNU/Linux
```

I tried using the kernel extracted from 1.32.6 and 1.62.6, and the results were the same. The kernel currently in use is the one from 1.62.6.

duhow, 2024-12-29:
Unable to reproduce with 1.32.6 . If you did any changes in package build or similar, or if you can provide any more details, that may help.

```
0.000483000 "shairport.c:2291" Version String: "4.3.2-AirPlay2-smi10-alac-OpenSSL-Avahi-external_mdns-ALSA-stdout-pipe-soxr-convolution-metadata-dbus-mpris-sysconfdir:/etc"
```

firedevel, 2024-12-29:
Package.sh:

```
PACKAGES_TO_BUILD="update-libs update-binaries support shairport-sync"

```

It starts up normally, but it crashes when you try to stream something on it.

duhow, 2024-12-29:
I notice the device completely freezes, but no error shown on UART or app.

```
         0.004568583 "rtp.c:513" AP1 control thread: set_ntp_anchor_info: rtptime: 1996236851, networktime: 36bdafbb5e59d4da, frame adjustment:  94.888.
         0.000587542 "player.c:1155" Check packet from buffer 15177, timestamp 1996245545, 0.195198 seconds ahead.
         0.004457833 "player.c:1155" Check packet from buffer 15178, timestamp 1996245887, 0.198494 seconds ahead.
         0.004450500 "player.c:1155" Check packet from buffer 15179, timestamp 1996246229, 0.201799 seconds ahead.
         0.004454709 "player.c:1155" Check packet from buffer 15179, timestamp 1996246229, 0.197346 seconds ahead.
         0.004244500 "player.c:1155" Check packet from buffer 15180, timestamp 1996246571, 0.200854 seconds ahead.
         0.002878625 "player.c:1155" Check packet from buffer 15180, timestamp 1996246571, 0.197977 seconds ahead.
         0.004157583 "player.c:1155" Check packet from buffer 15181, timestamp 1996246913, 0.201574 seconds ahead.
         0.006446625 "player.c:1155" Check packet from buffer 15181, timestamp 1996246913, 0.195129 seconds ahead.
         0.004239292 "player.c:1155" Check packet from buffer 15182, timestamp 1996247255, 0.198644 seconds ahead.
         0.001447791 "mdns_avahi.c:487" no change...
         0.000062667 "dacp.c:503" mutex_lock "&dacp_server_information_lock".
         0.002853583 "player.c:1155" Check packet from buffer 15183, timestamp 1996247597, 0.202035 seconds ahead.
         0.002197709 "player.c:1155" Check packet from buffer 15183, timestamp 1996247597, 0.199838 seconds ahead.
         0.004249541 "player.c:1155" Check packet from buffer 15184, timestamp 1996247939, 0.203341 seconds ahead.
         0.002734000 "player.c:1155" Check packet from buffer 15184, timestamp 1996247939, 0.200608 seconds ahead.
```

duhow, 2024-12-30:
I think it is related to the 🔊  volume + power rather than a build issue. I was able to play for a couple of minutes without problem, and when I increased the volume, it would freeze.

firedevel, 2024-12-30:
Thanks for your work again, It seems that the problem I had isn’t the same, I can’t successfully connect to AirPlay, and the same goes for HomeKit.

But why does the firmware you compiled work, have you enabled  AirPlay 2 support?

I will test and provide more information tomorrow.

duhow, 2024-12-30:
AirPlay 2 support is enabled by default to all builds, although it depends on the sender (PC or device) that it may not be triggered, and use AirPlay Classic instead.
I'm currently sending audio via Linux PC Pulseaudio with `pactl load-module module-raop-discover` .

firedevel, 2024-12-30:
Additionally, I manually set the BUILD_MODEL in package.sh. What should I do before building the Docker image?

firedevel, 2024-12-31:
[image]
[image]

duhow, 2024-12-31:
Thanks for the report. Do you have any other speaker model to test as well?

I suspect that the issue is not an old OpenSSL version, but instead LX01 kernel / system which is too old.
This cannot be fixed by upgrading to latest xiaoai base version, as they still run the same Linux kernel version `3.4.39` , but it's possible that we can do a patch to openssl to workaround this issue.

firedevel, 2024-12-31:
I found that `getentropy` requires Linux 3.14 or higher to work properly.

1.32.6 with glibc 2.24(getentropy not included):
```
root@LX01-2982:/# shairport-sync 
Fatal: no entropy gathering module detected
Aborted (core dumped)
```
Maybe it's not just about OpenSSL, just like https://github.com/openssl/openssl/issues/13002, it could be a problem with the cross-compilation toolchain, because many packages provide old methods, but they have not been enabled.

libgcrypt/configure.ac:  
```
LIST_MEMBER(getentropy, $random_modules)
if test "$found" = "1" ; then
   GCRYPT_RANDOM="$GCRYPT_RANDOM rndgetentropy.lo"
   AC_DEFINE(USE_RNDGETENTROPY, 1, [Defined if the getentropy RNG should be used.])
fi

LIST_MEMBER(linux, $random_modules)
if test "$found" = "1" ; then
   GCRYPT_RANDOM="$GCRYPT_RANDOM rndoldlinux.lo"
   AC_DEFINE(USE_RNDOLDLINUX, 1, [Defined if the /dev/random RNG should be used.])
fi
```

firedevel, 2024-12-31:
It’s probably caused by `libgcrypt` or `openssl` because I removed `openssl.status` `libsodium.status` `libgcrypt.status` `shairport-sync.status` `uuid.status` when compiling.

firedevel, 2024-12-31:
I reran the entire build process to ensure that there were no additional modifications and
 confirm that the issue is most likely with `libgcrypt`.  
I have removed these from `configure.ac `and `configure`:
```
LIST_MEMBER(getentropy, $random_modules)
if test "$found" = "1" ; then
   GCRYPT_RANDOM="$GCRYPT_RANDOM rndgetentropy.lo"
   AC_DEFINE(USE_RNDGETENTROPY, 1, [Defined if the getentropy RNG should be used.])
fi
```
```
name=getentropy
list=$random_modules
found=0

for n in $list; do
  if test "x$name" = "x$n"; then
    found=1
  fi
done

if test "$found" = "1" ; then
   GCRYPT_RANDOM="$GCRYPT_RANDOM rndgetentropy.lo"

$as_echo "#define USE_RNDGETENTROPY 1" >>confdefs.h

fi
```
And the error changed from
```
# shairport-sync -c /data/shairport-sync.conf -o pipe
Fatal: getentropy is not supported: Function not implemented
Aborted (core dumped)
```
to
```
# shairport-sync -c /data/shairport-sync.conf 
Fatal: no entropy gathering module detected
Aborted (core dumped)
```

firedevel, 2025-01-01:
I’ve finally fixed it, now I’m going to prepare the PR.

closed 2025-01-01 by firedevel