duhow/xiaoai-patch #108: i need a firmware for L05G  Mi smart speaker IR control

ithaqua974, 2025-01-20:
I bought a Mi Smart Speaker IR Control, but I made a mistake and bought a Chinese instead of a global or French one.
It is too late to return the product and I don't have money to buy another one with the right firmware.
I was hoping it to be possible to change the firmware to one that supports French but unfortunately, you don't have this one and it is the only place I can find firmware for Xiaomi speakers.

Do you plan to have this device firmware?
And is it possible to do what I want to do?

thanks a lot

duhow, 2025-01-21:
L05B and L05C are unsupported.
L05G is unsupported.

And at least, given the Chinese edition uses NuttX instead of Linux, I don't think the conversion is quite straight forward.

Still, you can find the firmware in the [research/updates.md](https://github.com/duhow/xiaoai-patch/blob/master/research/updates.md) file. Anything you do is at your risk.

closed 2025-01-21 by duhow

ithaqua974, 2025-01-21:
I looked at my device once more and I found another mistake I thought I got an L05G but when I watched under the speaker the label said it was an L05C not an IR control but a player-enhanced edition

[image]