duhow/xiaoai-patch #110: How do I make my LX06 a voice assistant?

flowerdanceee, 2025-01-30:
Dear duhow,

   I successfully flashed the firmware to my lx06 and found it on homeassistant.
    But now it is just a Bluetooth speaker in my hand. How can I wake it up and make it my voice assistant?
    By the way, I would also like to ask, can the lx06 in your hand be connected to the API of language models such as chatGPT?
    Thank you very much for your contribution as a developer of XiaoAi.

flowerdance

duhow, 2025-01-30:
Hi! Once you flash the speaker, Home Assistant may find different services (MPD, UPNP, Airplay). 👈🏻 ❌ Those do not provide Voice Assistant capabilities.

You need to configure the speaker with a token of Home Assistant, as explained in 📝 https://github.com/duhow/xiaoai-patch/blob/master/docs/voice-assistant.md .

If you go to the IP address of the speaker (eg. `http://192.168.1.40`) , you will see a website to configure STT and options.

XiaoAi speaker won't connect to ChatGPT, instead it's up to **Home Assistant** - [Agent Conversation](https://my.home-assistant.io/redirect/voice_assistants/), so you can configure it there.

flowerdanceee, 2025-01-30:
Yes, I found the IP address of the lx06 and can configure it by entering parameters through the web page. But if I want to manually configure more information, where can I find the folder 'data/listener'?

duhow, 2025-01-31:
You can `ssh` into the speaker with the modified password - patch changes it to `root`.

closed 2025-01-31 by duhow

flowerdanceee, 2025-02-01:
After I flashed the patch, do I need to solder the motherboard to ssh into my lx06? Although I know the IP under the LAN, I don’t know what the initial username is. How can I ssh into my lx06? Please save me (T_T)

wjcroom, 2025-03-22:
so are all the official features gone, @flowerdanceee how did it work please? I'd like to try it, I have one LX06 too.
Recently I've been accessing homeassitant and using xiaomi miot auto to interact with it, voice activated to play movies on TV from the NAS.   [this is the artical ](https://blog.csdn.net/wjcroom/article/details/146357771)
I'd like to add the cue word commands to the wildcard commands if I can keep the official functionality yet.
I haven't found the flashing steps yet, can you guide me?
so  官方功能是否都没有了， @flowerdanceee   请问怎么成功的？我也想试试，我也有一个。  
最近在接入 homeassitant   并用 xiaomi miot auto ，去交互，声控去播放NAS上的电影。
要是能保留官方功能，我想把提示词指令加入通配符指令呢，
我还没找到刷机步骤，可否指导一下？

wjcroom, 2025-03-22:
我找到文章说，打开盖板， 接上ttl 触点[是这样的方法](https://javabin.cn/2021/xiaoai_fm.html)