duhow/xiaoai-patch #112: S12 (aka MDZ-25-DT Xiaomi Mi AI Speaker 1 Gen )

MrC0nst, 2025-02-15:
Hi everyont!

Is it possible to flash this Xiaomi AI Speaker?

[image]

[image]

[image]

**UART:** soldered, J31
**USB:** not soldered, J32

**Printenv**
```
s12#printenv
aml_dt=xiaomi_s12a_v01
baudrate=115200
board_id=2
boot_failcnt=0
boot_failed=if itest ${boot_failcnt} == 1; then setenv boot_failcnt 2; setenv boot_part boot1; else if itest ${boot_failcnt} == 2; then setenv boot_failcnt 1; setenv boot_part boot0; else run set_boot_flag;fi;fi;
boot_part=boot0
bootargs=rootfstype=ramfs init=/init console=ttyS0,115200 no_console_suspend quiet earlycon=aml_uart,0xff803000 jtag=apao reboot_mode=cold_boot uboot=U-Boot 2015.01 (Feb 28 2019 - 04:51:12)
bootcmd=run storeboot
bootdelay=1
deviceid=18090/981388144
dtb_mem_addr=0x1000000
factory_detect=if keyman read deviceid ${loadaddr} str; then echo HAVE SN Code ...; else echo Switch to boot1 system, because of NOT found SN Code ...; setenv boot_failcnt 2; setenv boot_part boot1; fi;
fdt_high=0x20000000
firstboot=1
identifyWaitTime=1000
initargs=rootfstype=ramfs init=/init console=ttyS0,115200 no_console_suspend quiet earlycon=aml_uart,0xff803000 
jtag=apao
loadaddr=1080000
preboot=run storeargs;if test ${reboot_mode} = cold_boot; then run try_auto_burn; fi;
product_model=S1202
reboot_mode=cold_boot
rpmb_state=0
set_boot_flag=if test ${boot_part} = boot0; then setenv boot_failcnt 1; else setenv boot_failcnt 2; fi;
stderr=serial
stdin=serial
stdout=serial
storeargs=setenv bootargs ${initargs} jtag=${jtag}; setenv bootargs ${bootargs} reboot_mode=${reboot_mode} uboot=${version};keyman init 0x1234;
storeboot=if test ${reboot_mode} = cold_boot; then run set_boot_flag; run factory_detect; else run boot_failed; fi; saveenv; if imgread kernel ${boot_part} ${loadaddr}; then bootm ${loadaddr}; fi; reset;
try_auto_burn=update 500 1000;
ubootenv_version=1
upgrade_step=2
version=U-Boot 2015.01 (Feb 28 2019 - 04:51:12)

Environment size: 1775/65532 bytes
```

**Boot log**
```
AXG:BL1:d95c53:a4926f;FEAT:E0DC318C:2000;POC:F;EMMC:800;NAND:0;READ:0;0.0;CHK:0;
sdio debug board detected 
TE: 22920

BL2 Built : 18:30:39, Aug 28 2018. axg g56303a2-dirty - liang.yang@droid11-sz

set vcck to 1140 mv
set vddee to 1070 mv
Board ID = 2
CPU clk: 1200MHz
DDR low power enabled
DDR3 chl: Rank0 16bit @ 792MHz
bist_test rank: 0 31 0b 57 37 12 5d 2a 03 52 37 14 5b 00 00 00 00 00 00 00 00 00 00 00 00 612   - PASS
Rank0: 256MB(auto)-2T-11
AddrBus test pass!
NAND init
[... 186 lines cut ...]
Please press Enter to activate this console.
[    5.731325@1] cyttsp 0-0008: Failed to request firmware touch.cyacd
[    5.731959@1] cyttsp 0-0008: Failed to update firmware;
[    7.105936@2] i2c i2c-1: [aml_i2c_xfer] error ret = -5 (-EIO)
[    7.106063@2] i2c i2c-1: token 1, master_no(1) 300K addr 0x3c
[    7.581380@0] name: mac_wifi, size 17
[    7.597039@1] name: mac_bt, size 17
crond[1095]: crond (busybox 1.27.2) started, log level 5

BusyBox v1.27.2 () built-in shell (ash)

 _____  _              _____  ___    ___  _____ 
|     ||_| ___  ___   |   __||_  |  |_  ||  _  |
| | | || ||  _|| . |  |__   | _| |_ |  _||     |
|_|_|_||_||___||___|  |_____||_____||___||__|__|  
-------------------------------------------------
    Reboot (SNAPSHOT, 70-1-1)
-------------------------------------------------

=== WARNING! =====================================
There is no root password defined on this device!
Use the "passwd" command to set up a new password
in order to prevent unauthorized SSH logins.
--------------------------------------------------
```

duhow, 2025-02-15:
It should be supported, but I don't have any units to test. Feel free to build the image and flash it. Make sure to backup current flash!

MrC0nst, 2025-02-17:
I changed the issue title because the device is defined as S12A everywhere.

The USB connector was soldered:
[image]

Firmware was updated via chinese ai speaker app to s12a_mico_firmware_4babb_1.76.54.bin
I modded this firmware with following settings:

- username and password is root
- uart console enabled
- ssh enabled by default
- no DSA_verify with serial number and etc

```
BusyBox v1.27.2 () built-in shell (ash)

 _____  _              _____  ___    ___  _____
|     ||_| ___  ___   |   __||_  |  |_  ||  _  |
| | | || ||  _|| . |  |__   | _| |_ |  _||     |
|_|_|_||_||___||___|  |_____||_____||___||__|__|
-------------------------------------------------
   ROM Type:release / Ver:1.76.54
-------------------------------------------------

root@S12A:/#
```
[File was uploaded to temporary storage and will be deleted in 1 week](https://limewire.com/d/de13ac3b-9a72-4b22-8741-f5afc828e96d#GDL-QsdBj62a5-dgFHik9TVz3rEXjmHFwyu6ui-xzrQ)

Additional info

**uboot settings**

```
root@S12A:/# fw_env -p
[ubootenv] key: [aml_dt]
[ubootenv] value: [xiaomi_s12a_v01]

[ubootenv] key: [baudrate]
[ubootenv] value: [115200]

[ubootenv] key: [board_id]
[ubootenv] value: [2]

[ubootenv] key: [boot_failcnt]
[ubootenv] value: [0]

[ubootenv] key: [boot_failed]
[ubootenv] value: [if itest ${boot_failcnt} == 1; then setenv boot_failcnt 2; setenv boot_part boot1; else if itest ${boot_failcnt} == 2; then setenv boot_failcnt 1; setenv boot_part boot0; else run set_boot_flag;fi;fi;]

[ubootenv] key: [boot_part]
[ubootenv] value: [boot1]

[ubootenv] key: [bootargs]
[ubootenv] value: [rootfstype=ramfs init=/init console=ttyS0,115200 no_console_suspend quiet earlycon=aml_uart,0xff803000 jtag=apao reboot_mode=normal uboot=U-Boot 2015.01 (Sep 06 2019 - 07:13:45)]

[ubootenv] key: [bootcmd]
[ubootenv] value: [run storeboot]

[ubootenv] key: [bootdelay]
[ubootenv] value: [15]

[ubootenv] key: [deviceid]
[ubootenv] value: [18090/981388144]

[ubootenv] key: [dtb_mem_addr]
[ubootenv] value: [0x1000000]

[ubootenv] key: [factory_detect]
[ubootenv] value: [if keyman read deviceid ${loadaddr} str; then echo HAVE SN Code ...; else echo Switch to boot1 system, because of NOT found SN Code ...; setenv boot_failcnt 2; setenv boot_part boot1; fi;]

[... 128 lines cut ...]
saveenv - save environment variables to persistent storage
sdc_burn- Burning with amlogic format package in sdmmc
sdc_update- Burning a partition with image file in sdmmc card
set_trim_base- cpu temp-system
set_usb_boot- set usb boot mode
setenv  - set environment variables
showvar - print local hushshell variables
sleep   - delay execution for some time
store   - STORE sub-system
systemoff- system off
temp_triming- cpu temp-system
test    - minimal test like /bin/sh
true    - do nothing, successfully
unpackimg- un pack logo image into pictures
update  - Enter v2 usbburning mode
usb     - USB sub-system
usb_burn- Burning with amlogic format package in usb
usb_update- Burning a partition with image file in usb host
usbboot - boot from USB device
version - print monitor, compiler and linker version
vpp     - vpp sub-system
watchdog- enable or disable watchdog
write_trim- cpu temp-system
write_version- cpu temp-system
```

MrC0nst, 2025-02-17:
@duhow do You have any guide how to build docker container with all this stuff? I'm not familiar enough with docker and github to build a container with the correct settings.

noviardi1986, 2025-02-27:
Hi MrC0nst would mind to upload the firmware again?

Artkkon, 2025-03-27:
Привет. Не сохранились ли у тебя файлы с прошивкой этого хитрого китайского девайса?

dth97410, 2025-04-29:
Hi 
I came across your post on GitHub regarding the Xiaomi Mi AI Speaker (S12A / MDZ-25-DT) and saw that you had shared a modified firmware file (s12a_mico_firmware_4babb_1.76.54.bin). Unfortunately, the download link is no longer available.
Would you be willing to re-upload the file or share a new download link? I’m currently trying to restore or modify my speaker, and your work would be incredibly helpful.
Thank you i

cybergrom, 2026-11-06:
Confirmed: xiaoai-patch builds, flashes and boots on real S12A hardware.

Mi AI Speaker 1st Gen (S12A / MDZ-25-DT). It's been up and stable, running the patched
"Reboot" firmware with procd as init. Proof from the unit:

    # uname -a
    Linux S12A 4.9.61 #1 SMP PREEMPT ... aarch64 GNU/Linux
    # /etc/os-release
    PRETTY_NAME="LEDE Reboot SNAPSHOT"   ID="lede"   BUILD_ID="70-1-1"
    # /proc/1/comm
    procd
    # amlogic-dt-id
    xiaomi_s12c_v02
    # uptime
    up 5:56, load average 0.17  (stable)
    ubus/procd responsive

Everything works on the patched image: buttons (evdev keys-cyttsp), the 12-LED
is31fl3236a ring, the ad82584f speaker amp, and the PDM mic.

Hardware difference you asked about: my board's amlogic-dt-id is **xiaomi_s12c_v02**,
while the earlier report here matched xiaomi_s12a_v01 — same model/SoC/U-Boot 2015.01,
different board-revision DTB (so the correct DTB must be picked per unit).

I can open a PR with a detailed research/s12a/ page: UART + root procedure, mtd / A-B
partition layout, squashfs-RO root + UBI /data, build notes, and the full hardware map
(audio / buttons / LEDs). @duhow — which MODEL codename / make targets do you want for it?