duhow/xiaoai-patch #114: misprint in duhow/xiaoai-patch/blob/master/research/lx06/install.md

evgental, 2025-03-13:
on page https://github.com/duhow/xiaoai-patch/blob/master/research/lx06/install.md
in section backup, where we doing backup of original mtdblocks misprint where we backing up system0
update.exe mread store system0 normal _0x2820000_ mtd4.img 
should be 
update.exe mread store system0 normal **0x2800000** mtd4.img

evgental, 2025-03-16:
And so after several days of trying I was finally able to get root access to the speaker. What was done for this:
1 - Flashing the file from [research/lx06/install.md](https://github.com/duhow/xiaoai-patch/blob/04dd4d8b1c38f1ca079cd9723898fa1e6c50ade5/research/lx06/install.md#option-b-the-manual-way-credits-goes-to-httpjavabincn2021xiaoai_fmhtml)
### **option b: the manual way**
2 - loading this file via update.exe partition system1 system1.img
setting the boot order via ttl setenv boot_part boot1
3 - after flashing, root access to the system appears via ttl/ssh
4 - copying the original firmware using dd if=/dev/mtd4 of=/tmp/system0.img and then to the PC using SCP
5 - editing the original firmware file according to the [instructions](https://github.com/duhow/xiaoai-patch/blob/04dd4d8b1c38f1ca079cd9723898fa1e6c50ade5/research/lx06/install.md#option-b-the-manual-way-credits-goes-to-httpjavabincn2021xiaoai_fmhtml)
6 - loading the original_modified firmware into the device.
update.exe partition system1 system1_modified.img
7 - now the device has the original firmware ver.1.88.55 with root access and ssh is enabled.
I'm stuck here, because I can't figure out what to do next and how to make the device have a web interface on port 80

[image]

duhow, 2025-03-16:
Got a interesting key in here: 

> Parameters that correspond to my speaker

It seems that default partition size may differ, probably depending on the board ID that we use.
For instance, my Board ID `1 : LX06  8977 256MB` has this partition layout, which differs from yours.

```
mtd4: 02820000 00020000 "system0"
mtd5: 02800000 00020000 "system1"
mtd6: 013e0000 00020000 "data"
```

On the other side, since you flashed a custom image by yourself, you can add any programs that you'd like.
If you want the web interface for the speaker, make sure to build the package dependencies - or simply download and flash the prebuilt image.

duhow, 2025-03-16:
What are the steps you did to flash the image?

evgental, 2025-03-16:
> What are the steps you did to flash the image?

Download mico_firmware_f238c2e_241231_lx06.tar, unpack it.
Take rootfs.squahfs and boot.img
1st flash boot.img with update.exe partition boot0 boot.img
2nd flash root.squashfs with update.exe partition system0 root.squashfs

evgental, 2025-03-16:
Also I wondered that my speake has only 128 of ram, it shows when stop uboot.

[image]

omnight, 2025-04-06:
@evgental 

I also ran into this issue, newly purchased LX06 speaker got stuck in a boot loop when attempting to apply the prebuilt image.

I am far from a Linux expert but tried my hand at getting it working and made some progress.

I extracted the system partition mtd4 from my device I made as a backup, and then ran the patches via the makefile. Some patches didn't apply but I think that's ok since there are multiple versions of them. I missed to build the docker packages first time, so there was a lot of stuff missing.

The makefile release step didn't quite work for some reason but I manually packed the system image back using mksquashfs. I also restored the original boot image to the device.

From here on the device booted but there were a few other issues

Speaker would not connect to wifi. I figured out it was using the incorrect path for the wifi settings which was pointing to /data/wifi/wpa.conf instead of the expected /data/wifi/wpa_supplicant.conf

This can be edited in /etc/init.d/wireless

```
#WIRELESS_CONF="/data/wifi/wpa.conf"
WIRELESS_CONF="/data/wifi/wpa_supplicant.conf"
```

After I got the device to connect it still would not get an IP. Turns out udhcpc was not starting automatically, ok just create a new symlink in **etc/rc.d/** 

`S62dhcpc -> ../init.d/dhcpc`

That makes the dhcpc client starts after the wireless service, but it still doesn't start automatically when doing initial config (requires a reboot).

Possibly there are some more things that need an adjust, but at least it can play music and react to voice commands.

duhow, 2025-04-16:
> It trying to boot from mtd5 which not ready at the moment

@evgental If you flashed `system0`, that is `mtd4`. You should flash also `system1` and `boot1` if they are empty.

duhow, 2025-04-16:
@omnight if you manually patched the same firmware version from your backup, probably it is newer than the **Target version** which the patch is meant for, this is `1.74.10` in LX06.
If you have any additional log details regarding the boot loop you faced and how did you do the flashing might help.